espressif/esp-idf #9059: esp32c3: Unreliable connection with modem sleep and CONFIG_ESP_PHY_MAC_BB_PD=y

HiFiPhile, 2022-05-30:
## Environment

- Development Kit:     ESP32-C3-DevKitC-02
- Kit version v1.1
- Module or chip used: ESP32-C3-WROOM-02
- IDF version (run ``git describe --tags`` to find it):
    v5.0-dev-3055-g179ea878c7
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    esp-2021r2-patch3-8.4.0
- Operating System:     Windows
- (Windows only) environment type: Plain Command Prompt
- Using an IDE?: VS Code
- Power Supply:         USB

## Problem Description

With CONFIG_ESP_PHY_MAC_BB_PD and modem sleep enabled, WiFi connection become unreliable and fail to send tcp packet after some time, then WiFi disconnect and unable to reconnect.

With 4 devkits tested:
- 2 doesn't fail after 24 hours
- 1 fail in about 10 hours
- 1 always fail in 3-5 minutes
So it looks like the issue is related to chip process or external components variation.

To eliminate other factors, I've:
- Ensure WiFi signal strength is good, about -45dBm
- Reboot router
- Change USB port and verify power supply with oscilloscope

### Expected Behavior

A stable WiFi connection with CONFIG_ESP_PHY_MAC_BB_PD and modem sleep enabled.

### Actual Behavior

With attached minimal TCP example, after some time TCP socket send fail with error:
```
E (507881) power_save: Socket unable to connect: errno 113
E (507881) power_save: Error occurred during sending: errno 113
```

Wait a little longer WiFi will disconnect and unable to reconnect.

### Steps to reproduce

1. Create a TCP socket test server
2. Change WiFi credential & TCP server address of attached example
3. Start example and wait

### Code to reproduce this issue
[power_save.zip](https://github.com/espressif/esp-idf/files/8797012/power_save.zip)

## Debug Logs
[Full log](https://github.com/espressif/esp-idf/files/8797003/logfail.txt)

```
I (577881) power_save: Socket created, connecting to 51.75.13.186:8300
E (584381) power_save: Socket unable to connect: errno 113
E (584381) power_save: Error occurred during sending: errno 113
I (586381) power_save: Socket created, connecting to 51.75.13.186:8300
E (592881) power_save: Socket unable to connect: errno 113
E (592881) power_save: Error occurred during sending: errno 113
I (594881) power_save: Socket created, connecting to 51.75.13.186:8300
E (601381) power_save: Socket unable to connect: errno 113
E (601381) power_save: Error occurred during sending: errno 113
I (603381) power_save: Socket created, connecting to 51.75.13.186:8300
E (609881) power_save: Socket unable to connect: errno 113
E (609881) power_save: Error occurred during sending: errno 113
I (611881) power_save: Socket created, connecting to 51.75.13.186:8300
E (618381) power_save: Socket unable to connect: errno 113
E (618381) power_save: Error occurred during sending: errno 113
I (619681) wifi:bcn_timout,ap_probe_send_start
I (620381) power_save: Socket created, connecting to 51.75.13.186:8300
I (622191) wifi:ap_probe_send over, resett wifi status to disassoc
I (622191) wifi:state: run -> init (c800)
I (622191) wifi:pm stop, total sleep time: 418668340 us / 648823779 us
I (622191) wifi:new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
E (622201) power_save: Socket unable to connect: errno 113
E (622201) power_save: Error occurred during sending: errno 113
I (624211) power_save: Socket created, connecting to 51.75.13.186:8300
E (624211) power_save: Socket unable to connect: errno 118
E (624211) power_save: Error occurred during sending: errno 128
I (626211) power_save: Socket created, connecting to 51.75.13.186:8300
E (626211) power_save: Socket unable to connect: errno 118
E (626211) power_save: Error occurred during sending: errno 128
W (627491) wifi:m f probe req l=0
W (627611) wifi:m f probe req l=0
W (627741) wifi:m f probe req l=0
W (627861) wifi:m f probe req l=0
I (628211) power_save: Socket created, connecting to 51.75.13.186:8300
E (628211) power_save: Socket unable to connect: errno 118
E (628211) power_save: Error occurred during sending: errno 128
W (628701) wifi:m f probe req l=0
W (628821) wifi:m f probe req l=0
W (628941) wifi:m f probe req l=0
W (629061) wifi:m f probe req l=0
W (629181) wifi:m f probe req l=0
W (629301) wifi:m f probe req l=0
W (629421) wifi:m f probe req l=0
W (629541) wifi:m f probe req l=0
W (629661) wifi:m f probe req l=0
W (629781) wifi:m f probe req l=0
W (629901) wifi:m f probe req l=0
```

## Other items if possible

In attached code.
- [x] sdkconfig file (attach the sdkconfig file from your project folder)
- [x] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)

HiFiPhile, 2022-05-30:
Log with `Level=verbose`: 
[verbose.txt](https://github.com/espressif/esp-idf/files/8797149/verbose.txt)

Strangely no more `W (628941) wifi:m f probe req l=0` with verbose log.

HiFiPhile, 2022-07-19:
This issue has been marked done, is there any update ?

igrr, 2022-07-19:
The fix to the Wi-Fi driver has been merged internally (hence the internal task was marked "done"). This Github issue will be automatically closed when the master branch is synchronized to Github.

Alvin1Zhang, 2022-07-20:
Thanks for reporting, fix on master branch is available at https://github.com/espressif/esp-idf/commit/27d096fb63f240735fe8d74cfdb87de703990b7b and on release/4.4 is available at https://github.com/espressif/esp-idf/commit/d446c4a18ff0d17c82fccea200f4ee1abd4cbec0, feel free to reopen.

closed 2022-07-20 by Alvin1Zhang

AxelLin, 2022-07-20:
@Alvin1Zhang 
Could you clarify if v4.3 also needs this fix?