espressif/esp-idf #9060: Could not determine GDB version after sending (-

ddomnik, 2022-05-30:
## Environment

- Development Kit:  ESP32-DevKitC
- Kit version (DevKitC): [v4]
- Module or chip used:  |ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it):
    commit 31b7694551620522cc3fde0623321a38bfea762f (grafted, HEAD -> master, origin/master, origin/HEAD)
    Author: Roland Dobai <roland@espressif.com>
    Date:   Mon May 16 21:50:16 2022 +0800
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: Eclipse Espressif
- Power Supply:  

## Problem Description

Running hello_world example in debug mode returns the following GDB error:
`Could not determine GDB version after sending: C:\Espressif\tools\xtensa-esp-elf-gdb\11.1_20220318\xtensa-esp-elf-gdb\bin\xtensa-esp32-elf-gdb.exe --version, response: 
`

### Misc
Running `.\xtensa-esp32-elf-gdb.exe --version` manually as shown in the image below, prints a version as expected:
```
PS C:\Espressif\tools\xtensa-esp-elf-gdb\11.1_20220318\xtensa-esp-elf-gdb\bin> .\xtensa-esp32-elf-gdb.exe --version
GNU gdb (esp-gdb) 11.1_20220318
```

### Steps to reproduce
1. Using Espressif IDF 5.0 from master
2. Import hello_world
3. Create Launch Configuration: ESP-IDF GDB OpenOCD Debugging
4. Leave everything at default values in Launch Config
5. Select "Debug" and newly created Launch Config.

// If possible, attach a picture of your setup/wiring here.
[image]

[hello_world ESP32 DEBUG.launch.txt](https://github.com/espressif/esp-idf/files/8798258/hello_world.ESP32.DEBUG.launch.txt)

Lapshin, 2022-05-30:
Hi @ddomnik,

Could you please run export script (`export.bat`) from the root of idf directory and post the result here?

kolipakakondal, 2022-05-31:
Hi @ddomnik  You could run `Espressif > Product Information` from the main toolbar to retrieve all the tools information and paths configured.

ddomnik, 2022-05-31:
@Lapshin 
```
PS C:\Espressif\frameworks\esp-idf-master> .\export.bat
Checking Python compatibility
Setting IDF_PATH: C:\Espressif\frameworks\esp-idf-master

Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool cmake found in PATH: 3.20.0.
Not using an unsupported version of tool ninja found in PATH: 1.11.0.
Not using an unsupported version of tool ccache found in PATH: 3.1.9.
    C:\Espressif\tools\xtensa-esp-elf-gdb\11.1_20220318\xtensa-esp-elf-gdb\bin
    C:\Espressif\tools\riscv32-esp-elf-gdb\11.1_20220318\riscv32-esp-elf-gdb\bin
    C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin
    C:\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin
    C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin
    C:\Espressif\tools\xtensa-clang\12.0.1-d9341b81fc\xtensa-esp32-elf-clang\bin
    C:\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin
    C:\Espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin
    C:\Espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin
    C:\Espressif\tools\cmake\3.20.3\bin
    C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20220411\openocd-esp32\bin
    C:\Espressif\tools\ninja\1.10.2\
    C:\Espressif\tools\idf-exe\1.0.3\
    C:\Espressif\tools\ccache\4.3\ccache-4.3-windows-64
    C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
    C:\Espressif\python_env\idf5.0_py3.8_env\Scripts

Checking if Python packages are up to date...
Downloading https://dl.espressif.com/dl/esp-idf/espidf.constraints.v5.0.txt
Destination: C:\Espressif\espidf.constraints.v5.0.txt.tmp
Done
Constraint file: C:\Espressif\espidf.constraints.v5.0.txt
Requirement files:
 - C:\Espressif\frameworks\esp-idf-master\tools\requirements\requirements.core.txt
Python being checked: C:\Espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe
Python requirements are satisfied.

Detected installed tools that are not currently used by active ESP-IDF version.
For removing amazon-corretto-11-x64-windows-jdk, espressif-ide, idf-driver, xtensa-clang use command 'python.exe C:\Espressif\frameworks\esp-idf-master\tools\idf_tools.py uninstall'
For free up even more space, remove installation packages of those tools. Use option 'python.exe C:\Espressif\frameworks\esp-idf-master\tools\idf_tools.py uninstall --remove-archives'.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build
```
The error persists.

@kolipakakondal 
```
Executing C:\Espressif\python_env\idf5.0_py3.10_env\Scripts\python.exe C:\Espressif\frameworks\esp-idf-master\tools\idf_tools.py list
* xtensa-esp-elf-gdb: GDB for Xtensa
  - 11.1_20220318 (recommended, installed)
* riscv32-esp-elf-gdb: GDB for RISC-V
  - 11.1_20220318 (recommended, installed)
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2021r2-patch3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2021r2-patch3-8.4.0 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2021r2-patch3-8.4.0 (recommended, installed)
* xtensa-clang: LLVM for Xtensa (ESP32, ESP32-S2) based on clang (optional)
  - 12.0.1-d9341b81fc (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-2021r2-patch3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 and ESP32-S3 ULP coprocessors
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
[... 2 lines cut ...]
  - v0.11.0-esp32-20220411 (recommended, installed)
* ninja: Ninja build system
  - 1.10.2 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.3 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 4.3 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)

IDF_PATH: C:\Espressif\frameworks\esp-idf-master
IDF_PYTHON_ENV_PATH: C:\Espressif\python_env\idf5.0_py3.10_env
PATH: C:\Espressif\tools\xtensa-esp-elf-gdb\11.1_20220318\xtensa-esp-elf-gdb\bin;C:\Espressif\tools\riscv32-esp-elf-gdb\11.1_20220318\riscv32-esp-elf-gdb\bin;C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin;C:\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin;C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin;C:\Espressif\tools\xtensa-clang\12.0.1-d9341b81fc\xtensa-esp32-elf-clang\bin;C:\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Espressif\tools\cmake\3.20.3\bin;C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20220411\openocd-esp32\bin;C:\Espressif\tools\ninja\1.10.2\;C:\Espressif\tools\idf-exe\1.0.3\;C:\Espressif\tools\ccache\4.3\ccache-4.3-windows-64;C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:\Espressif\python_env\idf5.0_py3.10_env\Scripts;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\cygwin64\bin;C:\Program Files\PuTTY\;C:\Espressif\frameworks\esp-idf-master\tools;C:\Espressif\tools\cmake\3.20.3\bin;C:\E [... cut ...] \AppData\Local\Programs\Git\cmd;C:\Users\ddufner\AppData\Local\Programs\Microsoft VS Code\bin

Operating System: windows 10
Java Runtime Version: 11.0.15+9-LTS
Eclipse Version: 4.22.0.v20211124-1800
Eclipse CDT Version: 10.5.0.202109270023
IDF Eclipse Plugin Version: 2.4.2.202203081345
fatal: No names found, cannot describe anything.
WARNING: Git version unavailable, reading from source
ESP-IDF v5.0.0
Python set for IDF_PYTHON_ENV: Python 3.10.4
Python declared in PATH variable: Python 3.8.0
```

Lapshin, 2022-05-31:
Seems the error is gone after updating gdb to https://github.com/espressif/binutils-gdb/releases/tag/esp-gdb-v11.2_20220529

ddomnik, 2022-05-31:
Can confirm the new version works. Thanks

FanhuaCloud, 2022-05-31:
Thanks, I had the same problem and it's resolved now