espressif/esp-idf #9065: Error in Bluedroid api in union spp_data_ind_evt_param

Make4soul, 2022-05-31:
`uint16_t            len;  `  need to replace with  `uint32_t            len;  `  
Otherwise, the pointer is offset.
https://github.com/espressif/esp-idf/blob/a2d50414922aa51b107a13a7cd42cb4661e509bc/components/bt/host/bluedroid/api/include/api/esp_spp_api.h#L178

BetterJincheng, 2022-05-31:
@Make4soul Thank you for your feedback. Could you please be more specific? Thank you!

Make4soul, 2022-06-01:
I use bluedroid classic.
In the next part of the code, the length of the message and the pointer to the data are not valid.
``` cpp
esp_spp_cb(esp_spp_cb_event_t event, esp_spp_cb_param_t *param){
   switch (event){
         case ESP_SPP_DATA_IND_EVT:
               param->data_ind.len;
                param->data_ind.data;
         break;
      }
}
```

BetterJincheng, 2022-06-01:
That doesn't seem reasonable, I hope you can check again. Alternatively, attach your complete code and I'll dig into it. Thanks

Make4soul, 2022-06-04:
I checked again...
Indeed, this change does not matter. In the new idf, everything works without such changes. Apparently the reason was something else. Thanks!

closed 2022-06-04 by Make4soul