espressif/esp-idf #9067: Unexpected bahaviour touch sensor on ESP32-S3 measurement time

Phigu77, 2022-05-31:
## Environment

- Development Kit: ESP32-S3-DevKitC-1
- IDF version v4.4.1
- Build System: CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.4.0
- Operating System:     Windows
- Using an IDE: VSCode
- Power Supply: USB

## Problem Description
On the dev-board the example code touch_sensor_v2 produces unexpected behavior. I tried 2 different boards and different pins.

### Expected Behavior
According to the documentation, the measurement time is set with touch_pad_set_meas_time(). The sensor value is supposed to be the count of charge and discharge cycles within the fixed measurement period. My ESP32-Module behaves exactly as described in the documentation.

### Actual Behavior
The ESP32-S3 however extends the measurement time when a sensor is being touched. The raw value increases value returned by touch_pad_read_raw_data() while being touched rather than decreasing. The measurement time is being increased with an increase of the measured capacity.

Measurement without touch (raw value ~68.000)
[image]

Measurement with touch (raw value ~270.000)
[image]

### Steps to reproduce

1. Compile and flash example touch_sensor_v2
2. Connect Oscilloscope and touch pad to any touch-pin.

L-KAYA, 2022-06-01:
Yes, the touch sensor on ESP32 and ESP32-S3 are different. Actually, the touch sensor on ESP32-S2 and ESP32-S3 adopt an opposite way to  measure the capacity, they will fixed  the count of charge and discharge cycles and then record the count of the clock cycles during the sensing period. 

That is to say, the result of these two version are also opposite, on ESP32, the capacity will get bigger with finger put on the touch pad, then the touch sensor will count less charge and discharge cycles in a fixed period, however, on ESP32-S2 and ESP32-S3, it will count more clock cycles with the fixed charge and discharge cycles.

Phigu77, 2022-06-01:
Thanks for the clarification regarding the measurement principle. I am referring to the [programming guide for ESP32-S3](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/touch_pad.html). From what you describe, it needs to be updated.
What is also very misleading is the function touch_pad_set_meas_time() used to specify a fixed measurement time, which it is obviously not longer doing. What is it doing instead? With the procedure you describe, I would have expected an API that lets me specify the number of charge and discharge cycle within one measurement cycle. Is such a function implemented?
Can you anticipate, when the documentation will be updated?

L-KAYA, 2022-06-01:
I agree, the API name `touch_pad_set_meas_time` are same on both two versions is quite misleading, but this API can still specify the number of charge and discharge cycle in one measurement cycle, the second parameter `meas_times` is exactly the charge and discharge times, and an appropriate value recommended to to keep the measurement time about 1 ms.

Since the API has public, its name can't be updated easily, we'll add detailed description for this API as soon as possible

Phigu77, 2022-06-01:
OK, thank you!

krzychb, 2022-06-05:
We will keep this issue open and close it once documentation is updated.

closed 2022-07-07 by espressif-bot