espressif/esp-idf #9071: app_version not set to git describe when building with docker image espressif/idf:v4.4.1 inside a github actions workflow

TonyWu98, 2022-05-31:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32
- IDF version (run ``git describe --tags`` to find it): v4.4.1
- Build System:         idf.py
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         Battery

## Problem Description

I have a github actions workflow to build our esp-idf project similar to this:

```
name: CI

on:
  push:
    branches: 
      - master
  pull_request:
    branches:
      - master

  workflow_dispatch:

jobs:
  build:
    runs-on: ubuntu-latest

    steps:
      - uses: actions/checkout@v3
        with:
          submodules: 'recursive'

      - name: Build with esp-idf v4.4.1
        shell: bash
        run: docker run --rm -v $(pwd):/project -w /project espressif/idf:v4.4.1 idf.py build

      - name: Upload Build Artifacts
        uses: actions/upload-artifact@v2
        with:
          name: build
          path: |
            ${{ env.ESP_APP_FOLDER }}/build/
            !**/esp-idf/
```

Upon updating from espressif/idf:v4.3.1 to espressif/idf:v4.4.1, the `app_version` is no longer set to the output of `git describe`, it falls back to "1".

It may have something to do with git describe failing inside the container as attempting to use an interactive version of the image (`docker run --rm -v $(pwd):/project -w /project/application -it espressif/idf:v4.4.1`) on a Ubuntu WSL2 instance results in:

```
fatal: unsafe repository ('/project' is owned by someone else)
To add an exception for this directory, call:

        git config --global --add safe.directory /project
```

I attempted some of the workarounds discussed [here](https://github.com/actions/checkout/issues/766) with no results.

### Expected Behavior

`app_version` is set to the output of `git describe --always --tags --dirty`

### Actual Behavior

`app_version` is set to 1

### Steps to reproduce

1. Set up git workflow similar to one mentioned above
2. Run git workflow to build esp idf app binary and upload to workflow artifacts.
3. Check resulting artifact app binary for `app_version`.

## Debug Logs

This is a snippet of the logs from the github actions step:
```
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

-- Found Git: /usr/bin/git (found version "2.25.1") 
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/esp/tools/xtensa-esp32-elf/esp-20[21](https://github.com/Oto-Lawn-Care/OtO_FW_3.0/runs/6679544975?check_suite_focus=true#step:5:22)r2-patch3-8.4.0/xtensa-esp[32](https://github.com/Oto-Lawn-Care/OtO_FW_3.0/runs/6679544975?check_suite_focus=true#step:5:33)-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /project/application/sdkconfig
-- Found Python3: /opt/esp/python_env/idf4.4_py3.8_env/bin/python3.8 (found version "3.8.10") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "project" version: 1
```

Note lines:

`-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.`
`-- App "project" version: 1`

## Other items if possible

N/A

TonyWu98, 2022-06-01:
I've found a temporary workaround in the git actions workflow:

```
      - name: Add safe directory to global git config
        shell: bash
        run: git config --global --add safe.directory /project

      - name: Build with esp-idf v4.4.1
        shell: bash
        run: |
          docker run -v ~/.gitconfig:/etc/gitconfig -v $(pwd):/project -w /project/${{ env.ESP_APP_FOLDER }} espressif/idf:v4.4.1 idf.py build
```

The first step adds `/project` as a safe directory to the global git config, which is stored in `~/.gitconfig`.

During the docker run, the global git config is mounted to the idf container which allows it to see the `safe.directory` setting and allow it to run git describe.

atanisoft, 2022-06-02:
You could use https://github.com/espressif/esp-idf-ci-action which allows specifying the version, it uses a similar approach of mapping directories into the docker. I can confirm that the version is correctly picked up from git using the esp-idf-ci-action action.

closed 2023-05-26 by espressif-bot

MAVProxyUser, 2024-06-29:
@TonyWu98  can you share the PyOto repo? It is missing from https://github.com/Oto-Lawn-Care/