espressif/esp-idf #9072: [Check core dump data integrity on boot] case delay

sxsx2yzyz, 2022-06-01:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-WROVER-E
- IDF version (run ``git describe --tags`` to find it):v4.4.1
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     macOS
- Power Supply:         external 3.3V

## Problem Description

    Enable [Check core dump data integrity on boot] and [Core dump data format (ELF format)] will cause a 3 second delay at startup.

### Expected Behavior
Call mian immediately
### Actual Behavior
Call mian  about 3 sencond after boot

I find it a bit odd that this checksum is taking so long. If you guys notice the problem I will close it.

gerekon, 2022-06-01:
Hi @sxsx2yzyz 

Just checked this on blink example and did not observe the delay you mentioned. 
```
I (324) esp_core_dump_flash: Init core dump to flash
I (329) esp_core_dump_flash: Found partition 'coredump' @ 110000 65536 bytes
I (347) esp_core_dump_flash: Core dump data checksum is correct
I (347) esp_core_dump_flash: Found core dump 9060 bytes in flash @ 0x110000
I (351) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (361) example: Example configured to blink GPIO LED!
I (371) gpio: GPIO[5]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (381) example: Turning the LED OFF!
```
What project do you use? Is it possible for you to share it? Or at least its `sdkconfig`

sxsx2yzyz, 2022-06-02:
> Are other flash operations works well/fast? Could you also provide UART output from `idf.py monitor`?

yes other flash operations works well.
This problem occurs when the coredump partition has the correct data.

idf.py monitor:
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0038,len:2720
load:0x40078000,len:23072
load:0x40080400,len:3612
0x40080400: _init at ??:?

entry 0x400805f0
E (00:00:03.997) main: main 1  heap : 189747
```

sxsx2yzyz, 2022-06-02:
> Hi @sxsx2yzyz
>
> Just checked this on blink example and did not observe the delay you mentioned.
>
> ```
> I (324) esp_core_dump_flash: Init core dump to flash
> I (329) esp_core_dump_flash: Found partition [...]

ok
```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET_ARCH_XTENSA=y
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_TARGET_ESP32=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32-elf-"
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager

#
# Bootloader config
#
CONFIG_BOOTLOADER_OFFSET_IN_FLASH=0x1000
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
CONFIG_BOOTLOADER_LOG_LEVEL_NONE=y
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_INFO is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=0
# CONFIG_BOOTLOADER_SPI_CUSTOM_WP_PIN is not set
CONFIG_BOOTLOADER_SPI_WP_PIN=7
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
[... 1794 lines cut ...]
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
# CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
CONFIG_TCPIP_TASK_AFFINITY_CPU1=y
CONFIG_TCPIP_TASK_AFFINITY=0x1
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1=y
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
# End of deprecated options

```

sxsx2yzyz, 2022-06-02:
my partition:
```
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
phy_init,data,phy,0xf000,4K,
ota_0,    app,  ota_0,0x10000 ,1792K,
ota_1,    app,  ota_1, , 1792K,
nvs,data,nvs,,300K,
otadata,  data, ota, ,  0x2000,
coredump, data, coredump,,        64K,encrypted,
nvs_key,    data, nvs_keys,, 0x1000, encrypted,

```

closed 2022-10-21 by sxsx2yzyz