espressif/esp-idf #9074: erase_flash takes about 1 minute in some ESP32 chips

balbet, 2022-06-01:
Hi,

We designed a hardware with an ESP32 of 16MB, and we found that different devices have different "erase_flash" times. Some of them takes about 1 minute, which corresponds to the time of the datasheet, but with some others it takes like 3 seconds. So, for some reason it seems that it's not being erased correctly. We have pending to make some tests, writing some bulk data + erasing + reading + verifying that it is being erased, but we want to know the reason of that huge difference, since it's a problem for us in production.

**The ESP32 with slow (expected) time:**

`esptool.py -p COM17 read_flash_status --bytes 3`
>     esptool.py v3.1-dev
>     Serial port COM17
>     Connecting....
>     Detecting chip type... ESP32
>     Chip is ESP32-D0WD-V3 (revision 3)
>     Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
>     Crystal is 40MHz
>     MAC: xx:xx:xx:xx:xx:xx
>     Uploading stub...
>     Running stub...
>     Stub running...
>     Status value: 0x600200
>     Hard resetting via RTS pin...

`esptool.py -p COM17 flash_id`
>     esptool.py v3.1-dev
>     Serial port COM17
>     Connecting.....
>     Detecting chip type... ESP32
>     Chip is ESP32-D0WD-V3 (revision 3)
>     Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
>     Crystal is 40MHz
>     MAC: xx:xx:xx:xx:xx:xx
>     Uploading stub...
>     Running stub...
>     Stub running...
>     Manufacturer: 20
>     Device: 4018
>     Detected flash size: 16MB
>     Hard resetting via RTS pin...

We disassembled one module, and the "Manufacturer: 20" corresponds to the flash XMC XM25QH128C

**The ESP32 with fast time:**

`esptool.py -p COM17 read_flash_status --bytes 3`
>     esptool.py v3.1-dev
>     Serial port COM17
>     Connecting........___
>     Detecting chip type... ESP32
>     Chip is ESP32-D0WD-V3 (revision 3)
>     Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
>     Crystal is 40MHz
>     MAC: xx:xx:xx:xx:xx:xx
>     Uploading stub...
>     Running stub...
>     Stub running...
>     Status value: 0x200200
>     Hard resetting via RTS pin...

`esptool.py -p COM17 flash_id`
>     esptool.py v3.1-dev
>     Serial port COM17
>     Connecting....
>     Detecting chip type... ESP32
>     Chip is ESP32-D0WD-V3 (revision 3)
>     Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
>     Crystal is 40MHz
>     MAC: xx:xx:xx:xx:xx:xx
>     Uploading stub...
>     Running stub...
>     Stub running...
>     Manufacturer: c8
>     Device: 4018
>     Detected flash size: 16MB
>     Hard resetting via RTS pin...

In this case, the flash is the GigaDevice GD25Q128E. In both cases, the "Chip erase" time specified in the datasheet is much higher than 3 seconds (50 seconds typ in one case, and 55 seconds min in the other one).

Thanks!

ginkgm, 2022-08-08:
Hi @balbet ,
The driver doesn't do any magic. It's the flash's behavior. Some flash vendors may skip some erasing process if the erasing is not required, to improve the flash's performance.

Michael

balbet, 2022-08-09:
Hi Michael,

Yes, we are using the embedded module.

Thanks!

closed 2022-08-09 by balbet