espressif/esp-idf #9075: WiFi mesh broadcast message are lost

santuari, 2022-06-01:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version: v1
- Module or chip used:  ESP32-WROOM-32
```
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
Manufacturer: 5e
Device: 4016
Detected flash size: 4MB
```
- IDF version: v4.3.2
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:  Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

We discover that broadcast message are lost in the WiFi mesh. In real scenario with 50 to 150 nodes, nodes from layer 4 to 9 receives less that 5% of broadcast messages. 

We simulate a basic scenario with two ESP32 Dev Kit in WiFi mesh: 1 ROOT, 1 Child. Distance between the two nodes is less than 1 meter and there is Line Of Sight between them. 

Broadcast messages sent by root:

1. are received by the root itself that should not be case according to the documentation ("When a node receives a broadcast packet with a source address matching its own MAC address, the node will discard the broadcast packet." point 7 of [Broadcasting](https://docs.espressif.com/projects/esp-idf/en/v4.3.2/esp32/api-guides/mesh.html?highlight=routing%20table#broadcasting))
3. most of messages are not received by the connected child: it receives only 5-8% of them.

### Expected Behavior

1. All messages sent as broadcast should be received by the child.
2. Root should not receive any broadcast message that it has sent.

### Actual Behavior

1. Only 5-8% of the messages are received by the child
2. The root node also receives broadcast messages that it has sent.

### Steps to reproduce

0. Setup two nodes in the same mesh: one will be the root and the other the child.
1. Run the script in the following section.
4. Look at the logs of the two nodes.

### Code to reproduce this issue

Using [this IDF4 example](https://github.com/espressif/esp-idf/tree/master/examples/mesh/internal_communication) and modifying the code as follow: 

```cpp

uint8_t broadcast_addr[6] = { 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF };

memcpy(route_table[0].addr, broadcast_addr, sizeof(broadcast_addr));

err = esp_mesh_send(&route_table[0], &data, MESH_DATA_P2P, NULL, 0);
ESP_LOGI(MESH_TAG,"SEND --> [ROOT-2-UNICAST:%d][L:%d]parent:"MACSTR" to "MACSTR", heap:%d[err:0x%x, proto:%d, tos:%d]",
send_count, mesh_layer, MAC2STR(mesh_parent_addr.addr),
MAC2STR(route_table[0].addr),
esp_get_minimum_free_heap_size(), err, data.proto,	data.tos);
      if (err) {
          ESP_LOGE(MESH_TAG,
                   "[ROOT-2-UNICAST:%d][L:%d]parent:"MACSTR" to "MACSTR", heap:%d[err:0x%x, proto:%d, tos:%d]",
                   send_count, mesh_layer, MAC2STR(mesh_parent_addr.addr),
                   MAC2STR(route_table[0].addr), esp_get_minimum_free_heap_size(),
                   err, data.proto, data.tos);
      } else if (!(send_count % 100)) {
          ESP_LOGW(MESH_TAG,
                   "[ROOT-2-UNICAST:%d][L:%d][rtableSize:%d]parent:"MACSTR" to "MACSTR", heap:%d[err:0x%x, proto:%d, tos:%d]",
                   send_count, mesh_layer,
                   esp_mesh_get_routing_table_size(),
                   MAC2STR(mesh_parent_addr.addr),
                   MAC2STR(route_table[0].addr), esp_get_minimum_free_heap_size(),
                   err, data.proto, data.tos);
      }
```

## Debug Logs
The child receives 37 on 502 broadcast messages.
[image]
The root receives every broadcast message that it has sent: 505 on 505.

## Other items if possible

- [x] sdkconfig file: [sdkconfig.txt](https://github.com/espressif/esp-idf/files/8815435/sdkconfig.txt)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

shenjun7, 2022-06-06:
Please disable the ps function by `menuconfig->Example Configuration->Disable mesh PS (power save) function`

moefear85, 2022-06-07:
Are you using antennas? Alot of packets get dropped without them, even at short distances.

santuari, 2022-06-08:
@shenjun7 we will re-run the test with "disable of mesh power save" option. I let you know.

@moefear85 yes, we are using antennas. 

Thanks.
Michele

santuari, 2022-06-14:
"Disable of mesh power save" option seems to work for broadcast but having some nodes with this option enable and other disable is a mess: we got multiple root nodes on the same mesh (for example: 2 with disable power save and 1 with enable power save over a total of 10 nodes in the mesh).

How can we proceed in the OTA on large deployment considering that we want to disable the mesh power save on all the deployment?

shenjun7, 2022-07-19:
Please keep the same ps configuration. If you turn on the ps function, you can reduce the packet loss rate by increasing the duty.

closed 2023-03-14 by Alvin1Zhang