espressif/esp-idf #9076: 关于ESP HTTP Client的错误

HongYan12345, 2022-06-01:
我想要通过一个thingsboard的api获取账号的token
[image]
我在代码里连上wifi后，使用esp_http_client访问
[image]
但是一直出现401错误
[image]
我该如何修改它呢？

ESP-YJM, 2022-06-06:
你能否提供下你可以正确访问的完整抓包或者包头和包 payload 的数据。你也可以和你目前无法访问的进行包头的对比。

HongYan12345, 2022-06-06:
> 你能否提供下你可以正确访问的完整抓包或者包头和包 payload 的数据。你也可以和你目前无法访问的进行包头的对比。
curl -X POST --header "Content-Type: application/json" --header "Accept: application/json" -d "{\"username\":\"ivan.cmaya@alumnos.upm.es\", \"password\":\"SBC21g09\"}" "https://thingsboard.cloud:443/api/auth/login"
完整的是这样的，我感觉是我的参数的设置问题，在postman是能跑通的，我不知道该把后面{}里面的东西设置在哪里

HongYan12345, 2022-06-06:
> 你试试这样设置，我这样设置可以拿到 token。
> 
> ```
>     esp_http_client_config_t config = {
>         .event_handler = _http_event_handler,
>         .url = "https://thingsboard.cloud:443/api/auth/login",
>     };
>     esp_http_client_handle_t client = esp_http_client_init(&config);
> 
>     // POST
>     const char *post_data = "{\"username\":\"ivan.cmaya@alumnos.upm.es\", \"password\":\"SBC21g09\"}";
>     esp_http_client_set_method(client, HTTP_METHOD_POST);
>     esp_http_client_set_header(client, "Content-Type", "application/json");
>     esp_http_client_set_post_field(client, post_data, strlen(post_data));
>     esp_err_t err = esp_http_client_perform(client);
>     if (err == ESP_OK) {
>         ESP_LOGI(TAG, "HTTP POST Status = %d, content_length = %lld",
>                 esp_http_client_get_status_code(client),
>                 esp_http_client_get_content_length(client));
>     } else {
>         ESP_LOGE(TAG, "HTTP POST request failed: %s", esp_err_to_name(err));
>     }
>    
>     esp_http_client_cleanup(client);
> ```

我调通了！非常感谢！后续如果遇到问题我能继续问你吗，这是我的毕业设计

closed 2022-06-07 by Alvin1Zhang