espressif/esp-idf #9082: Compile errors from freertos/timers.h in ESP-IDF 4.4.1

mbratch, 2022-06-01:
## Environment

- Development Kit:   ESP32-PICO-Kit
- Module or chip used:  ESP32-PICO-D4
- IDF version (run ``git describe --tags`` to find it): 4.4.1
    // v3.2-dev-1148-g96cd3b75c
- Build System:     CMake or idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.4.0
- Operating System:     Windows 11
- (Windows only) environment type: PowerShell
- Using an IDE?: Yes - VS Code with ESP-IDF plugin
- Power Supply:         USB

## Problem Description

I am unable to compile a source file that includes `freertos/timers.h` unless, before that header, I also include `driver/i2c.h`

### Expected Behavior

`freertos/timers.h` should not depend upon `driver/i2c.h` to successfully compile

### Actual Behavior

I get the following compile errors attempting to compile with `freertos/timers.h` without first including `driver/i2c.h`:

`
> Executing task: ninja  <

[2/10] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Users\mdbra\Documents\Projects\ESP32\BD-OTB\build\bootloader\esp-idf\esptool_py && C:\Users\mdbra\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/Users/mdbra/Documents/Projects/ESP32/BD-OTB/build/bootloader/bootloader.bin"
Bootloader binary size 0x62c0 bytes. 0xd40 bytes (12%) free.
[3/8] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/OutIns.cpp.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/OutIns.cpp.obj
C:\Users\mdbra\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -I../include -Iconfig -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/newlib/platform_include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/freertos/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/freertos/include/esp_additions/freertos -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/freertos/port/xtensa/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/freertos/include/esp_additions -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/esp_hw_support/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/esp_hw_support/include/soc -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/esp_hw_support/include/soc/esp32 -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/esp_hw_support/port/esp32/. -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/esp_hw_support/port/esp32/private_include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/heap/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/log/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/lwip/include/apps -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/lwip/include/apps/sntp -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/lwip/lwip/src/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/lwip/port/esp32/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/lwip/port/esp32/include/arch -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/soc/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/soc/esp32/. -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/soc/esp32/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/hal/esp32/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/hal/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/hal/platform_port/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/esp_rom/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/esp_rom/include/esp32 -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/esp_rom/esp32 -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/esp_common/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/esp_system/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/esp_system/port/soc -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/esp_system/port/public_compat -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/esp32/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/xtensa/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/xtensa/esp32/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/driver/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/driver/esp32/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/esp_pm/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/esp_ringbuf/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/efuse/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/efuse/esp32/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/vfs/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/esp_wifi/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/esp_event/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/esp_netif/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/esp_eth/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/tcpip_adapter/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/esp_phy/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/esp_phy/esp32/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/esp_ipc/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/app_trace/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/esp_timer/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/mbedtls/port/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/mbedtls/mbedtls/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/mbedtls/esp_crt_bundle/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/app_update/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/spi_flash/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/bootloader_support/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/nvs_flash/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/pthread/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/esp_gdbstub/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/esp_gdbstub/xtensa -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/esp_gdbstub/esp32 -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/espcoredump/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/espcoredump/include/port/xtensa -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/wpa_supplicant/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/wpa_supplicant/port/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/wpa_supplicant/esp_supplicant/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/ieee802154/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/console -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/asio/asio/asio/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/asio/port/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/cbor/port/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/unity/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/unity/unity/src -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/cmock/CMock/src -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/coap/port/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/coap/libcoap/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/nghttp/port/include -IC:/Users/mdbra/.espressif/frameworks/esp-idf-v4.4.1/components/nghttp/nghttp2/lib/includes -IC:/Users/mdbra/.espressif/frameworks/esp- [... cut ...]
#include <freertos/timers.h>    // The source file with just this in it will result in the error
```

### Work-around

My current work around is just to include `driver/i2c.h` even though I am not needing it. If I use another include, such as `driver/adc.h`, which I do need, I still get the errors:

```cpp
#include <driver/i2c.h>        // Must include this before the following include to avoid the compile errors
#include <freertos/timers.h>    // The source file with just this in it will result in the error
```

KaeLL, 2022-06-01:
> "include FreeRTOS.h must appear in source files before include timers.h"

mbratch, 2022-06-02:
Thanks @KaeLL. What threw me here was that I was getting away without doing that for quite some time, I assume because the ESP-IDF headers I was including before timers just happened to already include FreeRTOS.h. Then recently, the header I was using from ESP-IDF just so happened not to include it.

But all good now. Again, thanks.

closed 2022-06-02 by mbratch