espressif/esp-idf #9084: wear_levelling example don't work with 512K partitions size

nopnop2002, 2022-06-02:
## Environment
```
$ idf.py --version
ESP-IDF v5.0-dev-3202-ga2d5041492-dirty
```

## Problem Description
I changed partitions_example.csv:

```
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 1M,
storage,  data, fat,            , 512K,
#storage,  data, fat,     ,        1M,
```

## Debug Logs

example don't work.

```
I (29) boot: ESP-IDF v5.0-dev-3202-ga2d5041492 2nd stage bootloader
I (29) boot: compile time 09:34:25
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot:  3 storage2         Unknown data     01 82 00110000 00010000
I (100) boot:  4 storage          Unknown data     01 81 00120000 00010000
I (108) boot: End of partition table
I (112) boot_comm: chip revision: 1, min. application chip revision: 0
I (119) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=08864h ( 34916) map
I (140) esp_image: segment 1: paddr=0001888c vaddr=3ffb0000 size=02648h (  9800) load
I (145) esp_image: segment 2: paddr=0001aedc vaddr=40080000 size=0513ch ( 20796) load
I (155) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=1a704h (108292) map
I (195) esp_image: segment 4: paddr=0003a72c vaddr=4008513c size=06954h ( 26964) load
I (206) esp_image: segment 5: paddr=00041088 vaddr=50000000 size=00010h (    16) load
I (212) boot: Loaded app from partition at offset 0x10000
I (212) boot: Disabling RNG early entropy source...
I (226) cpu_start: Pro cpu up.
I (226) cpu_start: Starting app cpu, entry point is 0x40081040
0x40081040: call_start_cpu1 at /home/nop/esp-idf/components/esp_system/port/cpu_start.c:152

I (0) cpu_start: App cpu up.
I (241) cpu_start: Pro cpu start user code
I (241) cpu_start: cpu freq: 160000000 Hz
I (241) cpu_start: Application information:
I (245) cpu_start: Project name:     wear_levelling_example
I (252) cpu_start: App version:      v5.0-dev-3202-ga2d5041492-dirty
I (259) cpu_start: Compile time:     Jun  2 2022 09:37:54
I (265) cpu_start: ELF file SHA256:  1d6e5b839bbdb1ee...
I (271) cpu_start: ESP-IDF:          v5.0-dev-3202-ga2d5041492-dirty
I (278) heap_init: Initializing. RAM available for dynamic allocation:
I (285) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (291) heap_init: At 3FFB2F90 len 0002D070 (180 KiB): DRAM
I (297) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (304) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (310) heap_init: At 4008BA90 len 00014570 (81 KiB): IRAM
I (318) spi_flash: detected chip: generic
I (321) spi_flash: flash io: dio
I (326) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (336) example: Mounting FAT filesystem
W (536) vfs_fat_spiflash: f_mount failed (13)
I (536) vfs_fat_spiflash: Formatting FATFS partition, allocation unit size=4096
E (536) vfs_fat_spiflash: f_mkfs failed (14)
E (536) example: Failed to mount FATFS (ESP_FAIL)
```

igrr, 2022-06-02:
Hi @nopnop2002,
For the default 4kB FAT sector size, 512kB is a bit too short for the minimal FAT partition plus wear levelling data. FAT has a minimum of 128 sectors, so the FAT part needs at least 512kB. Wear levelling needs another 4 sectors (16kB) at this partition size.

If you need to use a partition smaller than 528kB, you can change the FAT sector size to 512 bytes in menuconfig.

nopnop2002, 2022-06-02:
Thank you for your quick response.

I added the following to sdkconfig.defaults.

It worked.

```
#
# Wear Levelling
#
CONFIG_WL_SECTOR_SIZE_512=y
CONFIG_WL_SECTOR_SIZE=512
CONFIG_WL_SECTOR_MODE_SAFE=y
CONFIG_WL_SECTOR_MODE=1
```

closed 2022-06-02 by nopnop2002

franz-ms-muc, 2023-03-08:
same Error in this Example: 

[https://github.com/espressif/esp-idf/blob/master/examples/peripherals/i2c/i2c_tools/main/i2ctools_example_main.c](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/i2c/i2c_tools/main/i2ctools_example_main.c)