espressif/esp-idf #9085: export.sh partially fails for custom tools path

francis2tm, 2022-06-02:
## Environment

- Development Kit: ESP32-DevKit
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): v4.4.1
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.4.0
- Operating System:     [Linux]
- (Windows only) environment type: [Plain Command Prompt].
- Using an IDE?: [No]

## Problem Description
Using the new install scripts with IDF_TOOLS_PATH to install the toolchain to a custom location, sourcing export.sh outputs some warnings. I don't know if this may be affecting my builds. I have multiple python versions installed in my system (3.6.9 and 3.9). python and python3 alias to python3.9. I setup the IDF_TOOLS_PATH env variable by doing:

```
export IDF_TOOLS_PATH="<absolute path>/esp-idf-tools/"
```
Before the installation (install.sh) and before source of export.sh

### Expected Behavior
A clean export with no warnings.
### Actual Behavior

```
Detecting the Python interpreter
Checking "python" ...
Python 3.6.9
"python" has been detected
Adding ESP-IDF tools to PATH...
Using Python interpreter in /home/francis/projetos/project/project-esp32/esp-idf-tools/python_env/idf4.4_py3.6_env/bin/python
Checking if Python packages are up to date...
/home/francis/.local/lib/python3.9/site-packages/pkg_resources/__init__.py:122: PkgResourcesDeprecationWarning: 0.18ubuntu0.18.04.1 is an invalid version and will not be supported in a future release
  warnings.warn(
The following Python requirements are not satisfied:
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22
idf-component-manager~=1.0
gdbgui==0.13.2.0
pygdbmi<=0.9.0.2
python-socketio<5
jinja2<3.1  # See https://github.com/espressif/esp-idf/issues/8760
itsdangerous<2.1
kconfiglib==13.7.1
reedsolo>=1.5.3,<=1.5.4
bitstring>=3.1.6
ecdsa>=0.16.0
construct==2.10.54
To install the missing packages, please run "/home/francis/projetos/project/project-esp32/deps/esp-idf/install.sh"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: /home/francis/projetos/project/project-esp32/esp-idf-tools/python_env/idf4.4_py3.6_env
    Python interpreter used: /usr/bin/python3.9
    Warning: python interpreter not running from IDF_PYTHON_ENV_PATH
    PATH: /home/francis/projetos/project/project-esp32/esp-idf-tools/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin:/home/francis/projetos/project/project-esp32/esp-idf-tools/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin:/home/francis/projetos/project/project-esp32/esp-idf-tools/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin:/home/francis/projetos/project/project-esp32/esp-idf-tools/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin:/home/francis/projetos/project/project-esp32/esp-idf-tools/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/francis/projetos/project/project-esp32/esp-idf-tools/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/francis/projetos/project/project-esp32/esp-idf-tools/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin:/home/francis/projetos/project/project-esp32/esp-idf-tools/python_env/idf4.4_py3.6_env/bin:/home/francis/projetos/project/project-esp32/deps/esp-idf/tools
```

dobairoland, 2022-06-02:
Hi @francis2tm. This is a known issue with v4.4 and older releases. Python cannot be aliased.

```
Checking "python" ...
Python 3.6.9
"python" has been detected
....
...idf4.4_py3.6_env
Python interpreter used: /usr/bin/python3.9
```

You can see in your logs that 3.6 and 3.9 are both used.

Please use `update-alternatives`  (https://github.com/espressif/esp-idf/blob/release/v4.4/tools/docker/Dockerfile#L33) instead of aliases.

Or just remove the alias. It will detect and use `python3` instead.

Just to note for other readers, that this has been significantly re-written for v5.0 and won't cause trouble anymore.

francis2tm, 2022-06-02:
Thanks @dobairoland , that indeed fixed the problem!

closed 2022-06-02 by francis2tm