espressif/esp-idf #9086: [Bug][SPI RAM] heap_caps_malloc_prefer() should not abort on allocation failure if there is available SPI RAM

chipweinberger, 2022-06-02:
This is very similar to:

https://github.com/espressif/esp-idf/pull/7569

In functions that attempt allocation multiple times, we should only abort if *all* attempts fail.

**As currently implemented:**
```
IRAM_ATTR void *heap_caps_malloc_prefer( size_t size, size_t num, ... )
{
    va_list argp;
    va_start( argp, num );
    void *r = NULL;
    while (num--) {
        uint32_t caps = va_arg( argp, uint32_t );
        r = heap_caps_malloc( size, caps );
        if (r != NULL) {
            break;
        }
    }
    va_end( argp );
    return r;
}
```

**What it should be:**
```
IRAM_ATTR void *heap_caps_malloc_prefer( size_t size, size_t num, ... )
{
    va_list argp;
    va_start( argp, num );
    void *r = NULL;
    while (num--) {
        uint32_t caps = va_arg( argp, uint32_t );
        r = heap_caps_malloc_base( size, caps );
        if (r != NULL) {
            break;
        }
    }
    if (r==NULL){
        heap_caps_alloc_failed(size, caps, __func__);
    }
    va_end( argp );
    return r;
}
```

These 3 functions all need to be fixed in the same way:
heap_caps_malloc_prefer
heap_caps_realloc_prefer
heap_caps_calloc_prefer (for calloc we also need to implement heap_caps_calloc_base)

chipweinberger, 2022-06-02:
Here are the full changes:

[heap_caps.c.txt](https://github.com/espressif/esp-idf/files/8823468/heap_caps.c.txt)

chipweinberger, 2022-06-06:
@o-marshmallow

Maybe you cant take this again? Very simple change.

o-marshmallow, 2022-06-08:
Hi @chipweinberger ,

I am going to check this, thanks for linking the change

chipweinberger, 2022-06-08:
Thanks Omar =)

closed 2022-07-19 by espressif-bot

chipweinberger, 2022-07-19:
Yay =)

chipweinberger, 2022-07-19:
@o-marshmallow, can this be back ported to v4.4? 

I'd love for v4.4 to be as bug free as possible for those that aren't ready to switch to 5.0.

chipweinberger, 2022-08-09:
Bump. Would still really love to get this in v4.4. It is a *bugfix*, and causes crashes in my code.

o-marshmallow, 2022-08-10:
Hi @chipweinberger ,

Sorry for the delay, thanks for remainding me, I have creating an internal merge request to backport this change to v4.4