espressif/esp-idf #9087: [Crash] [Bug] malloc should return NULL when largest block size is too small. It should not abort.

chipweinberger, 2022-06-02:
ESP-IDF: v4.4.1
ESP32-S3

In heap_tlsf.c,  we throw an assert if the largest block is too small: 

**assert failed: block_locate_free heap_tlsf.c:447 (block_size(block) >= size)**

This does not let the user handle allocation failure. Worse, functions that attempt allocation multiple times (like heap_caps_malloc_prefer()) fail when they should have succeeded.

**Current implementation:**
```
static inline  __attribute__((__always_inline__)) block_header_t* block_locate_free(control_t* control, size_t size)
{
         ...... more ....

	if (block) {
		tlsf_assert(block_size(block) >= size);
		remove_free_block(control, block, fl, sl);
	}
	return block;
}
```

**What it should be:**
```
static inline  __attribute__((__always_inline__)) block_header_t* block_locate_free(control_t* control, size_t size)
{
         ...... more ....

	if (block && block_size(block) >= size) {
		remove_free_block(control, block, fl, sl);
	} else {
		block = NULL;
	}
	return block;
}
```

**Example backtrace:** 

```
**Backtrace:0x403763d3:0x3fcbd3000x4038268d:0x3fcbd330 0x4038d5c6:0x3fcbd360 0x4038b7fa:0x3fcbd480 0x4038c529:0x3fcbd4b0 0x4038ca0b:0x3fcbd4e0 0x40376c01:0x3fcbd510 0x40376c4f:0x3fcbd540 0x4037709b:0x3fcbd570 0x4037711c:0x3fcbd5a0 0x40376690:0x3fcbd5f0 0x403766c0:0x3fcbd620 0x420eacdf:0x3fcbd650 0x4038d27b:0x3fcbd680 0x40039ae1:0x3fcbd6e0  |<-CORRUPTED
0x403763d3: panic_abort at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_system/panic.c:402

0x4038268d: esp_system_abort at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_system/esp_system.c:128

0x4038d5c6: __assert_func at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/newlib/assert.c:85

0x4038b7fa: block_locate_free at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/heap/heap_tlsf.c:447
 (inlined by) tlsf_malloc at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/heap/heap_tlsf.c:757

0x4038c529: multi_heap_malloc_impl at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/heap/multi_heap.c:197

0x4038ca0b: multi_heap_malloc at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/heap/multi_heap_poisoning.c:230

0x40376c01: heap_caps_malloc_base at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/heap/heap_caps.c:147

0x40376c4f: heap_caps_malloc at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/heap/heap_caps.c:167

0x4037709b: heap_caps_calloc at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/heap/heap_caps.c:441

0x4037711c: heap_caps_calloc_prefer at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/heap/heap_caps.c:299

0x40376690: wifi_calloc at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_wifi/esp32s3/esp_adapter.c:107

0x403766c0: wifi_zalloc_wrapper at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_wifi/esp32s3/esp_adapter.c:115

0x420eacdf: ieee80211_ampdu_request at ??:?

0x4038d27b: ieee80211_encap_esfbuf at ??:?**
```

chipweinberger, 2022-06-06:
@o-marshmallow
@projectgus 

Thoughts?

I'm running into crashes when starting to run low on RAM, when I would instead like to handle allocation failure.

o-marshmallow, 2022-06-08:
Hi @chipweinberger ,

I tried with an ESP32-S3 with IDF v4.4.1 but I was unable to reproduce the issue when malloc-ing more than the maximum amount of RAM. Which setting did you enable in aprticular?

chipweinberger, 2022-06-08:
That's interesting. Maybe you can look at the code deeper to determine what scenario it would return a block too small?

https://github.com/espressif/esp-idf/issues/9088
^this describes my configuration a bit. 

CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP=y
CONFIG_SPIRAM_USE_MALLOC=y
CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=256 (I tried changing this around)
CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL=152000 (I tried changing this around)
CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL=y

o-marshmallow, 2022-06-09:
@chipweinberger I did investiguate the code to understand how this could be possible. I found that the caller of `block_locate_free` first go throught the function `adjust_request_size`, which as its name states, adjust the size to be aligned on a word size (4) **and** checks if the size requested is bigger than the maximum block size:
(`heap_tlsf.c`, line 123)
```
static inline __attribute__((__always_inline__)) size_t adjust_request_size(size_t size, size_t align)
{
	size_t adjust = 0;
	if (size)
	{
		const size_t aligned = align_up(size, align);

		/* aligned sized must not exceed block_size_max or we'll go out of bounds on sl_bitmap */
		if (aligned < block_size_max)
		{
			adjust = tlsf_max(aligned, block_size_min);
		}
	}
	return adjust;
}
```
Thus, if the (aligned) size we want to allocate is bigger than the maximum size of a block in our pool, this function returns 0, then `block_locate_free` is called with a size of 0.

I just tried with the same CONFIG options you provided and I still can't reproduce the issue. I tried allocating several big buffers (512KB, 2MB, 16MB), I didn't get any assert. From what I see in the code, the only reason we could get the assert to be triggered would be if the heap gets corrupted, which could be due to a buffer overflow, a double free, etc... 

I also tried the function you provide in #9088 to allocate memory, still not getting the assert. I know it may be hard but can you please try making a small reproductible example?

chipweinberger, 2022-06-09:
Omar,  I _really_ appreciate this great effort you have tried to reproduce the issue.

I trust that your code inspection is good. The fact that this does not reproduce after many large allocations demonstrates that my understanding of this assert was incorrect. 

**It is okay to close this issue unless I reproduce it again!**

Just for records, this was my complete sdkconfig at the time of the crash:
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/8866844/sdkconfig.txt)

chipweinberger, 2022-06-09:
I ran my code from when I filed this bug, **but I cannot get it to reproduce**. At the time, I was reproducing on every boot, so I must not have my configuration exactly the same

In the future I'll be more careful to all repro steps. (and my local commit hash, etc)

I will close this issue w/ cannot reproduce.

Thank you for everything you tried! My apologies for not recording enough details.

closed 2022-06-09 by chipweinberger

chipweinberger, 2022-06-09:
I've also closed https://github.com/espressif/esp-idf/issues/9088 since I cannot reproduce that either.

o-marshmallow, 2022-06-09:
@chipweinberger No worries, it happens sometimes :smile: 

I noticed you also had a problem with the SPIRAM (#9120). In that issue, it got corrupted because of an atomic operation. I wonder whether this heap issue is also a side effect of that SPIRAM corruption? 
As SPIRAM is be used for as a heap allocator, if it get corrupted, the assert could be fired. That would explain it.

Anyway, thank you for coming back with more info :+1:

chipweinberger, 2022-06-09:
Yes I was wondering if it was related to SPIRAM as well. 

When I filed this bug, I was still using CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=512, so I think I was immune from #9120, but its hard to remember if maybe I changed something else!  😅