espressif/esp-idf #9090: MQTT client: limit outbox size

kotyara12, 2022-06-02:
Good day!

It is necessary to add the ability to limit the size of the outbox for mqtt_client or block it altogether (disable it).

The problem is the following. On a "good" WiFi connection, everything works fine, no problems.

But not all devices have a good WiFi connection, some of my devices are at the Wifi limit, -90dBi and below. And sometimes neighbors clog the channel so tightly that packet loss reaches 100%. In this case, mqtt queues messages. And this is good. But not much. The fact is that rather large JSON packets go to MQTT, from 512 bytes and above, sometimes 2-3 kilobytes. I can't quickly determine if there is a connection to the server or not. And the messages are queued. And the memory runs out very, very quickly, the device either freezes or reboots.

It is necessary to somehow limit the size of the outgoing mailbox, after which the "old" messages will be overwritten by new ones. I haven't found such an option yet.
And besides, it is very important that messages in the same topic are not duplicated in the outbox, but are filled with new ones.

euripedesrocha, 2022-08-02:
Hi @kotyara12 thanks for the suggestion. I'm looking into adding more control over the outbox, but you can already disable it via menuconfig. It's also possible to define a timeout for messages, using these two options you may have a temporary solution for your issue.

kotyara12, 2022-08-04:
Hello @euripedesrocha.

I don't quite understand how to disable the send queue via menuconfig. I understand that in order to use a custom implementation, you need to write it, and I do not have enough knowledge in this matter.
But this is not required. In fact, the send queue is a great feature and avoids problems in some situations when the device generates several packets at once to send to mqtt and "does not have time" to send them all. But there are times when this functionality brings problems. For example, right now the power was turned off, the devices switched to battery operation, but unfortunately the provider's equipment is apparently de-energized.
I've gotten around so far by checking the amount of free heap before JSON is generated and sent. Devices began to work much more stable, failures due to heap exhaustion were no longer observed.
However, if such functionality is at the library level, it will be "much more correct", in my opinion.
Thank you.

PS: In general, I am completely delighted with the multitasking ESP-IDF, after Arduino.

euripedesrocha, 2022-08-04:
Hi @kotyara12,
we agree that this a good feature to have in esp-mqtt client, and I'm working on it already.

I only pointed the custom outbox route because it's a way for you to not be blocked by our implementation time.

kotyara12, 2022-09-13:
Hello @euripedesrocha.

Pay attention to this mechanism:

Let's say we send a message "{big-big-json}" to the topic "device/topic1"
But there is no connection with the broker (for any reason), it is put in the outbox queue. After some time (but before the message storage timeout expires), another message is added to the SAME topic "device/topic1", but with different data.
In order not to know the extra space in the outbox, you need to somehow replace the data in the queue (deleting the old ones) and update the addition time.

Thank you.

matiasstack, 2022-12-05:
Hello, I also agree that this would be a great feature to have.
@euripedesrocha you mentioned the custom outbox option, which looks quite interesting. Could you point me to some existing implementation to use as "inspiration" for a custom one?
Thanks!

AxelLin, 2023-08-28:
@euripedesrocha 
Looks like this feature is implemented in esp-mqtt.
However, the esp-idf still uses a very old version of esp-mqtt (5afa3d347c189217c087a098f1bcfc2d3f52ee1f).
As a result, people cannot test latest esp-mqtt code. Any chance to update esp-mqtt submodule in esp-idf?

euripedesrocha, 2023-08-28:
@AxelLin, mqtt submodule is in process of update in the latest idf master and 5.1.

For 5.0, that you are referring to, there is still an ongoing discussion about which features will be backported.

kotyara12, 2023-09-06:
(Top) → Component config → ESP-MQTT Configurations → Number of queued events

This is it? How this option works (there is nothing about it in the help yet)

euripedesrocha, 2023-09-08:
@kotyara12 the outbox limitation is a runtime configuration. You can find the limit configuration in the mqtt client [config structure](https://github.com/espressif/esp-mqtt/blob/05b347643f6679cc60a50a9664b86a24ebf3ae10/include/mqtt_client.h#L363). The availability will depend on which IDF version you are using and the version of esp_mqtt.

kotyara12, 2023-09-08:
@euripedesrocha, many thanks!

closed 2023-10-19 by euripedesrocha