espressif/esp-idf #9091: MDNS example not working (mdns_query_a unsable to resolve hostname to IP)

ble0, 2022-06-03:
## Environment

- Development Kit:      ESP32S3-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-S3-WROOM-1
- IDF version (run ``git describe --tags`` to find it): v4.4.1
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System: Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply: USB

## Problem Description
mdns_query_a unable to resolve to IP address

### Expected Behavior
mdns_query_a should resolve to valid IP address

### Actual Behavior

### Steps to reproduce
1. Modify the `\examples\protocols\mdns\main.c` in the function `check_button`:
```
         // query_mdns_hosts_async("esp32-mdns");
        query_mdns_host("esp32-mdns");
```
2. Compile the example
3. Press GPIO_0 (BOOT) to trigger query 

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue
```
static void check_button(void)
{
    static bool old_level = true;
    bool new_level = gpio_get_level(EXAMPLE_BUTTON_GPIO);
    if (!new_level && old_level) {
        // query_mdns_hosts_async("esp32-mdns");
        query_mdns_host("esp32-mdns");
        query_mdns_service("_arduino", "_tcp");
        query_mdns_service("_http", "_tcp");
        query_mdns_service("_printer", "_tcp");
        query_mdns_service("_ipp", "_tcp");
        query_mdns_service("_afpovertcp", "_tcp");
        query_mdns_service("_smb", "_tcp");
        query_mdns_service("_ftp", "_tcp");
        query_mdns_service("_nfs", "_tcp");
    }
    old_level = new_level;
}
```

david-cermak, 2022-06-06:
Hi @ble0 

Are you trying to resolve *our own* host name? I'm sorry, but this will not work, the packets that bounce back are suppressed.

ble0, 2022-06-06:
I’m just curious. Then why some systems send mdns query their own hostname?

david-cermak, 2022-06-07:
> Then why some systems send mdns query their own hostname?

While querying their own names gets resolved correctly on some platforms (such as `avahi`), it's not supported in Espressif's mDNS library.
This was a design decision to keep the implementation simple and practical (and per mDNS spec: not to pollute the local network), and the way how the low level multicasts on WiFi are implemented.

> do you mean the "known suppression" mechanism?

No, this is unrelated to mDNS's definition of *known suppression*. This is how multacasts are treated on physical network interfaces (esp. WiFi)
And also in mdns sources:

https://github.com/espressif/esp-idf/blob/01d014c42def8d0c19e1ce55c07de6761e092ffa/components/mdns/mdns.c#L3307-L3322

PS: If you compile the mDNS library for a different platform (and remove the conditions above), you might be able to resolve the advertized service (you can check this [linux compilation](https://github.com/espressif/esp-idf/tree/master/components/mdns/host_test) example)

PPS: I'm sorry, this is not well documented that the mdns library doesn't resolve its own advertised names and services. Will update the docs.

ble0, 2022-06-07:
Hi David, thanks. Yep keep me posted when these doc updates are pushed.

closed 2022-06-07 by ble0