espressif/esp-idf #9097: [Nimble] [v4.4.1][ESP32-S3] ble_gattc_notify_custom returns ENOMEM (BLE_HS_ENOMEM)

chipweinberger, 2022-06-04:
ESP-IDF: v4.1.1 (ae0e3e2e2a, May 31 10:28:03)
ESP32-S3

ble_gattc_notify_custom() is returning ENOMEM, and there's no obvious reason.

1. I have plenty of free ram. 100+ KB of all RAM types.
2. It sometimes works. 
3. In my code I use a SPP over BLE architecture. For sending large amounts of text data, I call` split_notify_msg()`. 
5. AFAICT, It repros when`split_notify_msg()` msgLen > ~4000 bytes. This is bewildering to me. 
6. When It fails, it fails on the first call to ble_gattc_notify_custom(). 
7. MTU size is 512 bytes

My Code: 
```
bool split_notify_msg(uint16_t gapConnHandle, uint16_t attrHandle, const uint8_t* msg,  uint32_t msgLen)
{

    uint16_t mtu = get_mtu(gapConnHandle);

    uint8_t* pkt = (uint8_t*) calloc_safe2(MALLOC_CAP_SPIRAM, mtu, 1, HERE);

    // how many packets needed?
    uint32_t roundUp = msgLen % mtu == 0 ? 0 : 1;
    uint32_t numPackets = (msgLen / mtu) + roundUp;

    for (int i = 0; i < numPackets; i++){

    	// copy to pkt
        uint32_t offset = mtu * i;
        uint32_t pktLen = PDMIN(msgLen - offset, mtu);
        memcpy(pkt, msg + offset, pktLen);

    	// this is returning ENOMEM
    	struct os_mbuf* txom = ble_hs_mbuf_from_flat(msg, msgLen);

    	int rc = ble_gattc_notify_custom(gapConnHandle, attrHandle, txom);
    	if (rc != 0){
    		return false;
    	}
    }

    free_safe(pkt);
    return true
}

```

**I've tracked down that  ENOMEM is coming from ble_att_clt_tx_write_cmd in ble_att_clt.c**
```
int
ble_att_clt_tx_write_cmd(uint16_t conn_handle, uint16_t handle,
                         struct os_mbuf *txom)

    ... truncated....
    cmd = ble_att_cmd_get(BLE_ATT_OP_WRITE_CMD, sizeof(*cmd), &txom2);
    if (cmd == NULL) {
        os_mbuf_free_chain(txom);
        return BLE_HS_ENOMEM; // this is where it comes from
    }
```

But unfortunately **ble_att_cmd_get()** does not exist? It looks like it must be a proprietary blob from the linker?

chipweinberger, 2022-06-04:
**Adding a Sleep (or even a ESP_LOGI call) in between each call to ble_gattc_notify_custom() seems to fix the issue. Is there a recommended way to do rate limiting without Sleep?**

chipweinberger, 2022-06-04:
The bleprph_throughput example uses this approach:

```
                /* Check if the MBUFs are available */
                if (os_msys_num_free() >= MIN_REQUIRED_MBUF) {
                    om = ble_hs_mbuf_from_flat(payload, sizeof(payload));
                    if (om == NULL) {
                        /* Memory not available for mbuf */
                        ESP_LOGE(tag, "No MBUFs available from pool, retry..");
                        vTaskDelay(100 / portTICK_PERIOD_MS);
                        om = ble_hs_mbuf_from_flat(payload, sizeof(payload));
                        assert(om != NULL);
                    }

                    rc = ble_gattc_notify_custom(conn_handle, notify_handle, om);
                    if (rc != 0) {
                        ESP_LOGE(tag, "Error while sending notification; rc = %d", rc);
                        notify_count -= 1;
                        xSemaphoreGive(notify_sem);
                        /* Most probably error is because we ran out of mbufs (rc = 6),
                         * increase the mbuf count/size from menuconfig. Though
                         * inserting delay is not good solution let us keep it
                         * simple for time being so that the mbufs get freed up
                         * (?), of course assumption is we ran out of mbufs */
                        vTaskDelay(10 / portTICK_PERIOD_MS);
                    }
                } else {
                    ESP_LOGE(tag, "Not enough OS_MBUFs available; reduce notify count ");
                    xSemaphoreGive(notify_sem);
                    notify_count -= 1;
                }
```

chipweinberger, 2022-06-04:
This was very tricky to find. 

Googling ble_gattc_notify_custom() and BLE_HS_ENOMEM yeilded nothing useful I could find. 

These search terms should be added to the FAQ: https://espressif-docs.readthedocs-hosted.com/projects/espressif-esp-faq/en/latest/software-framework/ble-bt.html

ocESP, 2022-06-06:
Hello @chipweinberger, 

You can find `ble_att_cmd_get()` function [here](https://github.com/espressif/esp-nimble/blob/e6cc38908c58a5d3590bff96a2904fc70d746346/nimble/host/src/ble_att_cmd.c#L46). 

This issue occurs when we try to send too many packets before controller can process them. In this case, the packets get queued. As we queue more number of packets, we reach a point where we run out of memory. In this case, `BLE_HS_ENOMEM` is returned. 

There are two methods in which this issue can be solved - 
1. Using suggested default data length read/write commands at the time of connection. These commands, will suggest controller to increase the number of data packets that can be transmitted. Please find the PR for the same [here](https://github.com/apache/mynewt-nimble/pull/1252).
2. Introducing a delay (sleep) gives time for packet transmission, and the issue is not seen. Thus, adding delay works. If you want to avoid using delay, you can add check to retry the transmission in case of failure.

chipweinberger, 2022-06-06:
Thanks for the detailed reply.

I'll close this issue as your suggestions do work.

It would be useful for others to get this error code documented, and all the examples to use these suggestions.

closed 2022-06-06 by chipweinberger

Zyuangeorge, 2025-01-22:
> Adding a Sleep (or even a ESP_LOGI call) in between each call to ble_gattc_notify_custom() seems to fix the issue. **Is there a recommended way to rate limit without Sleep?**

May I ask what frequency are you calling the ble_gattc_notify_custom()? I am using 100Hz, seems like there are errors raised when I am using a self-developed BLE monitor as a client. But when I use the nRF mobile application, even with 100Hz, there is no BLE_HS_ENOMEM error raised. I also tested 25Hz and the number of errors was reduced. 

According to Google, the BLE connection interval is 7.5ms to 4s, it seems like 10ms is close to the minimum value thus the error increased.