espressif/esp-idf #9099: ESP32-WebServer gone from mDNS discovery after 45 minutes

ble0, 2022-06-04:
## Environment
- Development Kit:      ESP32S3-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-S3-WROOM-1
- IDF version (run ``git describe --tags`` to find it): v4.4.1
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System: Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply: USB

## Problem Description
ESP32-WebServer gone from mDNS discovery after 45 minutes 

### Expected Behavior
ESP32-WebServer should stay alive as long as it takes

### Actual Behavior
ESP32-WebServer is gone after 45 minutes

### Steps to reproduce
1. Go to the `\examples\protocols\mdns\
2. Compile the example
3. Wait 45 minutes

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue
Refer to example code

Other info:
- Host is **NOT** reachable via `ping esp32-mdns` for Windows or` ping esp32-dns.local` for MAC.

david-cermak, 2022-06-07:
Hi @ble0 

Is there anything else that's running on that `ESP32-WebServer`? Sending periodic queries or something?
I've just tested the default example on `v4.4.1` and the service is still available after the default TTL expiry (75 minutes).

ble0, 2022-06-07:
hi @david-cermak 
Nope, I'm just running the example. I also noticed that it didn't resend advertisements before or close to TTL expiry.

david-cermak, 2022-06-08:
@ble0 Noticed your other issue, where you've updated the example to periodically query it's own services. This modification might affect mDNS, too. Could you also please quickly check if the service disappears even if you don't send any queries?

> it didn't resend advertisements before or close to TTL expiry.

It should advertize only once (or every time we call `mdns_init()` or update network settings), as the resolvers are responsible to re-query all records that are about to expire (typically 1/2 of TTL). And this is what I think you're seeing on other devices, they don't periodically re-advertize, just reply to responders (with shared records, so other responders update their caches)

Anyway, thanks for sharing your configuration, will try to reproduce the problem on esp32-s3 and your setup.

ble0, 2022-06-08:
hi @david-cermak 
For this issue, I just use the example from espressif and only modified the `menuconfig` for wifi connection and IPv6. I tried to connect as little equipment as possible to my router to isolate the problem.
A slight modification to my set-up, Router - Mac laptop - iPhone 12 - ESP32, I did dns-sd for both my iPhone 12 and Mac and the ESP32 is gone again after a few minutes.

> Could you also please quickly check if the service disappears even if you don't send any queries?
 yes, even I don't send anything, but of course I need to browse the top level (.local)

ble0, 2022-06-09:
I also tried using `dig` to do a one-shot query but it could not be found anymore.

ble0, 2022-06-15:
hi @david-cermak 
I tried adding `esp_wifi_set_ps(WIFI_PS_NONE);` but it did not work.

MaxwellAlan, 2022-07-12:
Hi @ble0 
Can you try with the latest master? These days we have fix some S3 bugs.
Thanks.

ble0, 2022-07-12:
hi @MaxwellAlan 
What idf version should I be using then?

ble0, 2022-07-14:
Hi @MaxwellAlan 
I'm able to compile it now,
(1) I copied the project `examples/protocols/mdns` from esp-idf4.4.1
(2) Created `examples/protocols/idf_component.yml` and added
```
dependencies:
  espressif/mdns: "^1.0.3"
```
Is this correct way to compile it? 
Because I'm able to compile the project now.

ble0, 2022-07-17:
I'm using this commit `9b8c558e63d95b491355a22e69f247834e899b47` and it still happens, can you @MaxwellAlan check it on your side?

MaxwellAlan, 2022-07-18:
> I'm using this commit `9b8c558e63d95b491355a22e69f247834e899b47` and it still happens, can you @MaxwellAlan check it on your side?

Ok, I'll try to reproduce this issue.

xueyunfei998, 2022-07-26:
hi @ble0 
please retest with this commit :47b6a8c85a243e3f290967ebc0ecf925d0beddf0

after checkout this commit please update submodule:`git submodule update -f`

ble0, 2022-07-26:
hi @xueyunfei998 
I can still see the issue

xueyunfei998, 2022-07-27:
hi @ble0 

My system is like this, a windows system computer, an Asus router, and an esp32s3 board.

Connect the router with the computer ESP32s3, and then ping esp32-mdns on the computer.

I used the latest 4.4 branch and hung up to test the example of protocols/mdns, but the problem didn't reappear in two days.

- Please provide the log when something goes wrong.
- Please provide sniffer 

Assuming that the channel of the router is 1,

use a linux laptop,You can use it to sniff the channel 1.

ifconfig wifi. down
iwconfig wifi. mode monitor
ifconfig wifi. up
iwconfig wifi. channel 1
wireshark

ble0, 2022-07-27:
hi @xueyunfei998  
Let me try it again one more time.

xueyunfei998, 2022-07-29:
hi @ble0 
any update?

ble0, 2022-07-29:
ill give u an update by tomorrow

ble0, 2022-07-30:
hi @xueyunfei998 
We can close this issue. It is working now.

closed 2022-07-30 by ble0

thorrak, 2022-08-19:
Out of curiousity, what was ultimately done to resolve this issue? I'm having [similar issues](https://github.com/espressif/arduino-esp32/issues/7156) with Arduino-ESP32 where calling `WiFi.setSleep(false)` (which in turn calls `esp_wifi_set_ps(WIFI_PS_NONE);`) seems to solve it, but this means that I can't use Bluetooth + WiFi coexistence.

everslick, 2022-11-20:
I have the same problem as @thorrak and I also would like to know, what is the commit that was done for the fix. I guess it didn't land in Arduino yet. @ble0 want to chime in?