espressif/esp-idf #9102: Delay in wifi re-connection if wifi_config.sta.bssid_set is set to true

chenlijun99, 2022-06-05:
## Environment

- Development Kit:     ESP32-S2-Saola-1
- Module or chip used:   ESP32-S2-WROVER
- IDF version: v4.4.1
- Build System:  idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

After WiFi connection has been established, the following sequence of statements causes re-connection to be delayed for at least one second

```c
memcpy(wifi_config.sta.bssid, MY_TMP_MAC, 6);
wifi_config.sta.bssid_set = true; // set to true to reproduce the issue
ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &wifi_config));

ESP_ERROR_CHECK(esp_wifi_disconnect());

ESP_ERROR_CHECK(esp_wifi_connect());
```

while not setting `wifi_config.sta.bssid_set` to true, as in the following code,

```c
memcpy(wifi_config.sta.bssid, MY_TMP_MAC, 6);
wifi_config.sta.bssid_set = false; // set to true to reproduce the issue
ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &wifi_config));

ESP_ERROR_CHECK(esp_wifi_disconnect());

ESP_ERROR_CHECK(esp_wifi_connect());
```
re-connection happens immediately.

### Expected Behavior

Re-connection happens immediately even if `wifi_config.sta.bssid_set` is set to `true`.

### Actual Behavior

Re-connection is delayed by ~1 second if `wifi_config.sta.bssid_set` is set to `true`.

### Steps to reproduce

1. https://gist.github.com/chenlijun99/507973cc8d22bafd2e7e0b6270b49033 flash the gist
2. The desidered behaviour should be observed. I.e. after connection, we force disconnection using `esp_wifi_disconnect()` and then re-connect using `esp_wifi_connect()`. Re-connection happens quickly, without one second delay.
3. Set `wifi_config.sta.bssid_set` to true at line 84 of the gist and re-flash.
4. Observe the problem: re-connection using `esp_wifi_connect()` has a delay of one second.

### Code to reproduce this issue

https://gist.github.com/chenlijun99/507973cc8d22bafd2e7e0b6270b49033

In particular, look at line 84.

## Debug Logs

[bssid_set_false.txt](https://github.com/espressif/esp-idf/files/8839610/bssid_set_false.txt)
[bssid_set_true.txt](https://github.com/espressif/esp-idf/files/8839611/bssid_set_true.txt)

What's interesting is the `wifi:connect_op:` log.

bssid_set_false.txt
```
D (1981) wifi:connect_op: status=0, auth=8, cipher=3 
D (3091) wifi:auth mode is not none
```

bssid_set_true.txt
```
D (1951) wifi:connect_op: status=0, auth=8, cipher=3 
D (3061) wifi:auth mode is not none
```

On both cases, after the first `wifi:connect_op:` we see 1 second of delay. But only in the case of bssid_set_true.txt, the second `wifi:connect_op:` is followed by 1 second delay.

bssid_set_false.txt
```
D (4331) wifi:connect_op: status=0, auth=8, cipher=3 
D (4331) wifi:auth mode is not none
```

bssid_set_true.txt
```
D (4271) wifi:connect_op: status=0, auth=8, cipher=3 
D (5381) wifi:auth mode is not none
```

So it seems that by setting `wifi_config.sta.bssid_set` to true, something in the WiFi stack gets resetted.

chenlijun99, 2022-06-06:
Further information: apparently alternating between `wifi_config.sta.bssid_set` set to true and set to false at each re-connection causes the 1 second delay at each re-connection. OTOH, if `wifi_config.sta.bssid_set` is set to true from the beginning, then the delay is not present on subsequent reconnections.
So it seems that by changing the value of `wifi_config.sta.bssid_set`, something in the WiFi stack gets resetted.

chenlijun99, 2022-06-27:
I tested also on master (c2ccc383dae2a47c2c2dc8c7ad78175a3fd11361) and the problem persists.

mhdong, 2024-06-13:
Hi @chenlijun99 
When re-connection with wifi_config.sta.bssid_set = false, we do not change any config. While call function wpa_set_passphrase, we do not calculate psk.
When re-connection with wifi_config.sta.bssid_set = true, we change config. While call function wpa_set_passphrase, we will re calculate psk.

please reference:
wpa_set_passphrase(char * passphrase, u8 *ssid, size_t ssid_len)
{
    struct wifi_ssid *sta_ssid = esp_wifi_sta_get_prof_ssid_internal();
    struct wpa_sm *sm = &gWpaSm;

    if (passphrase == NULL) return;

    /*
     *  Here only handle passphrase string.  Need extra step to handle 32B, 64Hex raw
     *    PMK.
     */
    if (sm->key_mgmt == WPA_KEY_MGMT_SAE)
        return;

    /* This is really SLOW, so just re cacl while reset param */
    if (esp_wifi_sta_get_reset_param_internal() != 0) {
        // check it's psk
        if (strlen((char *)esp_wifi_sta_get_prof_password_internal()) == 64) {
            if (hexstr2bin((char *)esp_wifi_sta_get_prof_password_internal(),
                           esp_wifi_sta_get_ap_info_prof_pmk_internal(), PMK_LEN) != 0)
                return;
        } else {
            pbkdf2_sha1((char *)esp_wifi_sta_get_prof_password_internal(), sta_ssid->ssid, (size_t)sta_ssid->len,
                        4096, esp_wifi_sta_get_ap_info_prof_pmk_internal(), PMK_LEN);
        }
        esp_wifi_sta_update_ap_info_internal();
        esp_wifi_sta_set_reset_param_internal(0);
    }

    if (sm->key_mgmt == WPA_KEY_MGMT_IEEE8021X) {
    /* TODO nothing */
    } else {
        memcpy(sm->pmk, esp_wifi_sta_get_ap_info_prof_pmk_internal(), PMK_LEN);
        sm->pmk_len = PMK_LEN;
    }
}

chenlijun99, 2024-06-13:
Hi @mhdong,
thanks for the information. I'm not using this anymore, but it's still nice to hear an explanation for this issue. I think it would be nice to add this caveat to the documentation of `esp_wifi_set_config`.

Anyway, feel free to close this issue.

closed 2024-06-14 by Sherry616