espressif/esp-idf #9103: Allow to connect to WiFi AP without scanning.

chenlijun99, 2022-06-05:
**Is your feature request related to a problem? Please describe.**

We would like to connect to a WiFi AP without the "implicit" scanning performed by `esp_wifi_connect()`. We would want to rely on custom vendor IE in the beacons to determine the best AP to connect to. The `wifi_sort_method_t` provided by `esp_wifi_connect()` is too limiting.

**Describe the solution you'd like**

A `esp_wifi_connect_to_ap(uint8_t *bssid)` function that directly tries to connect to the specified AP, without doing any scanning.

chenlijun99, 2022-07-17:
I'm sure such a function already exists in the existing wifi library, maybe it can be relatively easy to just expose it?

mhdong, 2022-07-18:
hi @chenlijun99 We have done similar functions for customers. We will merge this function into the master branch as soon as possible.

chenlijun99, 2022-07-18:
Hi @mhdong. Thanks for the reply! Can't wait to see it merged! :smile:

AxelLin, 2023-05-02:
> hi @chenlijun99 We have done similar functions for customers. We will merge this function into the master branch as soon as possible.

@mhdong How is the status for merging this function now?

jack0c, 2023-07-25:
Sorry for the delay. This function break some old functions, not merged yet. If we still need this function, we need to clear all issues found in the internal test. We will do it recently.

chenlijun99, 2023-07-25:
Personally I don't have urgent need for this function anymore. Still, I can see how it may be useful to many projects.
Thanks for the update and thanks for your work!

AndyMNSenso, 2023-09-22:
I would like to add a bump to this feature request.

In our application we have the need to switch APs as quickly as possible. We are currently using ESP-IDF 4.4.2

All our APs are working on the same single channel (channel 1). 
We scan the APs prior to connecting to an AP. 
This scan only checks channel 1 as configured:

```
V (63919) wifi:set_scan_id=0
V (63922) esp_adapter: thread sem create: sem=0x3ffde6fc
V (63927) esp_adapter: thread sem get: sem=0x3ffde6fc
D (63932) wifi:Start wifi scan
D (63935) wifi:ssid=broadcastyyy, channel=1, hidden=1, type=active, active=<0,0>, passive=0
D (63943) wifi:filter: set rx policy=3
D (63947) wifi:clear scan ap list
D (63950) wifi:start scan: type=0x8400, priority=3, cb=0x0, arg=0x3f801278, ss_state=0x1, time=60329655, index=0
V (63960) wifi:inter_channel_timeout: arg=0x0, ss_state=0x1
D (63965) wifi:perform scan: ss_state=0x9, chan<1,0>, dur<0,120>
V (63971) wifi:change: chan<1,0>, dur<0,120>
V (63975) wifi:enter start op, arg=0x3f80108d
V (63979) wifi:scan operation: state=0x3, chan<1,0>, arg=0x3f80108d, status=0
V (63986) wifi:scan specific ssid=broadcastyyy
V (63990) wifi:start max timer
V (63993) wifi:get_scan_id=0
V (63999) wifi:find matched ssid state
V (63999) wifi:set terminate flag
V (64002) wifi:scan histroy table is not full yet, add
V (64006) wifi:add ssid=broadcastyyy, chan=1 to scan history
V (64012) wifi:find matched ssid state
V (64015) wifi:set terminate flag
V (64018) wifi:update scan history ssid=broadcastyyy, channel=1
V (64095) wifi:get_scan_id=0
D (64113) wifi:scan end: arg=0x0, status=0, ss_state=0x7
V (64113) wifi:hint scan, stop scan
V (64113) wifi:back home chan=<1,0>, current chan=<1,0>
D (64115) wifi:filter: set rx policy=4
D (64119) wifi:first chan=1
V (64121) wifi:scan_done: arg=0x0, status=0, cur_time=60500938, scan_id=128, scan state=0
D (64129) wifi:Send scan done event: status=0, apnum=2
D (64134) event: running post WIFI_EVENT:1 with handler 0x400e3314 and context 0x3ffdd8a8 on loop 0x3ffdc3b0
```

But when we connect to an AP that was scanned, the esp_wifi_connect seems to scan all channels:

```
V (64228) esp_adapter: thread sem get: sem=0x3ffde6fc
D (64233) wifi:clear blacklist
D (64236) nvs: nvs_get_str_or_blob sta.ssid
D (64241) nvs: nvs_get sta.pmf_e 1
V (64244) esp_adapter: thread sem get: sem=0x3ffde6fc
D (64248) wifi:Start wifi connect
D (64251) wifi:connect status 0 -> 0
D (64255) wifi:connect chan=0
D (64257) wifi:first chan=1
D (64260) wifi:connect status 0 -> 1
D (64263) wifi:filter: set rx policy=3
D (64267) wifi:clear scan ap list
D (64270) wifi:start scan: type=0x50f, priority=2, cb=0x4012171c, arg=0x0, ss_state=0x1, time=60649614, index=0
0x4012171c: cnx_start_handoff_cb at ??:?

V (64279) wifi:inter_channel_timeout: arg=0x0, ss_state=0x1
D (64285) wifi:perform scan: ss_state=0x9, chan<1,0>, dur<0,120>
V (64291) wifi:change: chan<1,0>, dur<0,120>
V (64294) wifi:enter start op, arg=0x3f80108d
V (64299) wifi:scan operation: state=0x3, chan<1,0>, arg=0x3f80108d, status=0
V (64305) wifi:scan specific ssid=broadcastyyy
V (64310) wifi:start max timer
D (64318) wifi:rsn valid: gcipher=1 ucipher=1 akm=5

[... 185 lines cut ...]
I (67538) wifi:state: assoc -> run (10)
D (67539) wifi:start 10s connect timer for 4 way handshake
V (67542) wifi:wpa_psk start
V (67546) wifi:wpa_psk handle succeed
V (67562) wifi:wpa_psk start
V (67564) wifi:wpa_psk handle succeed
I (67566) wifi:connected with broadcastyyy, aid = 1, channel 1, BW20, bssid = e8:31:cd:e7:60:0d
I (67567) wifi:security: WPA2-PSK, phy: bg, rssi: -36
D (67571) wifi:remove all except e8:31:cd:e7:60:0d from rc list
D (67576) wifi:clear blacklist
D (67579) nvs: nvs_set sta.chan 1 1
D (67583) nvs: nvs_set_blob sta.apinfo 700
D (67590) wifi:filter: set rx policy=7
I (67590) wifi:pm start, type: 0

D (67594) wifi:Send sta connected event
D (67597) wifi:connect status 1 -> 5
D (67600) wifi:obss scan is disabled
D (67604) wifi:start obss scan: obss scan is stopped
D (67609) event: running post WIFI_EVENT:4 with handler 0x400ecddc and context 0x3ffdd5a4 on loop 0x3ffdc3b0
```

All STA and AP configs only have a single channel set, country configuration is also custom with only a single channel set, but still the esp_wifi_connect function scans all channels.

This is a big time waste and we would prefer a function to connect to a specific MAC or a fix to esp_wifi_connect so that it obeys our channel specifications and does not look at any other channel. If the MAC is not found it should simply return with an error.

AxelLin, 2024-10-13:
> Sorry for the delay. This function break some old functions, not merged yet. If we still need this function, we need to clear all issues found in the internal test. **We will do it recently.**

@jack0c 
It has been 1 year since your last reply, how is the status?