espressif/esp-idf #9104: Memory Allocation in C++ with SPIRAM

f-hoepfinger-hr-agrartechnik, 2022-06-05:
Hello, 

for C i can get Memory from SPIRAM via heap_caps_malloc.

in C++ i have new, but where does "new" get the Memory of ? 

is there a way to say "please place this Object in SPIRAM" ? 

Thanks, 
Franz

atanisoft, 2022-06-05:
You can use https://github.com/atanisoft/esp32usb/blob/master/include/psram_allocator.h (or a variant of this) to allocate in PSRAM. Example of usage:
```
std::vector<fat_long_filename_t,
                PSRAMAllocator<fat_long_filename_t>> lfn_parts;
```

moefear85, 2022-06-07:
either you have to use functions from esp-idf to do the allocation (and they are all C functions), or you have to setup sdkconfig to always use SPIRAM and prefer it over normal RAM for all heap allocations. Otherwise you have to create your own C++ wrapper if the goal is to have an elegant pure C++ way of doing things. You can't specify these things with the "new" keyword.

atanisoft, 2022-06-07:
> You can't specify these things with the "new" keyword.

That isn't true, you can use a custom allocator (such as what I linked to above) or use the "new with placement" approach.

moefear85, 2022-06-07:
yes it is. you're code is not standard c++. its a wrapper you wrote on your own. anyone can do that. that's not the question.

moefear85, 2022-06-07:
my mistake. i just learned something.

0xjakob, 2022-07-29:
@f-hoepfinger-hr-agrartechnik Did @atanisoft's suggestion help you solving this issue?

0xjakob, 2022-11-30:
@f-hoepfinger-hr-agrartechnik Please use a custom allocator as suggested. Alternatively, you can overload the `operator new` for specific types or overload the global `operator new`.

Since there hasn't been any activity, we are closing this issue. Feel free to re-open it if the suggested approaches didn't help.

closed 2022-11-30 by 0xjakob

lanistor, 2023-10-31:
Do like this:
```cpp
void* operator new(size_t size, void* ptr = NULL) {
  if (ptr) {
    return ptr;
  }
  return heap_caps_malloc(size, MALLOC_CAP_SPIRAM);
}
```

diplfranzhoepfinger, 2024-09-26:
wouldnt it be nice to have a Standard `heap_caps_new` ? 

```
void* operator heap_caps_new(size_t size, void* ptr = NULL) {
  if (ptr) {
    return ptr;
  }
  return heap_caps_malloc(size, MALLOC_CAP_SPIRAM);
}
```

diplfranzhoepfinger, 2024-09-26:
> Do like this:
> 
> ```c++
> void* operator new(size_t size, void* ptr = NULL) {
>   if (ptr) {
>     return ptr;
>   }
>   return heap_caps_malloc(size, MALLOC_CAP_SPIRAM);
> }
> ```

this brings a Error: 

```
error: redefinition of 'void* operator new(size_t, void*)'
```

pvginkel, 2024-10-12:
Working version:

```cxx
void* operator new(size_t size) {
    void* ptr = heap_caps_malloc(size, MALLOC_CAP_SPIRAM);
    if (!ptr) {
        abort();
    }
    return ptr;
}

void* operator new[](size_t size) {
    void* ptr = heap_caps_malloc(size, MALLOC_CAP_SPIRAM);
    if (!ptr) {
        abort();
    }
    return ptr;
}

void operator delete(void* ptr) noexcept { free(ptr); }

void operator delete(void* ptr, std::size_t) noexcept { free(ptr); }

void operator delete[](void* ptr) noexcept { free(ptr); }

void operator delete[](void* ptr, std::size_t) noexcept { free(ptr); }
```