espressif/esp-idf #9106: MQTT seems to be blocking API

diplfranzhoepfinger, 2022-06-05:
## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

diplfranzhoepfinger, 2022-06-05:
\### Actual Behavior

we have duration\_min/max 1000/1000 microseconds before MQTT run

we have duration\_min/max 34/364679 microseconds while connecting, this might be worse if using ESP-Modem or slower Internet connections. This Means that the Task-Scheduler is totally blocked at least once for about 400ms. in Best conditions.

we have duration\_min/max 649/1351 microseconds after the run of the sample, which is acceptable.

\### Steps to reproduce

run the Project

\### Code to reproduce this issue

https://github.com/DiplomInformatikerFranzHoepfinger/mqtt_and_watchdog

\## Debug Logs

\`\`\`  
Executing action: monitor  
Running idf\_monitor in directory /home/franz/git/franz/mqtt\_and\_watchdog/mqtt\_ssl  
Executing "/home/franz/.espressif/python\_env/idf4.4\_py3.8\_env/bin/python /home/franz/esp-idf-v4.4/tools/idf\_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 3 --print\_filter  /home/franz/git/franz/mqtt\_and\_watchdog/mqtt\_ssl/build/mqtt\_ssl.elf -m '/home/franz/.espressif/python\_env/idf4.4\_py3.8\_env/bin/python' '/home/franz/esp-idf-v4.4/tools/idf.py'"...  
\--- idf\_monitor on /dev/ttyUSB0 115200 ---  
\--- Quit: Ctrl+\] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---  
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON\_RESET),boot:0x13 (SPI\_FAST\_FLASH\_BOOT)  
configsip: 0, SPIWP:0xee  
clk\_drv:0x00,q\_drv:0x00,d\_drv:0x00,cs0\_drv:0x00,hd\_drv:0x00,wp\_drv:0x00  
mode:DIO, clock div:1  
load:0x3fff0030,len:6932  
load:0x40078000,len:15516  
load:0x40080400,len:4284  
0x40080400: \_init at ??:?

entry 0x40080680  
I (27) boot: ESP-IDF v4.4-dirty 2nd stage bootloader  
I (27) boot: compile time 17:31:46  
I (27) boot: chip revision: 3  
I (30) qio\_mode: Enabling default flash chip QIO  
I (36) boot.esp32: SPI Speed      : 80MHz  
I (40) boot.esp32: SPI Mode       : QIO  
I (45) boot.esp32: SPI Flash Size : 8MB  
I (50) boot: Enabling RNG early entropy source...  
I (55) boot: Partition Table:  
I (58) boot: ## Label            Usage          Type ST Offset   Length  
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00006000  
I (73) boot:  1 phy\_init         RF data          01 01 0000f000 00001000  
I (81) boot:  2 factory          factory app      00 00 00010000 00177000  
I (88) boot: End of partition table  
I (92) esp\_image: segment 0: paddr=00010020 vaddr=3f400020 size=1b2a4h (111268) map  
I (132) esp\_image: segment 1: paddr=0002b2cc vaddr=3ffb0000 size=02f2ch ( 12076) load  
I (135) esp\_image: segment 2: paddr=0002e200 vaddr=40080000 size=01e18h (  7704) load  
I (140) esp\_image: segment 3: paddr=00030020 vaddr=400d0020 size=8b864h (571492) map  
I (303) esp\_image: segment 4: paddr=000bb88c vaddr=40081e18 size=118e4h ( 71908) load  
I (327) esp\_image: segment 5: paddr=000cd178 vaddr=50000000 size=00010h (    16) load  
I (336) boot: Loaded app from partition at offset 0x10000  
I (337) boot: Disabling RNG early entropy source...  
I (349) cpu\_start: Pro cpu up.  
I (349) cpu\_start: Single core mode  
I (357) cpu\_start: Pro cpu start user code  
I (357) cpu\_start: cpu freq: 240000000  
I (357) cpu\_start: Application information:  
I (362) cpu\_start: Project name:     mqtt\_ssl  
I (367) cpu\_start: App version:      v0.1-19-g06a2a91-dirty  
I (373) cpu\_start: Compile time:     Jun  5 2022 17:43:18  
I (379) cpu\_start: ELF file SHA256:  52871ca846916717...  
I (385) cpu\_start: ESP-IDF:          v4.4-dirty  
I (391) heap\_init: Initializing. RAM available for dynamic allocation:  
I (398) heap\_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM  
[... 66 lines cut ...]
I (25735) MQTTS\_EXAMPLE: sent subscribe successful, msg\_id=48076  
I (25740) MQTTS\_EXAMPLE: sent unsubscribe successful, msg\_id=5989  
I (25936) MQTTS\_EXAMPLE: MQTT\_EVENT\_SUBSCRIBED, msg\_id=37197  
I (25938) MQTTS\_EXAMPLE: sent publish successful, msg\_id=0  
I (25990) example: duration\_min/max 34/364679 microseconds  
I (26142) MQTTS\_EXAMPLE: MQTT\_EVENT\_SUBSCRIBED, msg\_id=48076  
I (26143) MQTTS\_EXAMPLE: sent publish successful, msg\_id=0  
I (26346) MQTTS\_EXAMPLE: MQTT\_EVENT\_UNSUBSCRIBED, msg\_id=5989  
I (26490) MQTTS\_EXAMPLE: MQTT\_EVENT\_DATA  
TOPIC=/topic/qos0  
DATA=data  
I (26755) MQTTS\_EXAMPLE: MQTT\_EVENT\_DATA  
TOPIC=/topic/qos0  
DATA=data  
I (30990) example: duration\_min/max 31/1970 microseconds  
I (35990) example: duration\_min/max 576/1424 microseconds  
I (40990) example: duration\_min/max 623/1377 microseconds  
I (45990) example: duration\_min/max 567/1433 microseconds  
I (50990) example: duration\_min/max 726/1274 microseconds  
I (55990) example: duration\_min/max 649/1351 microseconds  
\`\`\`

\## Other items if possible

\- \[X \] sdkconfig file see github linked above.

david-cermak, 2022-06-07:
@diplfranzhoepfinger Yes most of the mqtt API is blocking, but for publishing you can use `esp_mqtt_client_enqueue()`

https://github.com/espressif/esp-mqtt/blob/684843a3096fea77346fd7f6994dcf984d65308b/include/mqtt_client.h#L354-L373

Which would just enqueue the message to be sent later in the mqtt thread context.

About connection: It's also happening in the mqtt task, so shouldn't block user threads. The default timeout is 10s, but it's configurable:

https://github.com/espressif/esp-mqtt/blob/master/include/mqtt_client.h#L221

franz-ms-muc, 2022-06-27:
Hello, 

any news about this ? 
Vasil from [www.rilabs.io](http://www.rilabs.io/) need non-Blocking API.

david-cermak, 2022-07-04:
Hi all,

If I understand correctly, there are two problems from the threading point of view:
1) mqtt task taking too much computing resources, that you're getting hit by the task WDT
2) other APIs, e.g. publish where you're not only waiting for the actual data to be sent out to the network, but also wait for the confirmation.

While, I cannot help much with the `1)`, as the TLS handshake is happening in the context of mqtt task (Here I'd suggest just adjusting the task watchdog timeout based on your needs, usually key/cert size), the second one could be easily handled by our [mqtt cxx wrapper](https://github.com/espressif/esp-idf/tree/master/examples/cxx/experimental/esp_mqtt_cxx), that helps creating your own actions after an asynchronous event happened, for example:

https://github.com/espressif/esp-idf/blob/c2ccc383dae2a47c2c2dc8c7ad78175a3fd11361/examples/cxx/experimental/esp_mqtt_cxx/ssl/main/mqtt_ssl_example.cpp#L26-L44

closed 2022-09-01 by diplfranzhoepfinger