espressif/esp-idf #9107: ESP_EARLY_LOGX calls can't be filtered even after startup

DavidJRobertson, 2022-06-05:
Calls to `ESP_EARLY_LOGX` macros cannot be filtered using `esp_log_level_set(...)`, even after the processor is up and running and normal `ESP_LOGX` calls are working.

This results in extra log spam that cannot be disabled. For example, from the efuse component.

I would suggest checking an 'initialised' flag and directing the call to either:
1. the direct `esp_rom_printf` call if not yet initialised
2. the `ESP_LOG_LEVEL` macro if initialised (which allows filtering to work)

negativekelvin, 2022-06-05:
It should be filtered based on default log level `esp_log_level_set("*", ESP_LOG_INFO);`
https://github.com/espressif/esp-idf/blob/01d014c42def8d0c19e1ce55c07de6761e092ffa/components/log/include/esp_log.h#L323

You have an interesting suggestion however I think ESP_EARLY_LOGX and ESP_DRAM_LOGX would first have to be fully distinguished in the codebase.

KonstantinKondrashov, 2024-08-15:
The log is printed by ESP_EARLY_LOGX if the `(LOG_LOCAL_LEVEL >= (log_level) && esp_log_get_default_level() >= (log_level))` allows it. esp_log_get_default_level() is directed by esp_log_level_set().

If you do not want to see startup logs you can set CONFIG_LOG_MAXIMUM_LEVEL=INFO and CONFIG_LOG_DEFAULT_LEVEL=WARN
```
    Default log verbosity (Warning)  --->
    Maximum log verbosity (Info)  --->
```
In the code raise the default log level till INFO using `esp_log_level_set()`, then info logs will go.

I close this issue because it was answered above. Thanks.

closed 2024-08-15 by KonstantinKondrashov