espressif/esp-idf #9108: wifi:bcn_timout,ap_probe_send_start when using esp_http_client to make post requests

listout, 2022-06-06:
## Environment

- Development Kit: N/A
- Kit version (for WroverKit/PicoKit/DevKitC): N/A
- Module or chip used: ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): 4.4.1
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

I'm writing polled values from IMU into flash, fatfs with wear leveling is the file system. After that when I try to upload the files to a cloud server, I get `wifi:bcn_timout,ap_probe_send_start` after uploading 2 - 3 files.

I've about 5 files of 1 MB. there is also another warning `TRANSPORT_BASE: Poll timeout or error, errno=Connection already in progress, fd=54, timeout_ms=5000`

P.S. My ESP32 is very close to AP so no question of losing connection

### Expected Behavior

Don't disconnect from WiFi AP and continue uploading.

### Actual Behavior

Disconnects from WiFi AP and stops uploading

### Steps to reproduce

1. Write some files to flash storage with fatfs and wearleveling
2. Try to make http requests with those files

### Code to reproduce this issue

The code for uploading
```cpp
#define HTTP_BUFFER_SIZE (1024 * 3)

static void http_rest_with_url()
{
	ESP_LOGW(TAG, "%d", heap_caps_get_free_size(MALLOC_CAP_8BIT));
	/*char local_response_buffer[MAX_HTTP_OUTPUT_BUFFER] = { 0 };*/
	/**
	 * NOTE: All the configuration parameters for http_client must be spefied either in URL or as host and path parameters.
	 * If host and path parameters are not set, query parameter will be ignored. In such cases,
	 * query parameter should be specified in URL.
	 *
	 * If URL as well as host and path parameters are specified, values of host and path will be considered.
	 */
	esp_http_client_config_t config = {
		.url = "https:/redacted/remote.php/dav/files/redacted/",
		.auth_type = HTTP_AUTH_TYPE_BASIC,
		.username = "redacted",
		.password =
			"redacted"
	};
	esp_http_client_handle_t client = esp_http_client_init(&config);

	//Get the derived MAC address for each network interface
	uint8_t mac_addr[6] = { 0 };
	esp_read_mac(mac_addr, ESP_MAC_WIFI_STA);

	char buffer[HTTP_BUFFER_SIZE];
	unsigned fileProgress = 0;
	int length = 1;

	DIR *d;
	struct dirent *dir;
	char url[2048];
	d = opendir("/d0/");
	if (d) {
		while ((dir = readdir(d)) != NULL) {
			ESP_LOGI(TAG, "Uploading %s", dir->d_name);
			snprintf(
				url, sizeof url,
				"https://redacted/remote.php/dav/files/redeacted/data_server_%s",
				dir->d_name);
			printf("%s\n", url);

			// POST
			length = 1;
			esp_http_client_set_url(client, url);
			esp_http_client_set_method(client, HTTP_METHOD_PUT);
			esp_http_client_set_header(client, "Content-Type", "text/plain");
			char test[255] = { "/d0/" };
			strcat(test, dir->d_name);
			ESP_LOGI(TAG, "File name: %s", test);
			FILE *f = fopen(test, "rb");
			if (f == NULL) {
				fclose(f);
				ESP_LOGE(TAG, "Could not open file for uploading");
				return;
			}
			struct stat st;
			stat(test, &st);
			esp_http_client_open(client, st.st_size);
			time_t fileTransferStart = 0;
			while (length) {
				length = fread(buffer, sizeof(char), HTTP_BUFFER_SIZE, f);
				if (length) {
					int wret = esp_http_client_write(client, buffer, length);
					ESP_LOGI(TAG, "write file:\t%d/%d", wret, length);
					if (wret < 0)
						return;

					fileProgress += wret;
					time_t now;
					time(&now);
				}
				vTaskDelay(100 / portTICK_PERIOD_MS);
			}
			fclose(f);
			ESP_LOGI(TAG, "HTTP POST Status = %d, content_length = %d",
				 esp_http_client_get_status_code(client),
				 esp_http_client_get_content_length(client));
			vTaskDelay(100 / portTICK_PERIOD_MS);
		}
		closedir(d);
	}

	esp_http_client_cleanup(client);
}
```

## Debug Logs

```
W (77571) TRANSPORT_BASE: Poll timeout or error, errno=Connection already in progress, fd=54, timeout_ms=5000
I (77571) file upload: write file:      0/3072
I (78467) wifi:bcn_timout,ap_probe_send_start
I (80969) wifi:ap_probe_send over, resett wifi status to disassoc
I (80970) wifi:state: run -> init (c800)
I (80971) wifi:pm stop, total sleep time: 37875396 us / 65686074 us
W (80975) wifi:<ba-del>idx
W (80977) wifi:<ba-del>idx
I (80980) wifi:new:<8,0>, old:<8,0>, ap:<255,255>, sta:<8,0>, prof:1
I (80986) prov: Disconnected. Connecting to the AP again...
E (80995) TRANSPORT_BASE: poll_write select error 113, errno = Software caused connection abort, fd = 54
W (81002) TRANSPORT_BASE: Poll timeout or error, errno=Connection already in progress, fd=54, timeout_ms=5000
I (81005) wifi:I (81012) file upload: write file:       -1/3072
new:<8,0>, old:<8,0>, ap:<255,255>, sta:<8,0>, prof:1I (81018) file upload: Finished uploading

I (81029) wifi:state: init -> auth (b0)
Committing updates in NVS ... I (81034) wifi:state: auth -> assoc (0)
I (81043) wifi:state: assoc -> run (10)
W (81069) wifi:<ba-add>idx:0 (ifx:0, 62:a4:b7:1a:9b:01), tid:5, ssn:0, winSize:64
I (81163) wifi:connected with test1234, aid = 3, channel 8, BW20, bssid = 62:a4:b7:1a:9b:01
I (81164) wifi:security: WPA2-PSK, phy: bgn, rssi: -76
I (81169) wifi:pm start, type: 1

I (81170) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (82032) wifi:state: run -> init (0)
I (82033) wifi:pm stop, total sleep time: 433762 us / 863983 us

W (82034) wifi:<ba-del>idx
I (82034) wifi:new:<8,0>, old:<8,0>, ap:<255,255>, sta:<8,0>, prof:1
I (82040) prov: Disconnected. Connecting to the AP again...
I (82070) wifi:flush txq
I (82070) wifi:stop sw txq

```

I've seen various issues about this and forum post, none of them ever provide a solution.

listout, 2022-06-06:
I was able to get somewhat of a better result with adding 1 second delay between each `esp_http_client_write`. But would like to do the uploading as fast as possible.

HiFiPhile, 2022-06-06:
Out of curiosity do you use modem sleep ?

listout, 2022-06-07:
@HiFiPhile Nope

MaxwellAlan, 2022-06-08:
@listout 

- From your log, maybe you need to disable modem sleep by calling: esp_wifi_set_ps(0);
- Also, when you still meet bcn_timout, you can calling: esp_wifi_statis_dump(0xffff); to dump the wifi internal statis.

Thanks.

listout, 2022-06-08:
@MaxwellAlan Are you sure this is not a memory issue? Because once I gave increased the delay between each read and write to flash I didn't face the issue. Going through forum posts on similar issue, all of them seems to suggests towards memory being full.

Thanks you for the recommendations BTW, I'll test and report back. Please keep the issue open for few more days.

AxelLin, 2022-06-08:
> @listout
> 
> * From your log, maybe you need to disable modem sleep by calling: esp_wifi_set_ps(0);

Why? I thought WIFI_PS_MIN_MODEM should work.

MaxwellAlan, 2022-06-09:
> > @listout
> > 
> > * From your log, maybe you need to disable modem sleep by calling: esp_wifi_set_ps(0);
> 
> Why? I thought WIFI_PS_MIN_MODEM should work.

Just for debuging whether related with sleep.

listout, 2022-06-09:
@MaxwellAlan using `esp_wifi_set_ps(0)` seems to fix the issue (at least for now). I tested it with the code I initially posted (not the one with 1 sec delay).

Thanks.

AxelLin, 2022-06-09:
@listout 
I suggest keeping this issue open until it's fixed with WIFI_PS_MIN_MODEM.

listout, 2022-06-11:
@AxelLin, sure. This will also be good for me, I'll do further tests on my end,

listout, 2022-06-23:
Any update on this?

HiFiPhile, 2022-06-23:
What's the value of `CONFIG_ESP_PHY_MAC_BB_PD` in your sdkconfig ?
I'm wondering it's the same issue as #9059

listout, 2022-06-24:
@HiFiPhile `CONFIG_ESP_PHY_MAC_BB_PD` is not defined in my sdkconfig, or at least not my default.

MaxwellAlan, 2022-07-31:
Hi @listout 
Can you upload a simple example for we to debug this issue?

listout, 2022-07-31:
@MaxwellAlan sure. I'll try to create a separate repo and update the comment

Janrupf, 2022-10-07:
I seem to be running into this issue - reproduction case is more or less the same, except I'm running a big download task. When looping while downloading a big file, the modem loses the beacon and has to reconnect (which it does successfully).

Note that I also have bluetooth enabled, which means I can't disable wifi power saving.

AxelLin, 2022-10-07:
@Janrupf 

If you are using master tree, please call esp_wifi_statis_dump(0xffff); when got WIFI_EVENT_STA_BEACON_TIMEOUT event, and share the output.
If you are using other branches, there is no WIFI_EVENT_STA_BEACON_TIMEOUT event, you probably need to add esp_wifi_statis_dump(0xffff) call regularly until you hit the issue. Then share the output after bcn_timout happened.

BTW, @MaxwellAlan  The "bcn_timout" looks like a typo, maybe it should be "bcn_timeout"?

AxelLin, 2022-10-07:
Below is the esp_wifi_statis_dump(0xffff) output when wifi:bcn_timout,ap_probe_send_start happened.
Test with v5.1-dev-1006-g02605f1a31 on esp32c3, and I have BT enabled.
Is this normal?
I (1305140) wifi:bcn_timout,ap_probe_send_start
I (1305142) wifi: BUFFER:
I (1305142) wifi: t  max alloc total_alloc total_free fail_fc  fail_oom
I (1305148) wifi: 1  32  0     136         136        0        0       
I (1305155) wifi: 2  32  1     14          13         0        0       
I (1305161) wifi: 3  32  0     681         681        0        0       
I (1305167) wifi: 4  32  0     0           0          0        0       
I (1305174) wifi: 5  0   0     4           4          0        0       
I (1305180) wifi: 6  0   0     0           0          0        0       
I (1305187) wifi: 7  32  0     1125        1125       0        0       
I (1305193) wifi: 8  32  0     2137        2137       0        0       
I (1305199) wifi: 9  0   0     0           0          0        0       
I (1305205) wifi: 10 2   2     2           2          0        0       
I (1305212) wifi:
I (1305213) wifi: LMAC TX:
I (1305216) wifi: tx_all   lrc src age to 
I (1305219) wifi: 741      0   0   0   0  
I (1305223) wifi: 0        0   0   0   0  
I (1305227) wifi: 91       0   0   0   0  
I (1305231) wifi: 0        0   0   0   0  
I (1305235) wifi: 0        0   0   0   0  
I (1305239) wifi:
I (1305240) wifi: LMAC RX:
I (1305243) wifi: ampdu    mdpu     mgmt     ctrl     frag
I (1305248) wifi: 0        1125     2159     0        0   
I (1305253) wifi:
I (1305255) wifi: HW RX:
I (1305257) wifi: rx_suc   rx_end   full#1   full#2  
I (1305262) wifi: cd4      14d4     0        0       
I (1305267) wifi:
I (1305268) wifi: Hardware:
I (1305271) wifi: rxbufbk  rxcckerr ofdmerr  rx_agc   rxsf     rx_abort rx_fcs   fifofull
I (1305279) wifi: 0        bfb      bfa      0        4fa7     7dbf     47c      0       
I (1305287) wifi: afull    rx_oth   tkiperr  samebm   ack_int  rts_int  cts_int  rifs_int
I (1305295) wifi: 0        76f0     0        1        2dc      10       0        0       
I (1305302) wifi: blockerr hoperr   txrts    txcts    track    trcts    trigger  txhung   panic   
I (1305311) wifi: 0        0        0        0        7a       10       0        0        0       
I (1305320) wifi:
I (1305321) wifi: DIAG:
I (1305323) wifi: diag0    diag1    diag2    diag3    diag4    diag5    diag6   
I (1305331) wifi: f00000   0        fa00a08  0        806b8001 1800080e 0       
I (1305338) wifi: diag7    diag8    diag9    diag10   diag12   diagsel 
I (1305344) wifi: 8240     0        82c000   40       0        0       
I (1305350) wifi:
I (1305352) wifi: Power save:
I (1305355) wifi: tbtt     beacon   timset   dtimset  tbttdrm  actto    txwake   null1    null0    bcndly   sleep    wake     sleept   waket    totalt   percent 
I (1305369) wifi: 12697    12592    41       1280     12696    114      70       369      311      423      12719    12718    1183537534 116589216 1300126750 91.03 %
I (1305383) wifi:
I (1305385) wifi: HMAC RX:
I (1305387) wifi: ifx all      data     ctl    mgmt   bcn/probe
I (1305393) wifi: sta 3263     1125     0      2138   2132 
I (1305398) wifi: ap  0        0        0      0      0    
I (1305404) wifi:
I (1305405) wifi: ifx auth     assoc    deauth action amsdu
I (1305411) wifi: sta 1        1        0      4      0    
I (1305416) wifi: ap  0        0        0      0      0    
I (1305421) wifi:
I (1305423) wifi: HMAC TX:
I (1305425) wifi: sta_tx   ap_tx    psq_tx psq_mc psq_uc ps_uc1 reorder oos   amsdu
I (1305433) wifi: 136      0        0      0      0      0       0    

happened again below:
I (1884763) wifi:bcn_timout,ap_probe_send_start
I (1884764) wifi: BUFFER:
I (1884765) wifi: t  max alloc total_alloc total_free fail_fc  fail_oom
I (1884771) wifi: 1  32  0     177         177        0        0       
I (1884778) wifi: 2  32  0     16          16         0        0       
[... 30 lines cut ...]
I (1884925) wifi: blockerr hoperr   txrts    txcts    track    trcts    trigger  txhung   panic   
I (1884934) wifi: 0        0        1        0        9a       17       0        0        0       
I (1884943) wifi:
I (1884944) wifi: DIAG:
I (1884946) wifi: diag0    diag1    diag2    diag3    diag4    diag5    diag6   
I (1884954) wifi: 400001   0        fa00a02  0        806b8001 18000891 80      
I (1884961) wifi: diag7    diag8    diag9    diag10   diag12   diagsel 
I (1884967) wifi: 8240     0        82c000   40       0        0       
I (1884973) wifi:
I (1884975) wifi: Power save:
I (1884978) wifi: tbtt     beacon   timset   dtimset  tbttdrm  actto    txwake   null1    null0    bcndly   sleep    wake     sleept   waket    totalt   percent 
I (1884992) wifi: 18357    18164    52       1827     18356    156      98       521      429      423      18389    18388    1716772001 162944958 1879716959 91.33 %
I (1885006) wifi:
I (1885008) wifi: HMAC RX:
I (1885010) wifi: ifx all      data     ctl    mgmt   bcn/probe
I (1885016) wifi: sta 4795     1748     0      3047   3041 
I (1885021) wifi: ap  0        0        0      0      0    
I (1885027) wifi:
I (1885028) wifi: ifx auth     assoc    deauth action amsdu
I (1885033) wifi: sta 1        1        0      4      0    
I (1885039) wifi: ap  0        0        0      0      0    
I (1885044) wifi:
I (1885046) wifi: HMAC TX:
I (1885048) wifi: sta_tx   ap_tx    psq_tx psq_mc psq_uc ps_uc1 reorder oos   amsdu
I (1885056) wifi: 177      0        0      0      0      0       0

Janrupf, 2022-10-07:
Here is my dump:
```
I (23128) wifi: BUFFER:
I (23128) wifi: t  max alloc total_alloc total_free fail_fc  fail_oom
I (23128) wifi: 1  32  0     44          44         0        0       
I (23138) wifi: 2  32  1     14          13         0        0       
I (23148) wifi: 3  32  0     53          53         0        0       
I (23148) wifi: 4  32  0     0           0          0        0       
I (23158) wifi: 5  0   0     4           4          0        0       
I (23168) wifi: 6  0   0     0           0          0        0       
I (23168) wifi: 7  32  2     124         78         22       0       
I (23178) wifi: 8  32  0     102         44         29       0       
I (23178) wifi: 9  0   0     0           0          0        0       
I (23188) wifi:
I (23188) wifi: LMAC TX:
I (23188) wifi: tx_all   lrc src age to 
I (23198) wifi: 80       0   0   0   0  
I (23198) wifi: 0        0   0   0   0  
I (23198) wifi: 31       0   0   0   0  
V (23208) wifi:I (23208) wifi:back home chan=<13,2>, current chan=<13,2> 0        0   0   0   0  

I (23218) wifi: 0        0   0   0   0  
I (23218) wifi:
I (23218) wifi: LMAC RX:
I (23218) wifi: ampdu    mdpu     mgmt     ctrl     frag
I (23228) wifi: 33       63       80       0        0   
I (23228) wifi:
I (23238) wifi: HW RX:
I (23238) wifi: rx_suc   rx_end   full#1   full#2  
I (23238) wifi: b0       65ac     0        0       
I (23248) wifi:
I (23248) wifi: Hardware:
I (23248) wifi: rxbufbk  rxcckerr ofdmerr  rx_agc   rxsf     rx_abort rx_fcs   fifofull
I (23258) wifi: 2        5e1f     5e6d     0        0        3b1      5e       0       
I (23268) wifi: afull    rx_oth   tkiperr  samebm   ack_int  rts_int  cts_int  rifs_int
I (23268) wifi: 0        2bd      0        0        49       18       1c       0       
I (23278) wifi: blockerr hoperr   txrts    txcts    track    trcts    trigger  txhung   panic   
I (23288) wifi: 0        0        42       0        8        18       3        1b       0       
I (23298) wifi:
I (23298) wifi: DIAG:
I (23298) wifi: diag0    diag1    diag2    diag3    diag4    diag5    diag6   
I (23308) wifi: 1b00001  0        fa00a01  0        1        a20801   0       
I (23318) wifi: diag7    diag8    diag9    diag10   diag12   diagsel 
I (23318) wifi: 2810     0        8ac000   40       0        0       
I (23328) wifi:
I (23328) wifi: Power save:
I (23328) wifi: tbtt     beacon   timset   dtimset  tbttdrm  actto    txwake   null1    null0    bcndly   sleep    wake     sleept   waket    totalt   percent 
I (23348) wifi: 57       27       1        10       55       16       17       26       25       430      67       66       9958178  6375768  16333946 60.97 %
I (23358) wifi:
I (23358) wifi: HMAC RX:
I (23358) wifi: ifx all      data     ctl    mgmt   bcn/probe
I (23368) wifi: sta 132      72       0      44     39   
I (23378) wifi: ap  0        0        0      0      0    
I (23378) wifi:
I (23378) wifi: ifx auth     assoc    deauth action amsdu
I (23388) wifi: sta 1        1        0      3      0    
I (23388) wifi: ap  0        0        0      0      0    
I (23398) wifi:
I (23398) wifi: HMAC TX:
I (23398) wifi: sta_tx   ap_tx    psq_tx psq_mc psq_uc ps_uc1 reorder oos   amsdu
I (23408) wifi: 44       0        0      0      0      0       2    
```

monkeytronics, 2023-04-24:
Any progress? I have seen similar problems during OTA update. I am able to recover the connection about 50% of the time by deinit & init the wifi again. But it's a huge problem.

zhangyanjiaoesp, 2023-05-23:
@monkeytronics Which IDF version do you use? Can this issue be reproduced using the official example `idf/examples/system/ota`? If not, can you provide a demo ?

AxelLin, 2023-05-25:
> @monkeytronics @AxelLin @Janrupf
> 
> 1. Which IDF version do you use?
> 2. Which chip do you use？
> 3. Can you confirm that your issue is related to modem sleep ? Does calling `esp_wifi_set_ps(0)`  function solve the problem?

https://github.com/espressif/esp-idf/issues/9108#issuecomment-1271130751 includes all the information you asked. I think you cannot set esp_wifi_set_ps(0) if BT is enabled.

AxelLin, 2023-05-25:
> @listout
> 
> * From your log, maybe you need to disable modem sleep by calling: esp_wifi_set_ps(0);
> * Also, when you still meet bcn_timout, you can calling: esp_wifi_statis_dump(0xffff); to dump the wifi internal statis.

@MaxwellAlan 
There are several esp_wifi_statis_dump() in the follow up responses.
Do you see anything unusual?

monkeytronics, 2023-05-25:
@AxelLin Hi there - 
IDF    - 4.3.2-dirty
Chip  - ESP32-WROOM-32E
Sleep - No! I did see a significant improvement when I switched sleep off though. 

I will add the status dump to the code. I don't have a device connected up right now, but as soon as I do, I'll blast it with a heap of OTA updates until it fails and report back.

zhangyanjiaoesp, 2023-06-30:
@AxelLin 
It seems normal from your statis dump logs. There are 3 things still need to check:
1. what BT was doing when the `bcn_timout` occur? 
2. Please provide a capture packets for this issue.We want to check whether the probe request packet has been send out successfully when `bcn_timout` occur.
3. It will be better if you can provide a demo for us to reproduce this problem.

Alvin1Zhang, 2024-03-18:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates, thanks.

closed 2024-03-18 by Alvin1Zhang

Ina1413, 2025-06-23:
Hello, having a very similar issue. ` esp_wifi_set_ps(WIFI_PS_NONE);  // Disable power-saving to avoid bcn_timeout` only reduced the frequency of the bcn timeout messages I am getting. I am trying to do something similar to @Janrupf  but my WiFi rarely reconnects and when it does I often have to restart to file upload task. I am running the latest ESP version so I'm not sure why this is happening.