espressif/esp-idf #9109: ESP32C2 tjpgd example run will crash

dreamcmi, 2022-06-06:
I try to run the example of tjpgd on the ESP8684-DevKitM-1 development board, but it will crash, the following is the error log, how can I solve this problem?
```shell
ESP-ROM:esp8684-api2-20220127
Build:Jan 27 2022
rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403805be
0x403805be: esp_restart_noos at /home/darren/esp/esp-idf/components/esp_system/port/soc/esp32c2/system_internal.c:112 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6108,len:0x18b4
load:0x403ae000,len:0x864
load:0x403b0000,len:0x2598
entry 0x403ae000
I (23) boot: ESP-IDF v5.0-dev-3290-g01d014c42d 2nd stage bootloader
I (23) boot: compile time 22:51:47
I (23) boot: chip revision: 0
I (26) boot.esp32c2: MMU Page Size  : 64K
I (31) boot.esp32c2: SPI Speed      : 60MHz
I (36) boot.esp32c2: SPI Mode       : DIO
I (40) boot.esp32c2: SPI Flash Size : 2MB
I (45) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 00100000
I (84) boot: End of partition table
I (88) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=18f38h (102200) map
I (128) esp_image: segment 1: paddr=00028f60 vaddr=3fca9d70 size=00ce8h (  3304) load
I (130) esp_image: segment 2: paddr=00029c50 vaddr=40380000 size=063c8h ( 25544) load
I (143) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=10c90h ( 68752) map
I (165) esp_image: segment 4: paddr=00040cb8 vaddr=403863c8 size=0399ch ( 14748) load
I (176) boot: Loaded app from partition at offset 0x10000
I (177) boot: Disabling RNG early entropy source...
I (188) cpu_start: Pro cpu up.
I (197) cpu_start: Pro cpu start user code
I (197) cpu_start: cpu freq: 120000000 Hz
I (197) cpu_start: Application information:
I (200) cpu_start: Project name:     lcd_tjpgd
I (205) cpu_start: App version:      1
I (210) cpu_start: Compile time:     Jun  6 2022 22:51:42
I (216) cpu_start: ELF file SHA256:  d14307d530b58591...
I (222) cpu_start: ESP-IDF:          v5.0-dev-3290-g01d014c42d
I (228) heap_init: Initializing. RAM available for dynamic allocation:
I (236) heap_init: At 3FCABCF0 len 0002C310 (176 KiB): D/IRAM
I (242) heap_init: At 3FCD8000 len 000074BC (29 KiB): STACK/DRAM
I (249) spi_flash: detected chip: generic
I (253) spi_flash: flash io: dio
W (257) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (271) sleep: Configure to isolate all GPIO pins in sleep state
I (277) sleep: Enable automatic switching of GPIO sleep configuration
I (284) cpu_start: Starting scheduler.
I (289) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (299) gpio: GPIO[1]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (299) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
total size:210892
free size:21588
minimum free size:18488
Heap summary for capabilities 0x00000800:
  At 0x3fcabcf0 len 181008 free 21044 allocated 158340 min_free 17944
    largest_free_block 19968 alloc_blocks 242 free_blocks 2 total_blocks 244
  At 0x3fcd8000 len 29884 free 544 allocated 27716 min_free 544
    largest_free_block 296 alloc_blocks 59 free_blocks 1 total_blocks 60
  Totals:
    free 21588 allocated 186056 min_free 18488 largest_free_block 19968
1free size:21588
Heap summary for capabilities 0x00000008:
  At 0x3fcabcf0 len 181008 free 21044 allocated 158340 min_free 17944
    largest_free_block 19968 alloc_blocks 242 free_blocks 2 total_blocks 244
[... 79 lines cut ...]
3fcae6e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcae700: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: d14307d530b58591
ELF file SHA256: d14307d530b58591

Rebooting...

```
I tried adding the following code, but didn't find the reason.
```c
//
    printf("total size:%d\n",heap_caps_get_total_size(MALLOC_CAP_INTERNAL));
    printf("free size:%d\n",heap_caps_get_free_size(MALLOC_CAP_INTERNAL));
    printf("minimum free size:%d\n",heap_caps_get_minimum_free_size (MALLOC_CAP_INTERNAL));
    heap_caps_print_heap_info ( MALLOC_CAP_INTERNAL ) ;
    // Allocate memory for the pixel buffers
    for (int i = 0; i < 2; i++) {
        printf("1free size:%d\n",heap_caps_get_free_size(MALLOC_CAP_DMA));
        heap_caps_print_heap_info (MALLOC_CAP_DMA) ;
        s_lines[i] = heap_caps_malloc(EXAMPLE_LCD_H_RES * PARALLEL_LINES * sizeof(uint16_t), MALLOC_CAP_DMA);
        assert(s_lines[i] != NULL);
    }

```

suda-morris, 2022-06-07:
@dreamcmi yes, OOM issue. If  you change the `PARALLEL_LINES` from 16 to 10, I think this example will be runnable again on C2

moefear85, 2022-06-07:
@dreamcmi 

it is not clear if you know what you are doing. please post your code so the mistake can be found. You are allocating DMA memory, which is not something one generally does for de/compressing an image, as it is very limited. Try using normal memory instead.

dreamcmi, 2022-06-08:
> @dreamcmi yes, OOM issue. If you change the `PARALLEL_LINES` from 16 to 10, I think this example will be runnable again on C2

Oh thanks, after I tried changing PARALLEL_LINES to 16, the ESP32C2 doesn't crash at runtime anymore.
I created a pull requests to fix this and hopefully can be merged
https://github.com/espressif/esp-idf/pull/9119

closed 2022-06-11 by dreamcmi