espressif/esp-idf #9112: UDP broadcast fails with errno 5

mzimmersfluidra, 2022-06-07:
## Environment

- Development Kit:      [none]
 Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): v4.4.1
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [|Yes (please give details)] VS Code
- Power Supply:         [external 5V]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

successful return of sendto() call.
### Actual Behavior
errno = 5.
### Steps to reproduce

1. run code

### Code to reproduce this issue

```cpp
iobAnnounce = socket(AF_INET, SOCK_DGRAM, IPPROTO_UDP);

setsockopt(iobAnnounce, SOL_SOCKET, SO_BROADCAST, &BROADCAST_ENABLE, sizeof(BROADCAST_ENABLE));

memset(&servaddrAnnounce, 0, sizeof(servaddrAnnounce));
servaddrAnnounce.sin_family = AF_INET;
servaddrAnnounce.sin_port = htons(30303);

netif = esp_netif_get_handle_from_ifkey("WIFI_STA_DEF");
esp_netif_get_ip_info(netif, &ip_info);
subnetBroadcast = (ip_info.ip.addr & ip_info.netmask.addr) | ~(ip_info.netmask.addr);
servaddrAnnounce.sin_addr.s_addr = subnetBroadcast;

else if(AnnounceCount >= 1 && AnnounceCount <= 6) {

retval_ann = sendto(iobAnnounce,
				udp_data,
				sizeof(udp_data),
				0,
				(const struct sockaddr *) &servaddrAnnounce,
				sizeof(servaddrAnnounce));
```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Other items if possible
C:\repos\iQPump-ESP32\fw\build\iQPump.elf
C:\repos\iQPump-ESP32\fw\sdkconfig
- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

ESP-YJM, 2022-06-07:
Hi @mzimmersfluidra , errno = 5 means ` ERR_ARG`, you could check the parameters you input.

mzimmersfluidra, 2022-06-07:
@ESP-YJM  servaddrAnnounce.sin_addr.s_addr is 0xff56a8c0 (host order). I do not convert it before the call to sendto().

Thanks...

mzimmersfluidra, 2022-06-07:
Problem found: [the problem](https://stackoverflow.com/questions/72523184/why-am-i-getting-i-o-error-on-a-udp-broadcast-sendto/72534157#72534157).

This may be documented somewhere, but I've never run across it.

closed 2022-06-07 by mzimmersfluidra