espressif/esp-idf #9113: 我该如何获取esp_http_client拿到的内容

HongYan12345, 2022-06-07:
我使用esp_http_client来调一个api，已经调通200了，返回长度也不是-1，但是我拿不到返回的内容

下面是我的代码

` 
    
    esp_http_client_config_t config = {
        .event_handler = _http_event_handler,
        .url = "https://thingsboard.cloud:443/api/auth/login",
    };
    esp_http_client_handle_t client = esp_http_client_init(&config);

    // POST
    const char *post_data = "{\"username\":\"ivan.cmaya@alumnos.upm.es\", \"password\":\"SBC21g09\"}";
    esp_http_client_set_method(client, HTTP_METHOD_POST);
    esp_http_client_set_header(client, "Content-Type", "application/json");
    esp_http_client_set_post_field(client, post_data, strlen(post_data));
    
    esp_err_t err = esp_http_client_perform(client);

    //esp_http_client_read(client, output_buffer, MAX_HTTP_OUTPUT_BUFFER);
    //int data_read = esp_http_client_read_response(client, output_buffer, MAX_HTTP_OUTPUT_BUFFER);
    if (err == ESP_OK) {
        int data_read = esp_http_client_read_response(client, output_buffer, MAX_HTTP_OUTPUT_BUFFER);
            if (data_read >= 0) {
                ESP_LOGI(TAG, "HTTP POST Status = %d, content_length = %d",
                esp_http_client_get_status_code(client),
                esp_http_client_get_content_length(client));
                ESP_LOG_BUFFER_HEX(TAG, output_buffer, strlen(output_buffer));
            } else {
                ESP_LOGE(TAG, "Failed to read response");
            }
    } else {
        ESP_LOGE(TAG, "HTTP POST request failed: %s", esp_err_to_name(err));
    }
   
    esp_http_client_cleanup(client);
`

返回的日志里显示
I (4159) HTTP_MP3_EXAMPLE: [ +++ ] 插入api获取代码
I (6839) HTTP_MP3_EXAMPLE: HTTP POST Status = 200, content_length = 1156
I (6839) HTTP_MP3_EXAMPLE: HTTP_EVENT_DISCONNECTED

我想讲返回的内容存到全局变量里面，但是并没有存进去，是哪里出了问题？

ESP-YJM, 2022-06-07:
@HongYan12345 你可以在 `_http_event_handler` 回调里 event `HTTP_EVENT_ON_DATA` 里获取到数据。参考 https://github.com/espressif/esp-idf/blob/master/examples/protocols/esp_http_client/main/esp_http_client_example.c#L65

HongYan12345, 2022-06-07:
> @HongYan12345 你可以在 `_http_event_handler` 回调里 event `HTTP_EVENT_ON_DATA` 里获取到数据。参考 https://github.com/espressif/esp-idf/blob/master/examples/protocols/esp_http_client/main/esp_http_client_example.c#L65

我使用的就是这个例子中的_http_event_handler, 但是我注意到日志里一直是HTTP_EVENT_DISCONNECTED， 说明没有进到HTTP_EVENT_ON_DATA，但是状态和长度都是对，我不知道问题出在哪里

ESP-YJM, 2022-06-07:
> 我使用的就是这个例子中的_http_event_handler, 但是我注意到日志里一直是HTTP_EVENT_DISCONNECTED， 说明没有进到HTTP_EVENT_ON_DATA，但是状态和长度都是对，我不知道问题出在哪里

`HTTP_EVENT_ON_DATA` 里的打印 LOG 等级是 debug 的。

HongYan12345, 2022-06-07:
> `HTTP_EVENT_ON_DATA` 里的打印 LOG 等级是 debug 的。

我成功拿到数据了，非常感谢。我还有一个问题， 我使用我拿到的这个token后，我要把它放到另一个api的包头里，但是这个token很长，有600+的字符

`  下面是我写的代码

        esp_http_client_config_t config = {
        .event_handler = _http_event_handler,
        .url = "https://thingsboard.cloud:443/api/tenant/subscription/data",
        .user_data = local_response_buffer,
    };
    esp_http_client_handle_t client = esp_http_client_init(&config);
    // GET
    esp_http_client_set_method(client, HTTP_METHOD_GET);
    esp_http_client_set_header(client, "Authorization", "Bearer eyJhbGciOiJIUzUxMiJ9.eyJzdWIiOiJpdmFuLmNtYXlhQGFsdW1ub3MudXBtLmVzIiwic2NvcGVzIjpbIlRFTkFOVF9BRE1JTiJdLCJ1c2VySWQiOiIzYTA2OGMzMC01OTBjLTExZWMtYTkxOS01NTZlOGRiZWYzNWMiLCJmaXJzdE5hbWUiOiJJdmFuIiwibGFzdE5hbWUiOiJDb2NhIE1heWEiLCJlbmFibGVkIjp0cnVlLCJpc1B1YmxpYyI6ZmFsc2UsImlzQmlsbGluZ1NlcnZpY2UiOmZhbHNlLCJwcml2YWN5UG9saWN5QWNjZXB0ZWQiOnRydWUsInRlcm1zT2ZVc2VBY2NlcHRlZCI6dHJ1ZSwidGVuYW50SWQiOiJlMmRkNjUwMC02NzhhLTExZWItOTIyYy1mNzQwMjE5YWJjYjgiLCJjdXN0b21lcklkIjoiMTM4MTQwMDAtMWRkMi0xMWIyLTgwODAtODA4MDgwODA4MDgwIiwiaXNzIjoidGhpbmdzYm9hcmQuY2xvdWQiLCJpYXQiOjE2NDc1MzUwODAsImV4cCI6MTY0NzU2Mzg4MH0.1X3grSDrdIjWzfZCU0h3iwljZLs6ibSMkTOaRkyERoGipAkV3NZ89CGQ6azr2eLXVDhAxBTKPYmwZOGfYRvFIg");
    esp_err_t err = esp_http_client_perform(client);
    //esp_http_client_read(client, output_buffer, MAX_HTTP_OUTPUT_BUFFER);
    //int data_read = esp_http_client_read_response(client, output_buffer, MAX_HTTP_OUTPUT_BUFFER);
    if (err == ESP_OK) {
            ESP_LOGI(TAG, "HTTP GET Status = %d, content_length = %d",
            esp_http_client_get_status_code(client),
            esp_http_client_get_content_length(client));
            printf("%s\n", local_response_buffer);
    } else {
        ESP_LOGE(TAG, "HTTP POST request failed: %s", esp_err_to_name(err));
    }
    esp_http_client_cleanup(client);`

，显示E (5681) HTTP_HEADER: Buffer length is small to fit all the headers， 那我应该如何调整才行？

ESP-YJM, 2022-06-08:
esp_http_client_config_t  配置里加上 `.buffer_size_tx = 1024`

Alvin1Zhang, 2022-07-11:
关闭此问题。如您有更多更新，请重新打开此问题，或者重新开一个 issue，谢谢。

closed 2022-07-11 by Alvin1Zhang