espressif/esp-idf #9115: Bug: MQTT QOS 2 duplicated messages

Sjurinho, 2022-06-07:
## Environment

- Development Kit: none
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.2.3-dirty
- Build System:  CMake
- Operating System: Linux
- Using an IDE?: VSCode
- Power Supply: external 5V

## Problem Description

Hi, I have set up an esp32 with PPPoS that communicates with a server over mqtt. I have specified QoS 2 for my messages, and I am using the `esp_mqtt_client_enqueue()` API. I have experienced that when sending more than one message in quick succession that I sometimes receive duplicate messages in my server. Does anyone share this experience, or is it most likely a broker issue?

### Expected Behavior
With QoS 2, messages are received by the server exactly once.

### Actual Behavior
Duplicated messages are seen on the server

david-cermak, 2022-06-07:
@Sjurinho Could you please capture the sent messages to check if they have the `DUP` flag set? I've seen this problem in the past, but was fixed by:

https://github.com/espressif/esp-mqtt/blob/684843a3096fea77346fd7f6994dcf984d65308b/mqtt_client.c#L1380-L1382

correctly setting the flag for potentially duplicated messages.

(you can use the standard WiFi or Ethernet interface for capturing the packets)

Sjurinho, 2022-06-10:
@david-cermak It does not have the `DUP` flag set, but I believe the issue comes from the fact that I receive two `PUBREC` messages for the same message id. 
```
D (99408) MQTT_CLIENT: mqtt_message_receive: first byte: 0x50
D (99409) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x2
D (99409) MQTT_CLIENT: mqtt_message_receive: total message length: 4 (already read: zu)
D (99410) MQTT_CLIENT: mqtt_message_receive: read_len=2
D (99411) MQTT_CLIENT: mqtt_message_receive: transport_read():zu zu
D (99411) MQTT_CLIENT: msg_type=5, msg_id=35286
D (99411) MQTT_CLIENT: received MQTT_MSG_TYPE_PUBREC
D (99657) Fullness: uxTaskGetHighStackWaterMark: 856
D (99708) MQTT_CLIENT: mqtt_message_receive: first byte: 0x50
D (99709) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x2
D (99709) MQTT_CLIENT: mqtt_message_receive: total message length: 4 (already read: zu)
D (99711) MQTT_CLIENT: mqtt_message_receive: read_len=2
D (99711) MQTT_CLIENT: mqtt_message_receive: transport_read():zu zu
D (99711) MQTT_CLIENT: msg_type=5, msg_id=35286
D (99711) MQTT_CLIENT: received MQTT_MSG_TYPE_PUBREC
D (100044) MQTT_CLIENT: mqtt_message_receive: first byte: 0x70
D (100046) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x2
D (100046) MQTT_CLIENT: mqtt_message_receive: total message length: 4 (already read: zu)
D (100047) MQTT_CLIENT: mqtt_message_receive: read_len=2
D (100048) MQTT_CLIENT: mqtt_message_receive: transport_read():zu zu
D (100048) MQTT_CLIENT: msg_type=7, msg_id=35286
D (100048) MQTT_CLIENT: received MQTT_MSG_TYPE_PUBCOMP
D (100048) MQTT_CLIENT: pending_id=35286, pending_msg_count = 1
D (100048) OUTBOX: DELETED msgid=35286, msg_type=3, remain size=0
D (100049) MQTT_CLIENT: Receive MQTT_MSG_TYPE_PUBCOMP, finish QoS2 publish
I (100049) MqttConnector: MQTT_EVENT_PUBLISHED, msg_id=35286
```
Does this indicate that the issue may lie with the server?

Sjurinho, 2022-06-13:
After checking the [http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.pdf](documentation), entry `[MQTT-4.3.3-1]`, shouldn't the `DUP` flag be set to 0 in case of QOS 2 @david-cermak?

euripedesrocha, 2022-06-28:
Hi @Sjurinho help me clarifying the scenario. From your logs client is receiving two `PUBREC` so it means that your server may be sending it twice. Can you check the server side? It's possible to share logs that have the full communication for a particular message ID, from `PUBLISH` to `PUBCOMP`? Are you using some custom broker implementation?

Alvin1Zhang, 2022-09-06:
@Sjurinho Thanks for reporting, would you please help share if any updates for the issue? Thanks.

Sjurinho, 2022-09-06:
Hi, sorry for the delay in my replies @euripedesrocha. We are still searching for the issue. We have stress tested our broker quite severely by spamming messages, and as far as I can see it never receives duplicated messages, but rather drops messages on high loads.
We are using a [MQTTNet](https://github.com/dotnet/MQTTnet) broker. 
I have some packet information, I will add a png that shows the communication between the server and client for a specific message id. Sorry that it is a bit sparse, but I would prefer not to share the entire log to avoid having my server information on full display:) Let me know if you have any suggestions to be able to send the entire log. 
We used a micropython version of the paho mqtt client earlier, also running on a esp32, and there we did not experience this issue.
[image]

euripedesrocha, 2022-09-19:
Hi @Sjurinho, I spent some time looking into our implementation. From my understanding, you may have a retry on sending the messages. Despite the retry being described on MQTT standard as REQUIRED only on reconnection, we retry with `message_retransmission_timeout` the default for this field is 1s.

I still want to understand about "sending more than one message in quick succession", could you provide a short code that I can use to reproduce?

Sjurinho, 2022-09-19:
@euripedesrocha Yes, I will try to create a small project reproducing the error as soon as I have some free time, hopefully by the end of this week. In words, what I mean is a burst of messages. So I use the `esp_mqtt_client_enqueue(message...) ` api to enqueue around 4 messages (in 2-3 different tasks) which all gets sent to the queue.

I can try to increase the `message_retransmission_timeout`

Sjurinho, 2022-09-20:
@euripedesrocha I have done some small load testing with an increased `message_retransmission_timeout`, and I have not been able to reproduce the issue. Whether or not the issue is solved needs some time to see, but so far it seems promising:)
I was earlier running `idf v4.2.3` where this parameter was not available.

Sjurinho, 2022-09-20:
@euripedesrocha Is it possible to add a debug log for the case when you re-request a message? This would make it easier to find the cases where you have too strict timing requirements on the communication protocol

euripedesrocha, 2022-09-20:
Yes, it is. I will add it, and of course we are always open to have a PR providing ways to improve the library.

closed 2023-01-13 by Alvin1Zhang