espressif/esp-idf #9118: ESP32S3: SPI bus not working after isolating SPI bus lines during deep sleep

samvangysegem, 2022-06-07:
## Environment

- Module or chip used: ESP32S3
- IDF version: v4.4.1-185-gc5deb7a971
- Build System: idf.py
- Operating System: macOS
- Using an IDE?: No
- Power Supply: external 3.3V

## Problem Description

During first startup, the node is able to communicate with external devices over SPI which is validated with an oscilloscope. After going to deep sleep, at which time the SPI bus signals (CLK, MISO and MOSI) are isolated using the `rtc_gpio_init` and `rtc_gpio_isolate` commands, and waking up, at which time the `rtc_gpio_deinit` method is called for the respective SPI bus signals, communication over the SPI bus is no longer working. The SPI bus initialisation using `spi_bus_initialize` is successful, as is adding devices using `spi_bus_add_device`. Reading or writing to or from the SPI bus is not working however, which is confirmed by an oscilloscope monitoring the SPI bus signals: neither CLK, MISO or MOSI go to 3V3 when the bus is initialised after deep sleep but all remain fixed a 0V. This problem does not occur when the SPI bus signals are not isolated during deep sleep. Hence, I believe the problem is related to the `rtc_gpio_deinit` command.

### Steps to reproduce

1. Isolate the signals of the SPI bus using the `rtc_gpio_init` and `rtc_gpio_isolate` commands (GPIO 11, 12 and 13 in case of the ESP32S3)
2. Enter deep sleep and wake back up (timer wakeup)
3. Call the `rtc_gpio_deinit` method for the corresponding SPI bus signals
4. Initialise the SPI bus and add a device
5. Write to the device (point of failure)

## Debug Logs

There are no specific debug logs since SPI bus initialisation and adding devices does not produce any errors. The failure of communication is recognised through the oscilloscope when reading registers from external devices does not lead to any monitored activity on the SPI bus.

samvangysegem, 2022-06-22:
Any updates on this issue?

skip-dt, 2022-10-01:
@ahne6000 Thanks for posting this!! It fixed my problem as well.  I had been pulling what little is left of my hair out...

wolfstrassen, 2023-03-15:
> Hi @samvangysegem, I could fix the same issue calling both `rtc_gpio_deinit` and `rtc_gpio_hold_dis`. Not quite sure if it's a bug or a missing bit in the espressif docs tho

Thanks for this, I was having a similar problem with I2C pins, this solved it.