espressif/esp-idf #9120: [Crash][esp32s3][v4.4.1] assert failed: spinlock_release spinlock.h:158 (core_id == lock->owner)

chipweinberger, 2022-06-08:
**Please see this thread:** https://www.esp32.com/viewtopic.php?f=13&t=27852&p=98615#p98615

**esp32s3**
**esp-idf 4.4.1** (latest commit: Tue May 31 10:28:03)

**I'm hitting this assert:** 
`assert(core_id == lock->owner); // This is a mutex we didn't lock, or it's corrupt`

**It does appear to be corruption:**
core_id: 0xabab 
lock->owner: 0xb33fffff

I don't think my code is the code is writing these corrupt values. 

**Logs:**
```
I (19385) pd_sntp: Starting SNTP
I (19395) pd led ux: home wifi connected
I (19405) pd mdns: mdns_delegate_hostname_add(192.168.0.180, pe6gzq)
I (19415) pd mdns: success
I (19565) : core_id 0xabab, lock->owner 0xb33fffff

assert failed: spinlock_release spinlock.h:163 (core_id == lock->owner)

Backtrace:0x40376407:0x3fcb95a00x4038286d:0x3fcb95d0 0x4038d99a:0x3fcb9600 0x40387e49:0x3fcb9720 0x40384068:0x3fcb9750 0x420cd39d:0x3fcb9790 0x420cd8db:0x3fcb97c0 0x420b94fc:0x3fcb97f0 0x420b95c2:0x3fcb9820 0x420ba832:0x3fcb9850 0x420c8aae:0x3fcb9880 0x420cd0f5:0x3fcb98d0 0x420b739d:0x3fcb9900 0x420b7444:0x3fcb9930 0x40387b5a:0x3fcb9960 
0x40376407: panic_abort at /esp_system/panic.c:402

0x4038286d: esp_system_abort at /esp_system/esp_system.c:128

0x4038d99a: __assert_func at /newlib/assert.c:85

0x40387e49: spinlock_release at /esp_hw_support/include/soc/spinlock.h:163
 (inlined by) vPortExitCritical at /freertos/port/xtensa/port.c:310

0x40384068: xQueueSemaphoreTake at /freertos/queue.c:1612

0x420cd39d: sys_mutex_lock at /lwip/port/esp32/freertos/sys_arch.c:82

0x420cd8db: sys_arch_protect at /lwip/port/esp32/freertos/sys_arch.c:480

0x420b94fc: do_memp_free_pool at /lwip/lwip/src/core/memp.c:370 (discriminator 2)

0x420b95c2: memp_free at /lwip/lwip/src/core/memp.c:440

0x420ba832: pbuf_free at /lwip/lwip/src/core/pbuf.c:786

0x420c8aae: ip6_input at /lwip/lwip/src/core/ipv6/ip6.c:701

0x420cd0f5: ethernet_input at /lwip/lwip/src/netif/ethernet.c:229

0x420b739d: tcpip_thread_handle_msg at /lwip/lwip/src/api/tcpip.c:180

0x420b7444: tcpip_thread at /lwip/lwip/src/api/tcpip.c:154

0x40387b5a: vPortTaskWrapper at /freertos/port/xtensa/port.c:131
```

chipweinberger, 2022-06-08:
I added a name to each spinlock:

```
typedef struct {
    NEED_VOLATILE_MUX uint32_t owner;
    NEED_VOLATILE_MUX uint32_t count;
    const char* name;
}spinlock_t;
```

And each time I crash I get this:
spinlock_release: name=queue.c-1 core_id=0xabab  lock->owner=0xb33fffff
spinlock_release: name=queue.c-2 core_id=0xabab  lock->owner=0xb33fffff

Which corresponds to the locks created by this code:

****esp-idf/components/freertos/queue.c**:**
```
BaseType_t xQueueGenericReset( QueueHandle_t xQueue,
                               BaseType_t xNewQueue )
{
   .......
#ifdef ESP_PLATFORM
    if( xNewQueue == pdTRUE )
    {
        portMUX_INITIALIZE(&pxQueue->mux, "queue.c-1");
    }
#endif // ESP_PLATFORM
  ........
}
```
```
static void prvInitialiseMutex( Queue_t * pxNewQueue )
    {
        if( pxNewQueue != NULL )
        {
       ..............
            /* In case this is a recursive mutex. */
            pxNewQueue->u.xSemaphore.uxRecursiveCallCount = 0;
#ifdef ESP_PLATFORM
            portMUX_INITIALIZE(&pxNewQueue->mux, "queue.c-2");
#endif // ESP_PLATFORM
        ...............
    }
```

Dazza0, 2022-06-08:
@chipweinberger We've recently run into a similar issue. There's a bug in the `spinlock.h` where the [external RAM atomic CAS function isn't being called on the ESP32-S3](https://github.com/espressif/esp-idf/blob/1329b19fe494500aeb79d19b27cfd99b40c37aec/components/esp_hw_support/include/soc/spinlock.h#L98).

Just to double check that you are seeing the same issue, could you...

- Attach your `sdkconfig`. In particular, what (if any) SPIRAM configurations are being enabled (i.e., is `malloc` allowed to allocated to SPIRAM)?
- The memory address of the spinlock. Is the spinlock being placed in SPIRAM (somewhere in address range `0x3C000000` to `0x3E000000`)

If it's the same issue, in `spinlock.h`, replacing

```c
#if defined(CONFIG_ESP32_SPIRAM_SUPPORT)
        if (esp_ptr_external_ram(lock)) {
            compare_and_set_extram(&lock->owner, SPINLOCK_FREE, &result);
        } else {
#endif
        compare_and_set_native(&lock->owner, SPINLOCK_FREE, &result);
#if defined(CONFIG_ESP32_SPIRAM_SUPPORT)
        }
#endif
```
with

```c
#if (CONFIG_ESP32_SPIRAM_SUPPORT || CONFIG_ESP32S3_SPIRAM_SUPPORT)
        if (esp_ptr_external_ram(lock)) {
            compare_and_set_extram(&lock->owner, SPINLOCK_FREE, &result);
        } else {
#endif
        compare_and_set_native(&lock->owner, SPINLOCK_FREE, &result);
#if (CONFIG_ESP32_SPIRAM_SUPPORT || CONFIG_ESP32S3_SPIRAM_SUPPORT)
        }
#endif
```
should resolve the issue.

chipweinberger, 2022-06-08:
Thank you so much for this information! I think you got it!

> is malloc allowed to allocated to SPIRAM?

Yes. 

> The memory address of the spinlock. Is the spinlock being placed in SPIRAM (somewhere in address range 0x3C000000 to 0x3E000000)

Yes. `0x0x3de4ecdc`. Probably due to me setting `CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=16` in a desperate try to save some RAM.

> If it's the same issue, in spinlock.h, replacing

Yes! That appears to have fixed it!! 🎉🎉🎉🎉🎉🎉

**Is there a timeline for gettting this fix in?**

Thankfully I found better ways to reduce ram, so I can disable those options for now.

ESP-Marius, 2022-06-09:
> Is there a timeline for gettting this fix in?

I've merged a fix to this on our internal master branch already, and it's being backported to 4.4. Should be available on github with the next sync.

closed 2022-06-11 by espressif-bot