espressif/esp-idf #9121: Use LVGL Direct mode in ESP32-S3 RGB panel example

itavero, 2022-06-08:
**Is your feature request related to a problem? Please describe.**

When using the RGB LCD panel peripheral, I still see quite some "tearing" due to parts of the buffer being updated.
I think this is partly due to how LVGL and the RGB peripheral have been integrated.

**Describe the solution you'd like**

The integration between LVGL and the RGB peripheral should be improved.
As described in [my forum post](https://esp32.com/viewtopic.php?f=13&t=28230), for these kind of peripherals it makes sense to use one screen-sized buffer and use [Direct mode](https://docs.lvgl.io/master/porting/display.html#direct-mode) in LVGL.
This will make sure that you only get a trigger once all the updates have been rendered.

Another thing that should probably be added is some synchronization/alignment to make sure that the actual "copying" of the buffer is done after the frame has been "transferred" to the RGB panel and before a new frame transfer is started.
In my forum post, I ask if I can use `on_frame_trans_done` for this. It has been "removed" from the example in the `master` branch, I believe due to an issue with the hardware, but I'm not 100% sure on that.

Note that the LVGL buffer probably also needs to be placed in the external PSRAM, as it will typically be too large for the internal RAM.

**Describe alternatives you've considered**

I tried using it in the same "partial buffer" mode as the example,  but for us the resulting experience when there are animations all over the screen are not acceptable.
Too much visible tearing/glitches occur.

I've not really tried any other examples, but I'm open to suggestions to get the experience as smooth as possible.

**Additional context**

* I'm using [this panel and dev kit](https://github.com/W00ng/ESP32-S3-RGB-Panel) for my tests, which I got from AliExpress (currently the only dev kit I could find that actually uses the RGB peripheral of the ESP32-S3.
* My [post on the Espressif forum](https://esp32.com/viewtopic.php?f=13&t=28230).
* An [older forum post on the LVGL forum](https://forum.lvgl.io/t/ssd1963-tearing-effect/653) about how tearing can be prevented.
   * I interpret it as: when the _flush_ callback is called from LVGL, wait until the current frame is done before updating the peripherals frame buffer and calling `lv_disp_flush_ready`
   * Perhaps `render_start_cb` can also be used to do some optimizations.

tore-espressif, 2022-06-15:
@itavero  Thanks for the detailed bug report and letting us know about this.

Unfortunately, there can be various reason for glitches with RGB interface. Having one screen-sized buffer in PSRAM a direct mode set in LVGL can definitely help. If your board allows it, set PSRAM frequency to 80MHz and octal mode to get maximum throughput.

The glitches usually depend on the RGB panel itself. Then, this section of configuration code is critical https://github.com/espressif/esp-idf/blob/master/examples/peripherals/lcd/rgb_panel/main/rgb_lcd_example_main.c#L116-128
For my particular LCD panel, I have to set `.flags.hsync_idle_low = true,` to get uncorrupted image. Try experimenting with theses settings and/or check you LCD panel datasheet.

Could you share some images/short videos of the glitches?

We also implemented some optimizations recently, that will be made public on GitHub within few day. I'll ping you when it is ready

tore-espressif, 2022-06-20:
Thank you for the provided code, it's good to hear that it works for you. I'll double check whether we can use the `on_frame_trans_done` 

> Looking forward to that!

The fix is available on our `master` branch now!

itavero, 2022-06-20:
> The fix is available on our `master` branch now!

Just tried it out, together with my aforementioned changes to "sync" access to the frame buffer, and the performance is indeed way better.
On one of our demos we got about 11 to 12 FPS before and now it's about 24 FPS (which looks way better).
Thanks for the update.

weilian1977, 2023-01-26:
关注一下

closed 2024-10-09 by tore-espressif