espressif/esp-idf #9124: ULP support for ESP32-S3

JitheshMurali, 2022-06-08:
As per the article, https://www.esp32.com/viewtopic.php?t=21906, which lists the status of various ESP-IDF features on ESP32-S3; ULP coprocessor is Not supported.
We would like to know when this support will be enabled as we are migrating from esp32 to esp32-s3 and our current esp32 based design use ULP for ADC wakeup in deep sleep mode.
Any alternate suggestions for the same use case will also be highly appreciated.

ESP-Marius, 2022-06-09:
Hi,

Basic ULP support for S3 is already merged to our internal master branch, and will be available the next time our internal branch is synced to Github

JitheshMurali, 2022-06-10:
Thank you @ESP-Marius. Will you be able to share a timeline on when this will be available on the Github?

ESP-Marius, 2022-06-13:
Looks like it synced this weekend, so it's possible to use the S3 ULP now. There is no example for using ADC from the ULP at the moment though, but I'm planning to work on adding one for the ULP-Riscv this week.

Are you looking at using the ULP-FSM or the ULP-Riscv core?

JitheshMurali, 2022-06-22:
Thank you @ESP-Marius . Sorry for the delay in response. Currently we are referencing the ULP-FSM ADC eg. But we are open to using the ULP-Riscv provided we can use the ADC.
Are the changes available in the stable/release/pre-release branches or the current master branch(4.3.3)?

ESP-Marius, 2022-06-28:
An example of using the ADC ULP-Riscv is under review at the moment and will be available on the latest master when merged. For the ULP FSM there is no ADC example yet (the current example is only tested for the ESP32), but it is on our todo list.

The changes for the ULP are currently only available on the latest master, but a backport for 4.4 is also on the way.

Alvin1Zhang, 2022-08-12:
Thanks for reporting, fix on master branch is available at https://github.com/espressif/esp-idf/commit/6cc871d793cbd6f7da9d3dcb276c963ea3a2203f, and on release/4.4 at https://github.com/espressif/esp-idf/commit/f8729d905e9556d17272698fd17d566bba2b0d6d. Feel free to reopen.

closed 2022-08-12 by Alvin1Zhang