espressif/esp-idf #9127: nimble ble_gatts_find_chr always zero

DGuimaras, 2022-06-08:
Hello,

I am testing the BLE nimble stack and on a central node was trying to get the handle of a given characteristic by calling the function 
ble_gatts_find_chr(BLE_UUID16_DECLARE(SERVICE_UUID), BLE_UUID16_DECLARE(CHAR_UUID), &handler_characteristic, &handle_value);
but it always returns zero in both. Is there a particular function that needs to be called first?

As an alternative, I tried to access the result of peer_disc_all(...). but no success. Where is the result stored and how can it be read?

My ultimate goal is to verify the characteristic UUID when the notification is received. Any other way this can be done?

chipweinberger, 2022-06-09:
I've never used that function, but you can get the handles like so, before you register:

```
extern uint16_t pd_ble_gatt_handle_midi;

// Define new custom service 
static const struct ble_gatt_svc_def gatt_defs[] = {
    {
        // Service: Midi
        .type = BLE_GATT_SVC_TYPE_PRIMARY,
        .uuid = (const ble_uuid_t*) &pd_ble_uuid_svc_midi,
        .includes = 0,
        .characteristics = (struct ble_gatt_chr_def[]) {
            {
                // Midi Characteristic 
                .uuid = (const ble_uuid_t*) &pd_ble_uuid_chr_midi,
                .access_cb = gatt_access_handler,
                .arg = 0,
                .descriptors = NULL,
                .flags = BLE_GATT_CHR_F_READ | 
                         BLE_GATT_CHR_F_WRITE_NO_RSP |
                         BLE_GATT_CHR_F_NOTIFY,
                .min_key_size = 0,
                .val_handle = &pd_ble_gatt_handle_midi,
            },
            {
                // No More Characteristics
                .uuid =  0,
                .access_cb = 0,
                .arg = 0,
                .descriptors = 0,
                .flags = 0,
                .min_key_size = 0,
                .val_handle = 0,
            }
        },
    },
    {
        // No More Services
        .type = 0,
        .uuid = 0,
        .includes = 0,
        .characteristics = 0
    }
};
```

IshaESP, 2022-06-09:
Hi @chipweinberger ,

1. gatts_* functions are to be used by server, gattc_* functions are primarily for client side.
2. ble_gatts_find_chr is for finding in local database. Client doesn't have database, server has. So using this function on client side may not give any information.
3. So to cross check the notification received on client side :
          a. client subscribes to notification using specific attr_handle, so it can be saved on client side and when it receives the    notification, it can be compared and verified.
          b. Or the other way, one can find the attr_handle associated with the specific UUID using this api - peer_chr_find_uuid()
              You can check it's demo implementation in examples/bluetooth/nimble/ble_spp/spp_client/main/main.c line 41
              This attr_handle obtained will be same to that when notification is received.

DGuimaras, 2022-06-09:
its solved. Thanks

closed 2022-06-09 by DGuimaras