espressif/esp-idf #9129: [Crash][v4.4.1][ESP32S3] Cache disabled but cached memory region accessed

chipweinberger, 2022-06-09:
Crashing with "Cache disabled but cached memory region accessed."

ESP32S3
esp-idf v4.4.1

Where would one begin debugging this issue? Is it related to SPI RAM? Is it related to https://github.com/espressif/esp-idf/issues/9120?

**I think it might be caused by an sdkconfig setting.**

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/8866381/sdkconfig.txt)

Details:
- crashes at seemingly random times, at different places, but usually not too long after boot

**Logs:**
```
I (10227) pd nvs: Setting Clock Start Time: 1654742890

Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). 

Core  0 register dump:
PC      : 0x40383d34  PS      : 0x00060234  A0      : 0x803795d7  A1      : 0x3fca0e50  
0x40383d34: xQueueGiveFromISR at /freertos/queue.c:1221

A2      : 0x3de5b948  A3      : 0x3fca0eb0  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0xbad00bad  A7      : 0x4038e0d0  A8      : 0x8001b85e  A9      : 0x3fca0e50  
0x4038e0d0: spi_flash_chip_generic_page_program at /spi_flash/spi_flash_chip_generic.c:231

A10     : 0x00000001  A11     : 0x00004cca  A12     : 0x3fcb6970  A13     : 0x3fcb6940  
A14     : 0x3fca318c  A15     : 0x3fcb698c  SAR     : 0x0000001f  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace:0x40383d31:0x3fca0e500x403795d4:0x3fca0e80 0x4037a51d:0x3fca0eb0 0x4037a549:0x3fca0ee0 0x4000ce7d:0x3fca0f00 0x40378836:0x3fca0f20 0x40389108:0x3fcb68d0 0x40389912:0x3fcb6900 0x4038e116:0x3fcb6940 0x4038e79e:0x3fcb6970 0x4037d361:0x3fcb69f0 0x420a2387:0x3fcb6a50 0x420a5be8:0x3fcb6a80 0x420a67a7:0x3fcb6ab0 0x420a701d:0x3fcb6ae0 0x420a75bd:0x3fcb6b30 0x420a5155:0x3fcb6b90 0x420a5757:0x3fcb6c00 0x420a3b32:0x3fcb6c30 0x420a3b87:0x3fcb6c70 0x420a3bf0:0x3fcb6ca0 0x42025a51:0x3fcb6cd0 0x4202d671:0x3fcb6d10 0x4202cd36:0x3fcb6d80 0x4202cd79:0x3fcb6dc0 0x4202ce0d:0x3fcb6df0 0x420c373d:0x3fcb6e20 0x420c8d19:0x3fcb6e70 0x420ce78e:0x3fcb6ea0 0x420b89b5:0x3fcb6ed0 0x420b8a58:0x3fcb6f00 0x40387b56:0x3fcb6f30 
0x40383d31: xQueueGiveFromISR at /freertos/queue.c:1221

0x403795d4: semphr_give_from_isr_wrapper at /bt/controller/esp32s3/bt.c:497

0x4037a51d: btdm_task_post_from_isr_impl at ??:?

0x4037a549: r_btdm_task_post_from_isr_hack at ??:?

0x40378836: _xt_lowint1 at /freertos/port/xtensa/xtensa_vectors.S:1111

0x40389108: spimem_flash_ll_cmd_is_done at /hal/esp32s3/include/hal/spimem_flash_ll.h:75 (discriminator 1)
 (inlined by) spi_flash_hal_poll_cmd_done at /hal/spi_flash_hal_common.inc:38 (discriminator 1)

0x40389912: spi_flash_hal_program_page at /hal/spi_flash_hal_iram.c:86

0x4038e116: spi_flash_chip_generic_page_program at /spi_flash/spi_flash_chip_generic.c:238

0x4038e79e: spi_flash_chip_generic_write at /spi_flash/spi_flash_chip_generic.c:266

0x4037d361: esp_flash_write at /spi_flash/esp_flash_api.c:907

0x420a2387: esp_partition_write_raw at /spi_flash/partition.c:517

0x420a5be8: nvs::NVSPartition::write_raw(unsigned int, void const*, unsigned int) at /nvs_flash/src/nvs_partition.cpp:50

0x420a67a7: nvs::Page::alterEntryState(unsigned int, nvs::Page::EntryState) at /nvs_flash/src/nvs_page.cpp:752

0x420a701d: nvs::Page::eraseEntryAndSpan(unsigned int) at /nvs_flash/src/nvs_page.cpp:418

0x420a75bd: nvs::Page::eraseItem(unsigned char, nvs::ItemType, char const*, unsigned char, nvs::VerOffset) at /nvs_flash/src/nvs_page.cpp:377

0x420a5155: nvs::Storage::writeItem(unsigned char, nvs::ItemType, char const*, void const*, unsigned int) at /nvs_flash/src/nvs_storage.cpp:376

0x420a5757: nvs::NVSHandleSimple::set_typed_item(nvs::ItemType, char const*, void const*, unsigned int) at /nvs_flash/src/nvs_handle_simple.cpp:29

0x420a3b32: int nvs::NVSHandle::set_item<unsigned long long>(char const*, unsigned long long) at /nvs_flash/include/nvs_handle.hpp:270

0x420a3b87: int nvs_set<unsigned long long>(unsigned int, char const*, unsigned long long) at /nvs_flash/src/nvs_api.cpp:356

0x420a3bf0: nvs_set_u64 at /nvs_flash/src/nvs_api.cpp:396

0x42025a51: pd_nvs_set_clock_start_time(unsigned long long) at /piano/pd_nvs.cpp:748

0x4202d671: pd_time_set_DS1302(unsigned long long, pd_time_source_t) at /piano/pd_time.cpp:331

0x4202cd36: pd_sntp_set_system_time_us(unsigned int, unsigned int) at /piano/pd_sntp.cpp:102

0x4202cd79: sntp_process(sntp_timestamps const*) at /piano/sntp2.cpp:326

0x4202ce0d: sntp_recv(void*, udp_pcb*, pbuf*, ip_addr const*, unsigned short) at /piano/sntp2.cpp:504

0x420c373d: udp_input at /lwip/lwip/src/core/udp.c:404

0x420c8d19: ip4_input at /lwip/lwip/src/core/ipv4/ip4.c:794

0x420ce78e: ethernet_input at /lwip/lwip/src/netif/ethernet.c:186

0x420b89b5: tcpip_thread_handle_msg at /lwip/lwip/src/api/tcpip.c:180

0x420b8a58: tcpip_thread at /lwip/lwip/src/api/tcpip.c:154

0x40387b56: vPortTaskWrapper at /freertos/port/xtensa/port.c:131
```

chipweinberger, 2022-06-09:
@Dazza0 , any thoughts on this?

ESP-Marius, 2022-06-09:
Are you still using `CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=16`? Might be some BT variables that end up in PSRAM even though they shouldn't.

My initial impression is that this isnt related to https://github.com/espressif/esp-idf/issues/9120, this exception happens even if you apply the patch provided in that issue right?

ESP-Marius, 2022-06-09:
Looking at the reg.dump seems to confirm my suspicion.  `A2 : 0x3de5b948` is the first argument passed to `QueueGiveFromISR` (`QueueHandle_t xQueue`), which is a address in external memory. So we get an exception here when this one is accessed while cache is disabled. Probably a malloc somewhere in the BT library that needs to be forced to internal memory.

I'll follow up on this with our BT team.

chipweinberger, 2022-06-09:
> I see from the sdkconfig that it's CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=72 now, are you able to run your program if you put this back to the default value? If yes, does that fix the issue?

I'll test. It will be a few days to test. It only crashes here ~3 times a day during development.

> this exception happens even if you apply the patch provided in that issue right?

Correct. I have the patch applied.

> Probably a malloc somewhere in the BT library that needs to be forced to internal memory.

This seems likely. I forgot to mention it crashes most often *after* I do some BT activity.

ESP-Marius, 2022-06-09:
The reason why the crash is rare is probably that it requires the BT interrupt to fire and access the PSRAM while NVS is writing to flash. 

Thanks for all the valuable feedback!

chipweinberger, 2022-07-11:
Bump. I am still routinely hitting this at random times. Really critical to have this fixed. Any updates?

@ESP-Marius @igrr

ESP-Marius, 2022-07-12:
@chipweinberger Sorry for the delay

I'll check with out BT team again to see if there is any progress, and remind them that this issue still needs to be addressed.

blueMoods, 2022-07-12:
@chipweinberger  Thanks very mush for you help.

The reason of this issues is not at nvs_set_blob(), the real reason is that: we try to access cache in interrupt, but cache is disabled. And nvs_set_blob() will write NVS and disable cache.

I have find some issues will cause assceeing cache in interrupt: queue, sempher and mutex will be placed into PSRAM if CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL is too small. I have fixed them in IDF, here is the patch for release/v4.4 
[fix_v4.4.zip](https://github.com/espressif/esp-idf/files/9089175/fix_v4.4.zip)
. And we are merging it to esp-idf.

Please make sure the queue, sempher and mutex you used in interrupt is mallocd in IRAM, you can refer to the function queue_create_wrapper, semphr_create_wrapperand mutex_create_wrapper in bt.c

If this issue re-produce after changes, please save the log and .elf file and send to me. thanks.

blueMoods, 2022-07-12:
@chipweinberger Thanks very mush for you help.

The reason of this issues is not at nvs_set_blob(), the real reason is that: we try to access cache in interrupt, but cache is disabled. And nvs_set_blob() will write NVS and disable cache.

I have find some issues will cause assceeing cache in interrupt: queue, sempher and mutex will be placed into PSRAM if CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL is too small. I have fixed them in IDF, here is the patch for release/v4.4
[fix_v4.4.zip](https://github.com/espressif/esp-idf/files/9089310/fix_v4.4.zip)

. And we are merging it to esp-idf.

Please make sure the queue, sempher and mutex you used in interrupt is mallocd in IRAM, you can refer to the function queue_create_wrapper, semphr_create_wrapperand mutex_create_wrapper in bt.c

If this issue re-produce after changes, please save the log and .elf file and send to me. thanks.

chipweinberger, 2022-07-12:
Thank you for the patch. I will try it shortly. 

I think it is better solution to change `xQueueCreate`/`xSemaphoreCreate` to always use internal memory.

- safer, fewer changes, lower complexity
- semaphores should probably use fast memory for performance, etc
- probably not much more memory, I think

@igrr, would we be able to make that change?

chipweinberger, 2022-07-12:
@blueMoods 

I have tried the patch. It is difficult to verify if https://github.com/espressif/esp-idf/issues/9129 (this issue) is fixed because it did not happen very often.

However, I can confirm that https://github.com/espressif/esp-idf/issues/9332#issuecomment-1181243560 still happens, which is likely a similar issue but perhaps unrelated to Bluetooth.

Also, sadly, I can confirm that settings `CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL = 92 = sizeof(StaticQueue_t)` also does not fix https://github.com/espressif/esp-idf/issues/9332#issuecomment-1181243560

blueMoods, 2022-07-12:
#9332  is different with this issue,   the reason of #9332 is the string in your logs is in  flash. you can not get it when cache disable.  Please try https://github.com/espressif/esp-idf/issues/9332#issuecomment-1181354144

closed 2022-07-29 by espressif-bot

AxelLin, 2022-08-05:
@ESP-Marius  Just remind v4.4.2 does not include this fix.

ESP-Marius, 2022-08-05:
> @ESP-Marius Just remind v4.4.2 does not include this fix.

The fix has been merged into release/v4.4, but it seems like it didnt make it in time for the v4.4.2 test build.