espressif/esp-idf #9130: Suspected miscompilation on IDF v4.2.3

jimmo, 2022-06-09:
## Environment

- Development Kit:     TinyPico
- Module or chip used:  ESP32-PICO-D4
- IDF version: v4.2.3
- Build System:         idf.py (inside MicroPython build)
- Compiler version:
  xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
  GNU assembler (crosstool-NG esp-2020r3) 2.31.1
- Operating System:    Linux
- Using an IDE?: No
- Power Supply:  USB

## Problem Description

I'm from the MicroPython project.

We noticed that one of our recent commits broke some basic parsing functionality with the way constant folding works in our compiler, but only on ESP32, and only with IDF v4.2 (works fine in 4.3 and 4.4). Any modifications (e.g. debugging, changing of inlining, etc) to this code caused the bug to go away.

The only diff for this function compiled on 4.2 vs 4.4 is:

```diff
  d3:   1aa1c2          movi    a12, 0x11a
  d6:   209770          or      a9, a7, a7     
  d9:   178976          loop    a9, f4 <push_result_rule+0xf4>
  dc:   030190          rsr.lend        a9
  df:   130090          wsr.lbeg        a9
> e2:   a8c992          addi    a9, a9, -88   (4.2, broken)
> e5:   06d992          addmi   a9, a9, 0x600
< e2:   000091          l32r    a9, fffc00e4 <push_result_rule+0xfffc00e4>   (4.4, working)
< e5:   0020f0          nop
  e8:   130190          wsr.lend        a9
  eb:   002000          isync
  ee:   030290          rsr.lcount      a9
  f1:   01c992          addi    a9, a9, 1
  f4:   1494e7          bne     a4, a14, 10c <push_result_rule+0x10c>
```

The 4.4 version references the constant `18:   9c 06 00 00`

Both IDF versions seem to use the same compiler version (8.4.0) and just substituting the gcc binary didn't solve the problem. However, substituting the assembler did. This led me to https://github.com/jcmvbkbc/binutils-gdb-xtensa/commit/403b0b61f6d4358aee8493cb1d11814e368942c9 @jcmvbkbc which was applied in binutils 2.33 (v4.2 has binutils 2.31, v4.4 has binutils 2.35).

We would like to move to v4.4 as our minimum supported IDF version, but that is not feasible right now as it causes other issues due to the differences in memory layout resulting in a smaller contiguous area available to create our garbage-collected heap. We are working on solving that with a split heat but that isn't ready yet.

Right now we have a "workaround" (by disabling inlining for a section of the relevant code) but this is fragile. I understand that v4.2 is maintained until mid 2023, so would you consider doing a new v4.2.x release with an updated binutils?

FYI, I did a search in our firmware image for other patterns of this addi/addmi code, and also found one other occurrence in `llm_hci_cmd_handler_tab_reset`. This appears to come from a binary blob (perhaps hasn't been recompiled with newer gcc/binutils?).

While investigating this I also came across https://github.com/espressif/esp-idf/commit/ee2f8b1a6236fcdcd57f8410d98119d96de40a3f -- which is not in v4.2. We do put a significant amount performance-sensitive code in IRAM, so although I don't have any evidence that we're being affected by this, it's still a bit concerning. Perhaps worth backporting to v4.2 also?

Also worth mentioning, I haven't checked IDF v4.1, but I suspect much of the above applies to v4.1 too.

Thanks! Please let me know if you need additional details.

cc @dpgeorge

jcmvbkbc, 2022-06-09:
If you decide to backport the commit https://github.com/jcmvbkbc/binutils-gdb-xtensa/commit/403b0b61f6d4358aee8493cb1d11814e368942c9 please note that the commit https://github.com/jcmvbkbc/binutils-gdb-xtensa/commit/548791769dc737f05cb12e5ee4190b7e853beac9 is related to this issue and should be backported as well.

chipweinberger, 2022-06-09:
This is some impressive investigation you did!

AxelLin, 2023-05-07:
> If you decide to backport the commit [jcmvbkbc/binutils-gdb-xtensa@403b0b6](https://github.com/jcmvbkbc/binutils-gdb-xtensa/commit/403b0b61f6d4358aee8493cb1d11814e368942c9) please note that the commit [jcmvbkbc/binutils-gdb-xtensa@5487917](https://github.com/jcmvbkbc/binutils-gdb-xtensa/commit/548791769dc737f05cb12e5ee4190b7e853beac9) is related to this issue and should be backported as well.

CC @igrr

AxelLin, 2023-12-10:
v4.2 branch is already EOL, this should be closed.

jimmo, 2023-12-11:
Indeed. We have moved to 5.0 (and soon 5.1) so this is no longer an issue for us.

closed 2023-12-11 by jimmo