espressif/esp-idf #9131: Guru Meditation Error trigger by nvs_flash_init when activated SystemView Tracing

nikidimitrow, 2022-06-09:
Hello support team,

## Environment
- Development Kit:   **ESP32-DevKitC**
- Kit version: **v4**
- Module or chip used:  **ESP32-S3-WROOM-1**
- IDF version: **v5.0-dev-3290-g01d014c42d**
- Build System:  **idf.py**
- Compiler version: **xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2021r2-patch3) 8.4.0** 
- Operating System: **Windows**
- (Windows only) environment type: **ESP Command Prompt**
- Using an IDE: **No**
- Power Supply: **USB**

## Problem Description

In our project we would like to do system Behavior Analysis with SEGGER SystemView. For this purpose from idf.py menuconfig I'm activated:

Component config --> Application Level Tracing --> Data Destomatopm 1 --> (JTAG) 
FreeRTOS SystemView Tracing --> SystemView Tracing Enable.

Before activated this option in idf.py menuconfig the system is working fine.

Once the above options in menuconfig  are save and the project is build  and flash system start to enter in the critical error:

_Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)._

From the trace I can see that nvs_flash_init is trigger this panic error.

See the log file below. Sdconfig file and .elf file are upload here: https://github.com/nikidimitrow/idf-report-crash

```
I (48) boot: ESP-IDF v5.0-dev-3290-g01d014c42d 2nd stage bootloader
I (48) boot: compile time 10:59:45
I (48) boot: chip revision: 0
I (51) boot.esp32s3: Boot SPI Speed : 80MHz
I (56) boot.esp32s3: SPI Mode       : DIO
I (61) boot.esp32s3: SPI Flash Size : 8MB
I (66) boot: Enabling RNG early entropy source...
I (71) boot: Partition Table:
I (75) boot: ## Label            Usage          Type ST Offset   Length
I (82) boot:  0 otadata          OTA data         01 00 0000d000 00002000
I (89) boot:  1 prod             WiFi data        01 02 0000f000 00001000
I (97) boot:  2 ota_0            OTA app          00 10 00010000 00280000
I (104) boot:  3 ota_1            OTA app          00 11 00290000 00280000
I (112) boot:  4 safe             WiFi data        01 02 00610000 00003000
I (119) boot:  5 config           WiFi data        01 02 00613000 00046000
I (127) boot:  6 appl             WiFi data        01 02 00659000 00004000
I (135) boot:  7 shared           WiFi data        01 02 0065d000 00003000
W (142) efuse: Loading virtual efuse blocks from flash
EFUSE_BLKx:
0) 0x01800301 0x000f0c02 0x49c40000 0x00500000 0x80000022 0x00000000
1) 0x03c0296c 0x000084f7 0x00000000 0x02040000 0x54991400 0x5005d47c
2) 0xfbecd701 0x24072709 0x50b01efb 0xcc734f00 0x70cb5271 0xd0d12edb 0xbed0d4e1 0x324cbcc8
3) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4) 0x7e68a2e3 0x08fe1124 0x53ed80b5 0xa7d5faa7 0x1b37941e 0x0b3d3af1 0x7e3e1a92 0x7dee2768
5) 0x0c115a96 0xe46eab50 0x87476dcd 0xc4ee6f86 0xa09cab43 0x8b2b625e 0xeabfd528 0x125e8ae2
6) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
7) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
8) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
9) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
10) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

I (234) boot:  8 efuse_em         efuse            01 05 00660000 00002000
I (242) boot:  9 nvs              WiFi data        01 02 00662000 00004000
I (249) boot: 10 nvs_key          NVS keys         01 04 00666000 00001000
I (257) boot: 11 spiffs           Unknown data     01 82 00667000 00199000
I (265) boot: End of partition table
I (269) boot: No factory image, trying OTA 0
[boot] Custom bootloader message defined in the KConfig file.
[... 121 lines cut ...]

0x4200a142: initServices() at C:/Users/xxxxxxx/Documents/ksc-2n/src/OS/os_wrapper/cfg/TaskCfg.cpp:81

0x4200a00b: app_main at C:/Users/xxxxxxx/Documents/ksc-2n/main/main.cpp:6

0x420d6de0: main_task at C:/Users/xxxxxxx/Documents/esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:124 (discriminator 2)

0x40381b21: vPortTaskWrapper at C:/Users/xxxxxxx/Documents/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:141

ELF file SHA256: 2e568e4c6cac14f4

Rebooting...
�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375b90
0x40375b90: esp_restart_noos at C:/Users/xxxxxxx/Documents/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:151 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd03e0,len:0x39b0
load:0x403b6000,len:0xbb8
load:0x403b9000,len:0x5184
entry 0x403b6260
```

chipweinberger, 2022-06-09:
This might be the same issue as: https://github.com/espressif/esp-idf/issues/9129

Are you using SPIRAM? Can you attach your sdkconfig? (you can upload it as a *.txt file)

nikidimitrow, 2022-06-09:
Hello @chipweinberger, 

thank for your reply.

> This is the same error code as: #9129
> 
> Are you using SPIRAM? Can you attach your sdkconfig? (you can upload it as a *.txt file)

Here is it:

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/8867437/sdkconfig.txt)

nikidimitrow, 2022-06-09:
> Thanks for reporting this. The solution is probably to add `app_trace_membufs_proto (noflash)` next to this line:
> 
> https://github.com/espressif/esp-idf/blob/01d014c42def8d0c19e1ce55c07de6761e092ffa/components/app_trace/linker.lf#L4

Hi @igrr, thank you for your reply and suggestion.

If I add the _app_trace_membufs_proto (noflash) next to app_trace (noflash) like this:

_[mapping:app_trace]
archive: libapp_trace.a
entries:
    app_trace (noflash) app_trace_membufs_proto (noflash)_

Linking phase of the project is failing due to following error in app_trace/linker.lf file:

`ERROR: failed to parse C:/Users/uie49957/Documents/esp-idf/components/app_trace/linker.lf
Expected end of text, found '['  (at char 0), (line:1, col:1)`

If I add line of code after app_trace (noflash) like this:

```
[mapping:app_trace]
archive: libapp_trace.a
entries:
    app_trace (noflash) 
    app_trace_membufs_proto (noflash)
    port_uart (noflash)
```
Still face the same error.

If I remove app_trace (noflash)  and add app_trace_membufs_proto (noflash) - again get panic error.

Appreciated your support.

nikidimitrow, 2022-06-10:
Hello support team,

any suggestion for the above mention problem?

nikidimitrow, 2022-06-22:
Hi support team,

what I can provide you more for some hints from your side to make NvS and SystemView working together?

ESP-Marius, 2022-06-23:
@nikidimitrow Sorry for the slow feedback!

Something like this should work:

```
entries:
    app_trace (noflash)
    app_trace_membufs_proto (noflash)
    port_uart (noflash)
    port (noflash)
    app_trace_util (noflash)
```
Added both `port` and `app_trace_membufs_proto`, seemed to fix it for me locally at least.

nikidimitrow, 2022-06-24:
> @nikidimitrow Sorry for the slow feedback!
>
> Something like this should work:
>
> ```
> entries:
>     app_trace (noflash)
>     app_trace_membufs_proto (noflash)
>     port_uart (noflash)
>     port [...]

@ESP-Marius [ESP-Marius](https://github.com/ESP-Marius) appreciated your support. The propose solution resolved the issue between NvS and SystemView.

Thank you and have a cold beers at the evening.

ESP-Marius, 2022-06-24:
@nikidimitrow Great to hear that it fixed it. 

If you don't mind i'll keep this issue open until we merge a fix for this. That way if anyone else encounters this issue then they'll be able to see this a still an unsolved problem.

KaeLL, 2022-06-24:
@ESP-Marius ETA?

ESP-Marius, 2022-06-27:
> @ESP-Marius ETA?

The MR will probably be merged today, but our syncs to Github haven't been that frequent lately (they depend on internal tests all passing), so I unfortunately can't give you an ETA for that.

closed 2022-07-07 by espressif-bot

leeebo, 2023-10-16:
@ESP-Marius The bug can be reproduced when using UART as the Systemview destination (The UART Driver does not support the configuration to IRAM)

```
I (307) sleep: Configure to isolate all GPIO pins in sleep state
I (314) sleep: Enable automatic switching of GPIO sleep configuration
I (321) esp_apptrace: Initialized TRAX on CPU0
I (326) esp_apptrace_uart: UART baud rate: 1000000
I (332) esp_apptrace_uart: Initialized UART on CPU0
I (337) app_start: Starting scheduler on CPU0
I (342) main_task: Started on CPU0
I (346) main_task: Calling app_main()
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). 

Core  0 register dump:
PC      : 0x4200c28a  PS      : 0x00060d34  A0      : 0x80379591  A1      : 0x3fc95040  
0x4200c28a: uart_get_buffered_data_len at /home/libo/esp/github_master/components/driver/uart/uart.c:1393

A2      : 0x3fc97cbc  A3      : 0x3fc95088  A4      : 0x00000000  A5      : 0x3fc9e950  
A6      : 0x403829c8  A7      : 0x3fca0250  A8      : 0x8037a0e9  A9      : 0x3fc95020  
0x403829c8: spi_flash_chip_generic_read at /home/libo/esp/github_master/components/spi_flash/spi_flash_chip_generic.c:226

A10     : 0x3fc95040  A11     : 0x3fc95040  A12     : 0x40382a35  A13     : 0x3fc95170  
0x40382a35: spi_flash_chip_generic_read at /home/libo/esp/github_master/components/spi_flash/spi_flash_chip_generic.c:246

A14     : 0x0000ccc0  A15     : 0x3fc948b8  SAR     : 0x00000019  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  
0x400570e8: memset in ROM

0x400570f3: memset in ROM

Core  0 was running in ISR context:
EPC1    : 0x4201f6c7  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4200c28a
0x4201f6c7: uart_hal_write_txfifo at /home/libo/esp/github_master/components/hal/uart_hal_iram.c:27

0x4200c28a: uart_get_buffered_data_len at /home/libo/esp/github_master/components/driver/uart/uart.c:1393

```

erhankur, 2023-10-16:
@leeebo can you attach your sdkconfig file or sample app to help me to reproduce the issue?

leeebo, 2023-10-16:
> @leeebo can you attach your sdkconfig file or sample app to help me to reproduce the issue?

Please try [sysview_tracing.zip](https://github.com/espressif/esp-idf/files/12915277/sysview_tracing.zip)