espressif/esp-idf #9132: How to get all Software Versions

franz-ms-muc, 2022-06-09:
Hello, 

how to get Version Name etc.. from

- running Partition (that is easy, just look in startup.c)
- the other OTA Partition
- the Test Firmware
- the Factory Partition
- the Bootloader

also can one add some Fields in the code which are "reserved" spaces in the bin File, which can be set by a script, and later read by the code ? 

Thanks,
Franz

AshUK, 2022-06-13:
+1 would also like to the be able to report the boot loader version to the cloud.

igrr, 2022-06-13:
@franz-ms-muc There is an open feature request for getting the IDF version used to compile the bootloader: https://github.com/espressif/esp-idf/issues/8800. Please follow that ticket for updates.

For the rest, @KaeLL is right, `esp_ota_get_partition_description` should give you the structure with the required information.

> also can one add some Fields in the code which are "reserved" spaces in the bin File, which can be set by a script, and later read by the code ?

If you add these fields during the compilation process (via rodata_custom_desc section), then yes. Overwriting parts of the binary after it is created isn't supported, because the hash of the binary image will become incorrect.

closed 2023-05-21 by espressif-bot