espressif/esp-idf #9134: Symbol 'STATE_IDLE' could not be resolved

vinothkrstatiq, 2022-06-09:
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/esp_ble_mesh/ble_mesh_wifi_coexist/main/main.c

JiaLWang, 2022-06-10:
Hi，

Could you please provide your error log and sdkconfig file?

vinothkrstatiq, 2022-06-10:
error log
Symbol 'STATE_IDLE' could not be resolved
sdkconfig file below
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET_ARCH_XTENSA=y
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_TARGET_ESP32=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32-elf-"
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager

#
# Bootloader config
#
CONFIG_BOOTLOADER_OFFSET_IN_FLASH=0x1000
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
CONFIG_BOOTLOADER_LOG_LEVEL_INFO=y
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=3
# CONFIG_BOOTLOADER_SPI_CUSTOM_WP_PIN is not set
CONFIG_BOOTLOADER_SPI_WP_PIN=7
# CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V is not set
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
# CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_ON_POWER_ON is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_ALWAYS is not set
CONFIG_BOOTLOADER_RESERVE_RTC_SIZE=0
# CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC is not set
[... 1862 lines cut ...]
# CONFIG_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=64
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
# End of deprecated options

JiaLWang, 2022-06-22:
Hi,
Sorry for the late reply.

I can't reproduce it. Can you provide detailed steps to reproduce it？ And your compiling environment.
I'm not certain if it's a  environment problem.

vinothkrstatiq, 2022-06-23:
I am using Espressif-IDE Version: 2.5.0
File->New->Espressif IDF Project
After creating project, I just copy pasted the  below code 
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/esp_ble_mesh/ble_mesh_wifi_coexist/main/main.c
**I am getting below error** 
[1004/1011] cmd.exe /C "cd . && C:\Espressif\tools\cmake\3.20.3\bin\cmake.exe -E rm -f esp-idf\wifi_provisioning\libwifi_provisioning.a && C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-ar.exe qc esp-idf\wifi_provisioning\libwifi_provisioning.a  esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/src/wifi_config.c.obj esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/src/wifi_scan.c.obj esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/src/manager.c.obj esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/src/handlers.c.obj esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/src/scheme_console.c.obj esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/proto-c/wifi_config.pb-c.c.obj esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/proto-c/wifi_scan.pb-c.c.obj esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/proto-c/wifi_constants.pb-c.c.obj esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/src/scheme_softap.c.obj && C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-ranlib.exe esp-idf\wifi_provisioning\libwifi_provisioning.a && cd ."
[1005/1011] ccache C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -IC:/Espressif/frameworks/esp-idf-v4.4/components/newlib/platform_include -IC:/Espressif/frameworks/esp-idf-v4.4/components/freertos/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/freertos/include/esp_additions/freertos -IC:/Espressif/frameworks/esp-idf-v4.4/components/freertos/port/xtensa/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/freertos/include/esp_additions -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_hw_support/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_hw_support/include/soc -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_hw_support/include/soc/esp32 -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_hw_support/port/esp32/. -IC:/Espressif/frameworks/esp-idf-v4.4/components/heap/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/log/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/lwip/include/apps -IC:/Espressif/frameworks/esp-idf-v4.4/components/lwip/include/apps/sntp -IC:/Espressif/frameworks/esp-idf-v4.4/components/lwip/lwip/src/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/lwip/port/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/lwip/port/esp32/include/arch -IC:/Espressif/frameworks/esp-idf-v4.4/components/soc/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/soc/esp32/. -IC:/Espressif/frameworks/esp-idf-v4.4/components/soc/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/hal/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/hal/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/hal/platform_port/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_rom/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_rom/include/esp32 -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_rom/esp32 -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_common/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_system/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_system/port/soc -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_system/port/public_compat -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/xtensa/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/xtensa/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/driver/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/driver/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_pm/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_ringbuf/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/efuse/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/efuse/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/vfs/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_wifi/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_event/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_netif/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_eth/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/tcpip_adapter/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_phy/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_phy/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_ipc/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/app_trace/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_timer/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/mbedtls/port/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/mbedtls/mbedtls/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/mbedtls/esp_crt_bundle/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/app_upda [... cut ...]
Build complete (1 errors, 0 warnings): C:\Users\statiq32\Desktop\esp32\bleidf\build
C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\Espressif\frameworks\esp-idf-v4.4\tools\idf_size.py C:/Users/statiq32/Desktop/esp32/bleidf/build/bleidf.map
C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v4.4/\components\partition_table\gen_esp32part.py C:\Users\statiq32\Desktop\esp32\bleidf\build\partition_table\partition-table.bin
Total time taken to build the project: 141,458 ms

JiaLWang, 2022-06-23:
Hi,
Are you using eclipse？ If yes, you can refer to the README. https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md

If you want to use the `ble_mesh_wifi_coexist` example, you can refer to `Create a new project using ESP-IDF Templates`.

JiaLWang, 2022-06-23:
OK, thanks for your info.

I will retry with this tool. And I will update ASAP.

vinothkrstatiq, 2022-06-24:
I am using Espressif-IDE Version: 2.5.0
Operating System: windows 10
Java Runtime Version: 11.0.14.1+10-LTS
Eclipse Version: 4.22.0.v20211124-1800
Eclipse CDT Version: 10.5.0.202109270023
IDF Eclipse Plugin Version: 2.5.0.202206080313
ESP-IDF v4.4-dirty
Python set for IDF_PYTHON_ENV: Python 3.8.7
Python declared in PATH variable: Python 3.10.4