espressif/esp-idf #9136: [protobuf-c] Move to 1.4.0 from 1.3.0

KonssnoK, 2022-06-10:
Protobuf-C 1.3.0 is from 2017,
since then there were multiple fixes to the library, which reached 1.4.0 in 2021.

Please update the referenced module commit !

Thanks.

mahavirj, 2022-06-10:
@KonssnoK This change is already available, please check https://github.com/espressif/esp-idf/commit/b09b716f1df13dabb15e911b181ab8e82aa2d25f. We shall backport to previous release branches till v4.3.

KonssnoK, 2022-06-10:
cool, we can close when it's backported to v4.4 then

mahavirj, 2022-06-23:
Fix for v4.4 is available at https://github.com/espressif/esp-idf/commit/0b05f1708dd8ce3ba984b7bd890a8bd47a1e485d. Closing this issue.

closed 2022-06-23 by mahavirj

AxelLin, 2022-09-14:
Hi @mahavirj 
FYI, protobuf-c 1.4.1 is released which includes below fixes:
https://github.com/protobuf-c/protobuf-c/releases/tag/v1.4.1

mahavirj, 2022-09-14:
@AxelLin Thanks for notifying. We will evaluate and then prepare for the update.

AxelLin, 2022-09-25:
> @AxelLin Thanks for notifying. We will evaluate and then prepare for the update.

It's a bugfix release, so I think you should upgrade it.

mahavirj, 2022-10-06:
@AxelLin 

> It's a bugfix release, so I think you should upgrade it.

This change has been merged internally. It shall appear to GH with the next sync.

AxelLin, 2023-12-25:
Hi @mahavirj
FYI, protobuf-c 1.5.0 is released: https://github.com/protobuf-c/protobuf-c/releases/tag/v1.5.0

AxelLin, 2025-02-02:
Hi @mahavirj 
FYI, protobuf-c 1.5.1 is released.
https://github.com/protobuf-c/protobuf-c/releases/tag/v1.5.1