espressif/esp-idf #9140: Bluetooth modem sleep with external 32.768kHz xtal not working on ESP32-PICO-MINI-02

karelchik, 2022-06-11:
## Environment

- Development Kit: none
- Module or chip used: ESP32-PICO-MINI-02
- IDF version (run ``git describe --tags`` to find it): v4.4.1
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System: Windows
- (Windows only) environment type: Plain Command Prompt
- Using an IDE?: no
- Power Supply: USB

## Problem Description

Using Blehr example (`esp-idf-v4.4.1\examples\bluetooth\nimble\blehr`), I added automatic light sleep according to https://github.com/espressif/esp-idf/issues/947#issuecomment-500312453:
```
#include "esp_err.h"
#include "esp_pm.h"
...
void app_main(void)
{
    esp_pm_config_esp32_t pm_config = {
        .max_freq_mhz = 80,
        .min_freq_mhz = 40,
        .light_sleep_enable = true,
    };
    ESP_ERROR_CHECK( esp_pm_configure(&pm_config) );
...
}
```
But I can't get this to work on my ESP32-PICO-MINI-02 based DIY breakout. I have got two types of output:
1) Connection works, but light sleep does not apply: measured 28-30 mA current consumption. (Also I changed `ESP_EARLY_LOGD` to `ESP_EARLY_LOGW` in `esp-idf-v4.4.1\components\esp_system\port\soc\esp32` to see the `RTC_SLOW_CLK calibration value`)
``` 
entry 0x40080694
I (28) boot: ESP-IDF v4.4.1-dirty 2nd stage bootloader
I (29) boot: compile time 18:34:01
I (29) boot: chip revision: 3
I (31) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 8MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 3, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0ed8ch ( 60812) map
I (133) esp_image: segment 1: paddr=0001edb4 vaddr=3ffbdb60 size=01264h (  4708) load
I (135) esp_image: segment 2: paddr=00020020 vaddr=400d0020 size=52d04h (339204) map
I (261) esp_image: segment 3: paddr=00072d2c vaddr=3ffbedc4 size=035e8h ( 13800) load
I (267) esp_image: segment 4: paddr=0007631c vaddr=40080000 size=186d4h (100052) load
I (308) esp_image: segment 5: paddr=0008e9f8 vaddr=50000000 size=00010h (    16) load
I (321) boot: Loaded app from partition at offset 0x10000
I (321) boot: Disabling RNG early entropy source...
I (332) cpu_start: Pro cpu up.
I (333) cpu_start: Starting app cpu, entry point is 0x4008138c
0x4008138c: call_start_cpu1 at C:/Users/kaare/Documents/esp-idf-v4.4.1/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
W (1674) clk: RTC_SLOW_CLK calibration value: 10518517
I (1680) cpu_start: Pro cpu start user code
I (1680) cpu_start: cpu freq: 160000000
I (1680) cpu_start: Application information:
I (1685) cpu_start: Project name:     blehr
I (1690) cpu_start: App version:      1
I (1694) cpu_start: Compile time:     Jun 11 2022 18:33:41
I (1701) cpu_start: ELF file SHA256:  3e345af420a48386...
I (1707) cpu_start: ESP-IDF:          v4.4.1-dirty
I (1712) heap_init: Initializing. RAM available for dynamic allocation:
I (1719) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1726) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1732) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1738) heap_init: At 3FFC4FE0 len 0001B020 (108 KiB): DRAM
I (1744) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1751) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1757) heap_init: At 400986D4 len 0000792C (30 KiB): IRAM
I (1765) spi_flash: detected chip: generic
I (1768) spi_flash: flash io: dio
I (1784) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1785) pm: Frequency switching config: CPU_MAX: 80, APB_MAX: 80, APB_MIN: 40, Light sleep: ENABLED
I (1815) BTDM_INIT: BT controller compile version [5688ed5]
W (1816) BTDM_INIT: 32.768kHz XTAL not detected, fall back to main XTAL as Bluetooth sleep clock
light sleep mode will not be able to apply when bluetooth is enabled
I (1827) system_api: Base MAC address is not set
I (1832) system_api: read default base MAC address from EFUSE
I (1841) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (2192) NimBLE_BLE_HeartRate: BLE Host Task Started
I (2239) NimBLE: Device Address:
I (2239) NimBLE: 30:83:98:d1:14:3a
I (2240) NimBLE:

I (2247) NimBLE: GAP procedure initiated: advertise;
I (2248) NimBLE: disc_mode=2
I (2248) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
I (2258) NimBLE:

I (48495) NimBLE: connection established; status=0
```
2) A crashing loop, does not even make it to advertising state.
```
entry 0x40080694
I (28) boot: ESP-IDF v4.4.1-dirty 2nd stage bootloader
I (29) boot: compile time 18:34:01
I (29) boot: chip revision: 3
I (31) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 8MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 3, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0ed8ch ( 60812) map
I (133) esp_image: segment 1: paddr=0001edb4 vaddr=3ffbdb60 size=01264h (  4708) load
I (135) esp_image: segment 2: paddr=00020020 vaddr=400d0020 size=52d04h (339204) map
I (261) esp_image: segment 3: paddr=00072d2c vaddr=3ffbedc4 size=035e8h ( 13800) load
I (267) esp_image: segment 4: paddr=0007631c vaddr=40080000 size=186d4h (100052) load
I (308) esp_image: segment 5: paddr=0008e9f8 vaddr=50000000 size=00010h (    16) load
I (321) boot: Loaded app from partition at offset 0x10000
I (321) boot: Disabling RNG early entropy source...
I (332) cpu_start: Pro cpu up.
I (333) cpu_start: Starting app cpu, entry point is 0x4008138c
0x4008138c: call_start_cpu1 at C:/Users/kaare/Documents/esp-idf-v4.4.1/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6996
load:0x40078000,len:16616
load:0x40080400,len:3808
0x40080400: _init at ??:?
```

I have built the exact same breakout for ESP32-WROOM-32D and I have no issues with that: using nRF Connect App for testing I measured 12-13 mA current consumption. So the 32k xtal is working there. Unfortunately the ESP32-WROOM-32D is too large for my application, so I need to get something smaller to work.

### Expected Behavior

The Blehr example works on ESP32-PICO-MINI-02 and consumes below 15 mA of current.

### Actual Behavior

On ESP32-PICO-MINI-02 the Blehr example: 
1) fails to apply light sleep, but otherwise works and consumes around 30 mA of current.
or
2) does not work at all (crashes continuously)

## Other items if possible

Sdkconfig and elf file are attached [blehr_issue.zip](https://github.com/espressif/esp-idf/files/8884371/blehr_issue.zip)
.

xiongweichao, 2022-06-13:
Hi @karelchik ,

1. Because there is no external 32K crystal, light sleep will not be used.
`W (1816) BTDM_INIT: 32.768kHz XTAL not detected, fall back to main XTAL as Bluetooth sleep clock
light sleep mode will not be able to apply when bluetooth is enabled`

2. The number of cycles for the RTC_SLOW_CLK calibration is too large, causing the watchdog to time out.

Thanks

xiongweichao, 2022-06-14:
Hi @karelchik ,

1. I set ESP32_RTC_CLK_CAL_CYCLES to 5000 and tested it, light sleep is no problem.

2. I haven't tested it with v4.2.3.

3. You can try to use this [API](https://github.com/espressif/esp-idf/blob/20f5e180eecccfaff815d707e1fcbba2f4d6a391/components/soc/esp32/include/soc/rtc.h#L278) to determine if there is an external 32K crystal.

Thanks

xiongweichao, 2022-06-15:
Hi @karelchik ，

Sorry, should use this [API](https://github.com/espressif/esp-idf/blob/20f5e180eecccfaff815d707e1fcbba2f4d6a391/components/soc/esp32/include/soc/rtc.h#L252). Sleep uses RTC slow clock.

I use ESP32-WROOM-32E.

Thanks

karelchik, 2022-06-15:
Hi @xiongweichao,

Using `printf("%i\n", rtc_clk_slow_freq_get());` I get `1` for all my breakouts (ESP32-PICO-MINI-02 as well as ESP32-WROOM-32D), which corresponds to 32 kHz xtal being used. (Breakout A sometimes fails to detect the xtal and gives `0`, but this is no news as it then also sends the message `W (1368) BTDM_INIT: 32.768kHz XTAL not detected, fall back to main XTAL as Bluetooth sleep clock
light sleep mode will not be able to apply when bluetooth is enabled`)

I'm not sure why You are using ESP32-WROOM-32E for testing, because I think the issue is related to ESP32-PICO-MINI-02. I have no issues with the same code on ESP32-WROOM-32D using the same schematic and components.

For me it would be really important te get this ESP32-PICO-MINI-02 working, or if someone could share the exact hardware setup and code for getting another smaller ESP module working with ble + 32 kHz xtal modem sleep. By smaller modules I mean something like ESP32-C3-MINI-1, ESP32-MINI-1.

Thanks

xiongweichao, 2022-06-23:
Hi @karelchik ,

Sorry, I don't have an ESP32-PICO-MINI-02. I have the ESP32C3-MINI-1 and have tested it with no issues.

Thanks

AxelLin, 2022-06-23:
> Hi @karelchik ,
> I have the ESP32C3-MINI-1 and have tested it with no issues.

@xiongweichao 
I thought ble modem sleep and light sleep is only supported in master tree fd90341138e6cfd2980fc03b85bb7fedd21e6dc2 now? Does it really work in v4.x branches? (As this issue was reported against v4.4.1).

xiongweichao, 2022-06-23:
Hi @AxelLin ,

Sorry, I don't quite understand what you mean, you mean ESP32C2? ESP32/ESP32C3/ESP32S3 supports ble modem sleep and light sleep in v4.4.1.

Thanks

AxelLin, 2022-06-23:
Oops, forget it. I misread the commit.

bakgab, 2022-08-01:
Hi @karelchik ,

It seems that I am having the exact same problem. I have two development boards, one with ESP32-WROOM, and the other is ESP32-PICO-MINI-02. I try to reduce power consumption with Bluetooth, and I did the same (try go to light sleep with external 32k crystal, which I soldered on the 32-33 pins by hand). 
I have the same error: "32 kHz XTAL not found", when uploading the code on the MINI-02. But when I upload it to the WROOM board, it runs fine. 

Did you mange to find the solution for this problem? I would be very grateful if you could write it.
(The IDF version I am using: 4.4)

Thanks

karelchik, 2022-08-01:
Hi @bakgab,

I managed to bypass the problem, but I think you might not like the answer: I switched to ESP32-C3-MINI-1 instead. There the 32k crystal is working perfectly (I'm using v4.4.1). 

But before the final design I also built some quick breakouts for crystal testing using ESP32-C3-MINI-1 and ESP32-PICO-MINI-02 as well to further investigate the issue. On those breakouts the crystal worked with both modules. However for the final design I decided going with ESP32-C3-MINI-1, because I found some breakouts using this module and featuring the crystal: https://www.espressif.com/sites/default/files/user-guide-aws-expresslink-devkit-en.pdf?v=1.0.

To this day it really perplexes me that the crystal did not work on my first ESP32-PICO-MINI-02 breakout, but did work on the second one. There were only a few minor differences: the first one used flex pcb, second was rigid. Also, on the second one the crystal was a bit further away from the module (about 7mm from the pins) and on the first it was about 4mm. Because of the small distance one lead to the crystal is about 2x shorter than the other, but I don't know if this can cause an issue. I am running out of ideas guessing why the crystal failed on my first breakout. I also built a few of them so contacts were very unlikely to be the problem. I mainly suspect some sort of issue regarding ESP32-PICO-MINI-02 module.

One last thing to make this story more complete for any readers out there. The final breakout (on which the crystal is working with ESP32-C3-MINI-1) is also flex and very similar to the first breakout (on which the crystal isn't working with ESP32-PICO-MINI-02).

xiongweichao, 2023-06-14:
Hi @karelchik and @bakgab ,

May I ask if there is a resistor connected in parallel?
[image]

bakgab, 2023-06-14:
> Hi @karelchik and @bakgab ,
> 
> May I ask if there is a resistor connected in parallel? [image]

Hi @xiongweichao ,
Yes, I used a resistor in parallel with a value of 5.1M Ohm, it worked, but I did not test what would happen if I wouldn't use one. 
The more important thing for me was finding the proper crystal. I tried several ones, and only those worked, which had a series resistance smaller than 70 kOhm (and they were SMD crystals in size 3x2 or 2x1.2 mm).

karelchik, 2023-06-14:
hi @xiongweichao, I did not have a resistor in parallel. Are you investigating this issue? If yes, please let us know about the details, I am still interested in what may have caused this. :)

xiongweichao, 2023-06-14:
Hi @karelchik , I reproduced this issue with ESP32-PICO-MINI-02. When I added a parallel resistor it didn't reproduce. You can add a parallel resistor to test it, and please let me know the test results. thanks！

karelchik, 2023-06-14:
hi @xiongweichao, that's promising. I will check if i still have those breakouts laying around.

karelchik, 2024-02-15:
Hi all. Just happened to brush past here and saw that I forgot to give an update. Better late than never, right? Sadly the update is going to be an uninteresting one: I no longer had those breakouts, so I could not test if the resistor helps. :(

closed 2024-07-24 by Alvin1Zhang