espressif/esp-idf #9142: New compile warnings after upgrading to esp-2022r1-RC1

AxelLin, 2022-06-13:
## Environment

- Module or chip used: ESP32-WROOM-32E
- IDF version: v5.0-dev-3481-g16a4ee7c36
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System: Linux
- Power Supply: USB

## Problem Description

Got new compile warnings aftre pull latest esp-idf update.

/home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/chap_ms.c:497:32: error: argument 2 of type 'const u_char[16]' {aka 'const unsigned char[16]'} with mismatched bound [-Werror=array-parameter=]
  497 |                   const u_char PasswordHash[MD4_SIGNATURE_SIZE],
      |                   ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/chap_ms.c:155:52: note: previously declared as 'const u_char *' {aka 'const unsigned char *'}
  155 | static void     ChallengeResponse (const u_char *, const u_char *, u_char[24]);
      |                                                    ^~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/chap_ms.c:875:16: error: argument 8 of type 'u_char[]' {aka 'unsigned char[]'} with mismatched bound [-Werror=array-parameter=]
  875 |         u_char authResponse[], int authenticator) {
      |         ~~~~~~~^~~~~~~~~~~~~~
/home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/chap_ms.c:179:29: note: previously declared as 'u_char[41]' {aka 'unsigned char[41]'}
  179 |                   u_char *, u_char[MS_AUTH_RESPONSE_LENGTH+1], int);
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
[536/1237] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/netif/ppp/lcp.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

antmak, 2022-06-13:
@AxelLin thank you for the catch! Indeed the new compiler is more sensitive in such cases. Please make a kind of change locally to comply the compiler:

```patch
diff --git i/src/netif/ppp/chap_ms.c w/src/netif/ppp/chap_ms.c
index 5a989c9b..60b86f25 100644
--- i/src/netif/ppp/chap_ms.c
+++ w/src/netif/ppp/chap_ms.c
@@ -152,7 +152,7 @@ extern void set_mppe_enc_types(int, int);
 
 static void    ascii2unicode (const char[], int, u_char[]);
 static void    NTPasswordHash (u_char *, int, u_char[MD4_SIGNATURE_SIZE]);
-static void    ChallengeResponse (const u_char *, const u_char *, u_char[24]);
+static void    ChallengeResponse (const u_char *, const u_char[MD4_SIGNATURE_SIZE], u_char[24]);
 static void    ChallengeHash (const u_char[16], const u_char *, const char *, u_char[8]);
 static void    ChapMS_NT (const u_char *, const char *, int, u_char[24]);
 static void    ChapMS2_NT (const u_char *, const u_char[16], const char *, const char *, int,
@@ -872,7 +872,7 @@ static void ChapMS(ppp_pcb *pcb, const u_char *rchallenge, const char *secret, i
  */
 static void ChapMS2(ppp_pcb *pcb, const u_char *rchallenge, const u_char *PeerChallenge,
        const char *user, const char *secret, int secret_len, unsigned char *response,
-       u_char authResponse[], int authenticator) {
+       u_char authResponse[MS_AUTH_RESPONSE_LENGTH+1], int authenticator) {
     /* ARGSUSED */
     LWIP_UNUSED_ARG(authenticator);
 #if !MPPE_SUPPORT

```

We will also add apparent changes in IDF soon!

AxelLin, 2022-06-13:
Hi @antmak 
Thanks for the prompt fix. Above change fixed this issue.