espressif/esp-idf #9144: Creating Custom Tables in ESP32 partition 3 MB app and 1 MB spiffs(4mb)

vinothkrstatiq, 2022-06-13:
I want to create 3 mb app and 1 mb spiffs partition   (4mb)esp32
I used CONFIG_PARTITION_TABLE_OFFSET 0x8000 
what are values should I  keep in the CSV file?

I tried many different values ,but could not succeed. if there is someone who has already implemented this partition can help me?

igrr, 2022-06-13:
You can try something like this (3MB app partition, 1MB - 64kB = 960kB SPIFFS partition):
```csv
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 3M,
storage,  data, spiffs,  ,        0xF0000,
```
or this (3MB - 64kB = 3008kB app partition, 1MB SPIFFS partition):
```csv
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 3008k,
storage,  data, spiffs,  ,        1M,
```
Don't forget to set the flash size to 4MB in menuconfig, under Serial flasher config menu.

vinothkrstatiq, 2022-06-13:
@igrr ,
Thanks for the help it worked for me

there some reference also I found
https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/huge_app.csv

closed 2022-06-13 by vinothkrstatiq