espressif/esp-idf #9146: about freeRTOS ERROR

KOOKOKOK, 2022-06-13:
I execute the following code. After a while, I will report an error, and then repeat the process. I use the esp32 development board and the editor vscode+esp IDF plug-in.

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"
#include <stdio.h>
#include"freertos/FreeRTOS.h"
#include "esp_log.h"
#include"freertos/task.h"
#include"nvs_flash.h"
static const char* TAG = "MyModule";

void myTask(){
    for (;;)
    {
        /* code */
        // ESP_LOGW(TAG, "Hello");
        // vPrintString("Hello");
        // vTaskDelay(1000);
    }
    
    vTaskDelay(1);
}
void app_main()
{
    ESP_ERROR_CHECK(nvs_flash_init());
    // esp_netif_init();
    // ESP_ERROR_CHECK(esp_event_loop_create_default());

    xTaskCreate(
        myTask,
        "mtTask",
        4096,
        NULL,
        1,
        NULL
    );
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
E (15390) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (15390) task_wdt:  - IDLE (CPU 1)
E (15390) task_wdt: Tasks currently running:
E (15390) task_wdt: CPU 0: IDLE
E (15390) task_wdt: CPU 1: mtTask
E (15390) task_wdt: Print CPU 0 (current core) backtrace
E (20390) task_wdt: Print CPU 1 backtrac
```

igrr, 2022-06-13:
That makes sense, mtTask is always running and prevents the Idle task on CPU1 from running. You can either modify the code of mtTask to yield once in a while, or disable Idle task watchdog on CPU1.

(I'll close https://github.com/espressif/esp-idf/issues/9145 as I assume you have accidentally opened two tickets for the same problem.)

KOOKOKOK, 2022-06-13:
ok,thanks.

closed 2022-06-13 by KOOKOKOK