espressif/esp-idf #9148: Creating esp32 image failes since compiler update?

0xFEEDC0DE64, 2022-06-13:
I'm compiling our existing firmware repo and after switching to the newest esp-idf version I got thousands of compiler errors, seems like gcc was finally updated from 8.4.0 to 11.2 (yea!).

But now after fixing all my compiler errors and linker errors I fail at the last step?

```
Creating esp32 image...
Merged 25 ELF sections

A fatal error occurred: secure_pad only applies if a SHA-256 digest is also appended to the image
ninja: build stopped: subcommand failed.
```

We are using secure boot and flash encryption

boarchuz, 2022-06-14:
Check that `ESPTOOLPY_FLASHSIZE_DETECT` ("Detect flash size when flashing bootloader") is disabled.

There was a recent change to the default value for this (y->n), and ``--dont-append-digest`` is now passed to elf2image when set.

This means that most existing projects will now rebuild _without appending a SHA256 to the image_ after updating IDF as their sdkconfig will have ESPTOOLPY_FLASHSIZE_DETECT set by default.

0xFEEDC0DE64, 2022-06-14:
thanks, that fixed it.

igrr, 2022-06-14:
Thanks for confirming this is indeed the case. Will keep this issue open until the fix is released.

closed 2022-06-21 by espressif-bot

derekthecool, 2025-01-07:
After some struggling I found that it was the sdkconfig value required to get past this same error was
`CONFIG_ESPTOOLPY_HEADER_FLASHSIZE_UPDATE=n`

Thanks for pointing me in the right direction with this ticket!