espressif/esp-idf #9149: NVS set and erase fail - flash appears to be read-only

jcolebaker, 2022-06-13:
## Environment

- Development Kit: none
- Module or chip used: ESP32-WROOM-32D
- IDF version:  ESP-IDF v4.3.1-29-ge493a4c30e-dirty
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): ??
- Operating System: Linux / Docker
- Using an IDE?: No
- Power Supply: external 5V

## Problem Description

We have run into an intermittent problem where **nvs_set_*** functions start to fail: They do not return any error code, but the new value is NOT added to the NVS. This effects updating an existing key (value doesn't change) and adding a new key (key not added; dump of all keys shows only keys which already existed). Once the problem has occurred,  **nvs_flash_erase** also fails to work. It returns a success code, but using nvs_entry_find / nvs_entry_next afterwards to iterate through the keys shows they have NOT been erased. The problem persists after a reboot, but can be fixed with "esptool.py erase_flash" command and firmware reinstall.

### Expected Behavior

NVS keys / values can be added / updated; All NVS keys can be erased with nvs_flash_erase

### Actual Behavior

NVS data becomes "read only"; update and erase don't work.

### Steps to reproduce

Intermittent

### Code to reproduce this issue

[Gist](https://gist.github.com/jcolebaker/20519fd1b4ce52e7cf08e6992fba9b41)

## Debug Logs

Output from failed "factory reset" (see gist code). NVS keys should be deleted but still exist.

```
I (2360495) WCUI: ************* FACTORY RESET *************
I (2360495) NVSC: ERASE NVS partition...
I (2360495) NVSC: NVS Erase finished.
I (2360525) NVSC: NVS Keys:
I (2360525) NVSC: Key: 'name_zone_1'; type 66
I (2360525) NVSC: Key: 'name_zone_2'; type 66
I (2360535) NVSC: Key: 'name_zone_3'; type 66
I (2360535) NVSC: Key: 'name_zone_4'; type 66
I (2360545) NVSC: Key: 'name_zone_5'; type 66
I (2360545) NVSC: Key: 'name_zone_6'; type 66
I (2360555) NVSC: Key: 'serial_engr'; type 33
I (2360565) NVSC: Key: '01_reg_data'; type 66
I (2360565) NVSC: Key: '02_reg_data'; type 66
I (2360575) NVSC: Key: 'use_stand_ble'; type 4
I (2360575) NVSC: Key: 'use_target_ble'; type 4
I (2360585) NVSC: Key: 'name_engr'; type 66
I (2360585) NVSC: NVS Stats: used entries 104; free entries 904; all entries 1008
```

## Other items if possible

Attached: Data read from flash after the bug occurred (including attempting to erase the nvs entries):
    flash.bin - First ~128 kB of flash, including partition table at 0x8000
    nvs_dat.bin - Just the nvs data partition

[flash_data.zip](https://github.com/espressif/esp-idf/files/8894416/flash_data.zip)

negativekelvin, 2022-06-13:
Possible some write protection bits got turned on in your flash chip?  See https://github.com/espressif/esp-idf/issues/7994

jcolebaker, 2022-06-27:
SOLVED:

We are now 99% sure this problem is actually caused by stray voltage on the MTDI pin during power up and boot; the bug occurs after a power interruption, and the boot log shows the strapping pins as 0x33, meaning VDD_SDIO has been set to 1.8v instead of 3.3v. 

This was not noticed on our original boards because the flash still worked correctly even at the lower voltage. However, more recent boards have a later ESP module with a different flash chip. I have now confirmed that on these boards, the flash can be read but cannot be written when VDD_SDIO has been set to 1.8v.

I will close this issue since it's not a bug with the flash or nvs libraries.

closed 2022-06-27 by jcolebaker

pacucha42, 2022-06-27:
@jcolebaker thank you for your investigation and sharing the results - closing our internal task as well