espressif/esp-idf #9150: error: "MEMP_NUM_SYS_TIMEOUT" redefined after upgrading to esp-idf-v5.0

AxelLin, 2022-06-14:
## Environment

- Module or chip used: ESP32-WROOM-32E
- IDF version: v5.0-dev-3424-gbb23d783c0
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System: Linux
- Power Supply: USB

## Problem Description

I have below in my CMakeLists.txt because I use the mdns from lwip.
target_compile_definitions(${lwip} PUBLIC -DMEMP_NUM_SYS_TIMEOUT=LWIP_NUM_SYS_TIMEOUT_INTERNAL+LWIP_MDNS_RESPONDER)

However, I got below build error after pull recent changes in esp-idf master / v5.0 branch.
/home/axel/esp/esp-idf/components/lwip/port/esp32/include/lwipopts.h:1372: error: "MEMP_NUM_SYS_TIMEOUT" redefined [-Werror]
 1372 | #define MEMP_NUM_SYS_TIMEOUT            LWIP_NUM_SYS_TIMEOUT_INTERNAL_WOUT_IGMP_MLD6

The commit a6154a5f1060  causes problem.

AxelLin, 2022-07-08:
After update to v5.0-dev-4001-g495d35949d, now I got another similar error.
/home/axel/esp/esp-idf/components/lwip/port/esp32/include/lwipopts.h:669: error: "LWIP_NUM_NETIF_CLIENT_DATA" redefined [-Werror]

I have below line in my CMakeLists.txt
target_compile_definitions(${lwip} PUBLIC -DLWIP_MDNS_RESPONDER=1 -DLWIP_NUM_NETIF_CLIENT_DATA=1 PRIVATE -DLWIP_NETIF_EXT_STATUS_CALLBACK=1)
So I got redefined error. (It was fine when build with esp-idf-v4.x)

I hope my application can be compiled with differet esp-idf versions.
What's your suggestion to fix this?

AxelLin, 2022-07-14:
@david-cermak 

I think esp-idf should not define MEMP_NUM_SYS_TIMEOUT otherwise the application cannot change it.
If esp-idf define MEMP_NUM_SYS_TIMEOUT, you must provide a way for users to change the setting.

This is not a big issue but it's an annoy issue because now I need to either modify esp-idf or my CMakeLists.txt
every time when build with esp-idf-v5.0.

david-cermak, 2022-07-14:
@AxelLin

From IDF `v5.0` we're switching to lwip's `2.1.3` and have reworked many Espressif's patches from `esp-lwip` to IDF's port layer in [`lwip/port`](https://github.com/espressif/esp-idf/tree/master/components/lwip/port/esp32) (in our effort to converge to lwip upstream).
The benefit for users is that we can simply switch to upstream; the drawback is that will have to find a way for users to apply these "unsupported" (=missing in menuconfig) options (something similar what's been done with the hook file: lwip's hook file was taken by IDF hooks, but provided a means to inject an IDF-lwip hook file)

As for the `MEMP_NUM_SYS_TIMEOUT`, you're correct. The reason to update it was the patch applied directly to lwip's code 
(https://github.com/espressif/esp-lwip/commit/33e3d3eb2ffbb15e4aaa17115c9c216abb3624c8) and thus the config opts should be defined/updated in the same place. The fix for this issue still pending, sorry for the delay. You can evaluate it by switching to [`2.1.3-esp-dev`](https://github.com/espressif/esp-lwip/commits/2.1.3-esp-dev) and deleting these lines:

https://github.com/espressif/esp-idf/blob/9b8c558e63d95b491355a22e69f247834e899b47/components/lwip/port/esp32/include/lwipopts.h#L1408-L1409

closed 2022-07-19 by espressif-bot

david-cermak, 2022-07-21:
@AxelLin About your comment above https://github.com/espressif/esp-idf/issues/9150#issuecomment-1178575555 and the `LWIP_NUM_NETIF_CLIENT_DATA`: Just wanted to ask if you use any non-default config number in

https://github.com/espressif/esp-idf/blob/c585618b97d47514b7bf5f361944a8af05348169/components/lwip/Kconfig#L253-L258

Or is it set to `0` via `sdkconfig.h`?

AxelLin, 2022-07-21:
Hi @david-cermak 

Now I use below workaround:
In my CMakeLists.txt:
```
if (${IDF_VERSION_MAJOR} STRGREATER "4")        #IDF-v5.0+
# Set LWIP_NUM_NETIF_CLIENT_DATA settings via sdkconfig.defaults CONFIG_LWIP_NUM_NETIF_CLIENT_DATA
target_compile_definitions(${lwip} PUBLIC -DLWIP_MDNS_RESPONDER=1 PRIVATE -DLWIP_NETIF_EXT_STATUS_CALLBACK=1)
else()  #IDF-v4.x
target_compile_definitions(${lwip} PUBLIC -DLWIP_MDNS_RESPONDER=1 -DLWIP_NUM_NETIF_CLIENT_DATA=1 PRIVATE -DLWIP_NETIF_EXT_STATUS_CALLBACK=1)
endif()
```

Then also set LWIP_NUM_NETIF_CLIENT_DATA=1 in my sdkconfig.default