espressif/esp-idf #9152: TWAI Driver - How to use ISR to get the receiving data and alerts and Listen mode Bit error

dev-techshlok, 2022-06-14:
Hello,

I am trying to get the interrupt working for TWAI driver. 
I am trying to get an interrupt at the time when any Rx data is available. But it does not seem to work.

**BUG:** Also when I am using the built-in example for Listen mode only if I choose the wrong baud rate for TWAI my car's engine light turns on but listen-only mode must not interfere with the BUS.

**I am using the esp-idf version :  ESP-IDF v4.4-dirty
The platform is Windows 8.1**

An example is posted below
```
/* TWAI Network Listen Only Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

/*
 * The following example demonstrates a Listen Only node in a TWAI network. The
 * Listen Only node will not take part in any TWAI bus activity (no acknowledgments
 * and no error frames). This example will execute multiple iterations, with each
 * iteration the Listen Only node will do the following:
 * 1) Listen for ping and ping response
 * 2) Listen for start command
 * 3) Listen for data messages
 * 4) Listen for stop and stop response
 */
#include <stdio.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "freertos/semphr.h"
#include "esp_err.h"
#include "esp_log.h"
#include "driver/twai.h"

/* --------------------- Definitions and static variables ------------------ */
//Example Configuration
#define NO_OF_ITERS                     3
#define RX_TASK_PRIO                    9
#define TX_GPIO_NUM                     CONFIG_EXAMPLE_TX_GPIO_NUM
#define RX_GPIO_NUM                     CONFIG_EXAMPLE_RX_GPIO_NUM
#define EXAMPLE_TAG                     "TWAI Listen Only"

#define ID_MASTER_STOP_CMD              0x0A0
#define ID_MASTER_START_CMD             0x0A1
#define ID_MASTER_PING                  0x0A2
#define ID_SLAVE_STOP_RESP              0x0B0
#define ID_SLAVE_DATA                   0x0B1
#define ID_SLAVE_PING_RESP              0x0B2

static const twai_filter_config_t f_config = TWAI_FILTER_CONFIG_ACCEPT_ALL();
static const twai_timing_config_t t_config = TWAI_TIMING_CONFIG_500KBITS();
//Set TX queue length to 0 due to listen only mode
static const twai_general_config_t g_config = {.mode = TWAI_MODE_LISTEN_ONLY,
                                               .tx_io = TX_GPIO_NUM, .rx_io = RX_GPIO_NUM,
                                               .clkout_io = TWAI_IO_UNUSED, .bus_off_io = TWAI_IO_UNUSED,
                                               .tx_queue_len = 0, .rx_queue_len = 5,
                                               .alerts_enabled = TWAI_ALERT_ALL,
                                               .clkout_divider = 0
                                              };

static SemaphoreHandle_t rx_sem;

/* --------------------------- Tasks and Functions -------------------------- */

void check_erros() {
    // while (true) {
    uint32_t alerts;
    // printf("cheching erros\n");
    printf("In loop:%d\n", xPortGetCoreID());
    twai_read_alerts(&alerts, portMAX_DELAY);
    printf("%08x", alerts);
    if (alerts & TWAI_ALERT_ABOVE_ERR_WARN) {
        ESP_LOGI(EXAMPLE_TAG, "Surpassed Error Warning Limit");
    }
    if (alerts & TWAI_ALERT_ERR_PASS) {
        ESP_LOGI(EXAMPLE_TAG, "Entered Error Passive state");
    }
    if (alerts & TWAI_ALERT_BUS_OFF) {
        ESP_LOGI(EXAMPLE_TAG, "Bus Off state");
        // Prepare to initiate bus recovery, reconfigure alerts to detect bus recovery completion
        twai_reconfigure_alerts(TWAI_ALERT_BUS_RECOVERED, NULL);
        // for (int i = 3; i > 0; i--) {
        //     ESP_LOGW(EXAMPLE_TAG, "Initiate bus recovery in %d", i);
        //     vTaskDelay(pdMS_TO_TICKS(1000));
        // }
        // twai_initiate_recovery();    //Needs 128 occurrences of bus free signal
        ESP_LOGI(EXAMPLE_TAG, "Initiate bus recovery");
    }

    if (alerts & TWAI_ALERT_BUS_ERROR) {
        ESP_LOGI(EXAMPLE_TAG, "Bus Error state");
        // Prepare to initiate bus recovery, reconfigure alerts to detect bus recovery completion
        twai_reconfigure_alerts(TWAI_ALERT_BUS_RECOVERED, NULL);
        // for (int i = 3; i > 0; i--) {
        //     ESP_LOGW(EXAMPLE_TAG, "Initiate bus recovery in %d", i);
        //     vTaskDelay(pdMS_TO_TICKS(1000));
        // }
        // twai_initiate_recovery();    //Needs 128 occurrences of bus free signal
        ESP_LOGI(EXAMPLE_TAG, "Initiate bus recovery");
    }
   

    if (alerts & TWAI_ALERT_BUS_RECOVERED) {
        //Bus recovery was successful, exit control task to uninstall driver
        ESP_LOGI(EXAMPLE_TAG, "Bus Recovered");
        // break;
    }

}
static void twai_receive_task(void *arg)
{
    xSemaphoreTake(rx_sem, portMAX_DELAY);
    bool start_cmd = false;
    bool stop_resp = false;
    uint32_t iterations = 0;

    while (iterations < NO_OF_ITERS) {
        twai_message_t rx_msg;
        twai_receive(&rx_msg, portMAX_DELAY);
        check_erros();
        printf("Received identifier:%04x ", rx_msg.identifier);
        // printf("Received frame:%04x ",rx_msg.data_length_code);
        for (int i = 0 ; i < 8; i++) {
            printf("D Byte:%d : %04x ", i, rx_msg.data[i]);
        }
        printf("\n");
        vTaskDelay(pdMS_TO_TICKS(10));

    }

    xSemaphoreGive(rx_sem);
    vTaskDelete(NULL);
}

static void CAN_isr(void *arg_p) {
    printf("i was here");
}

void app_main(void)
{
    rx_sem = xSemaphoreCreateBinary();
    xTaskCreatePinnedToCore(twai_receive_task, "TWAI_rx", 4096, NULL, RX_TASK_PRIO, NULL, tskNO_AFFINITY);

    //Install and start TWAI driver
    ESP_ERROR_CHECK(twai_driver_install(&g_config, &t_config, &f_config));
    ESP_LOGI(EXAMPLE_TAG, "Driver installed");
    esp_intr_alloc(ETS_CAN_INTR_SOURCE, 0, CAN_isr, NULL, NULL);
    ESP_ERROR_CHECK(twai_start());
    ESP_LOGI(EXAMPLE_TAG, "Driver started");

    xSemaphoreGive(rx_sem);                     //Start RX task
    // vTaskDelay(pdMS_TO_TICKS(100));
    xSemaphoreTake(rx_sem, portMAX_DELAY);      //Wait for RX task to complete

    //Stop and uninstall TWAI driver
    ESP_ERROR_CHECK(twai_stop());
    ESP_LOGI(EXAMPLE_TAG, "Driver stopped");
    ESP_ERROR_CHECK(twai_driver_uninstall());
    ESP_LOGI(EXAMPLE_TAG, "Driver uninstalled");

    //Cleanup
    vSemaphoreDelete(rx_sem);
}

```
Please help me if possible. I was unable to find a clean example for ISR.

dev-techshlok, 2022-06-14:
@Dazza0 Please can you look into it.

micheleciaramicoli, 2022-11-16:
HAVE YOU SOLVED IT? I'm trying to do the same, recieve via twai with interrupts (it appears to me that twai_recieve API is not working in interrupt mode)

Dazza0, 2023-02-27:
> BUG: Also when I am using the built-in example for Listen mode only if I choose the wrong baud rate for TWAI my car's engine light turns on but listen-only mode must not interfere with the BUS.

@micheleciaramicoli There was a HW errata with the TWAI controller that allows active error frames to be sent even if the TWAI controller was in listen only mode (thus why it was interfering with your bus). A SW workaround has been merged on master (see https://github.com/espressif/esp-idf/commit/57b6be22a7779bbfd194ba254f66bdbe95a18a7c) and this will be backported to our currently oldest service release. I'll tag the backport commits to this issue when they have been merged.

closed 2023-02-27 by Dazza0

Dansenmansenhansen, 2023-09-27:
Hello!

I saw this thread beeing marked as closed, even though noone jumped on the part in which he asked:

> I am trying to get the interrupt working for TWAI driver. I am trying to get an interrupt at the time when any Rx data is available.

I neither found a way to get receiving Can-Telegrams notified and/or executed in an Interrupt. Examples and API don't seem to be made for Interrupt use. Is it(like most of the time) programmers fault, or is my assumption right this time?

Thanks in advance.

Dazza0, 2023-10-11:
@Dansenmansenhansen 

> I am trying to get the interrupt working for TWAI driver. I am trying to get an interrupt at the time when any Rx data is available.

The current driver API design does not intend for users to interact with the interrupt directly. The driver is buffered thus it will already handle all RX interrupts.

If for whatever reason you still require an RX interrupt (e.g., writing a bare metal driver), you'll need to parse the RX interrupt yourself. Refer to the TWAI Peripheral chapter in the ESP32's (or whichever relevant target) technical reference manual. The chapter will describe how the TWAI peripheral's interrupts operate, and how the interrupt status register works.

If you still have questions/issues, please open a new GitHub issue.

Oxygensun, 2024-11-09:
I can understand that driver has buffered all msg, but now can I got timestamp of each can msg?
can msg worked on car's, that mean even 1ms timestamp is differante. some msg must get response real time.