espressif/esp-idf #9156: Application fails to boot in v4.3 when SPI flash freq differs between bootloader and application

zevv, 2022-06-14:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

KonstantinKondrashov, 2022-06-15:
Hi @zevv!
Regarding commit 2373f11, it should not affect on any things because the max value in pkg field up to nowday still less or equal 7 . You can run **esptool.py chip_id** to make sure what exactly chip inside it. ESP32-WROOM-32D can be build on any chips, see https://espressif.com/en/products/socs.

Based on the log, probably the app is booted but stucked some where in the startup of app. I would suggest to set the verbos log for app to make sure in it. The next logs after "Disabling RNG ...." would be as below from app. 
I (189) cpu_start: Pro cpu up.
I (189) cpu_start: Starting app cpu, entry point is 0x4008111c
I will check it how to fix it.

zevv, 2022-06-15:
Hi, output of `chip_id` below.

```
$ esptool.py chip_id
esptool.py v3.1-dev
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 10:52:1c:52:2c:d4
Uploading stub...
Running stub...
Stub running...
Warning: ESP32 has no Chip ID. Reading MAC instead.
MAC: 10:52:1c:52:2c:d4
Hard resetting via RTS pin...
```

zevv, 2022-06-15:
App log at verbose gets me to here, after this nothing happens until the reboot:

```
I (0) cpu_start: App cpu up.
D (306) clk: RTC_SLOW_CLK calibration value: 3068954
```

My application at this time is only one `printf()`, the rest is essentially all unreachable code; I must admit I do not understand at all how 2373f11 can be related to my code not starting up, but still it's happening.

KonstantinKondrashov, 2022-06-15:
I tried hello_world on master. bootloader with 40Mhz and app with 80Mhz (ESPTOOLPY_FLASHFREQ), App is stucked looks like at the same place. 
@zevv could you try this fix for app?
```diff
diff --git a/components/bootloader_support/src/bootloader_efuse.c b/components/bootloader_support/src/bootloader_efuse.c
index ac5f9f7..9b1556a 100644
--- a/components/bootloader_support/src/bootloader_efuse.c
+++ b/components/bootloader_support/src/bootloader_efuse.c
@@ -9,13 +9,14 @@
 #include "bootloader_common.h"
 #include "hal/efuse_ll.h"
 #include "hal/efuse_hal.h"
+#include "esp_attr.h"
 
-uint8_t bootloader_common_get_chip_revision(void)
+IRAM_ATTR uint8_t bootloader_common_get_chip_revision(void)
 {
     return efuse_hal_get_chip_revision();
 }
 
-uint32_t bootloader_common_get_chip_ver_pkg(void)
+IRAM_ATTR uint32_t bootloader_common_get_chip_ver_pkg(void)
 {
     return efuse_ll_get_chip_ver_pkg();
 }
```

zevv, 2022-06-15:
Right, you hit the spot, that's some excellent quality support there!

I was already printf()ing my way through the boot processes and ended up in the same place, but I hadn't realized it was related to IRAM execution.

Is there a chance this will get backported to 4.3 as well?

Thanks!

KonstantinKondrashov, 2022-06-15:
@zevv Thanks a lot for confirmation! Yes, I will create backport for 4.4 and 4.3 as well.

closed 2022-06-21 by espressif-bot