espressif/esp-idf #9158: Enhancement request for console component help feature

dalbert2, 2022-06-14:
I have always implemented a command line interface for my projects that's similar to the console component.  I'm trying to switch to IDF standard components whenever possible. but an obstacle with the console component is the implementation of the help feature which generates excessive output once the number of commands grows larger than a trivial example.  

For my own component, I implement help as a two-tier function: 

1. the help command alone lists all commands, separated by commas and at the end suggests 
"use help <command> for more details".

2. help <command> displays the detailed help for the specified command

This way a system with more than a few commands can generate reasonable amounts of output for any particular help request.

I'd be happy to modify the component code to do this if the maintainers agree it is desirable and permit me to do so.  Otherwise, please consider this as an enhancement request.

dalbert2, 2022-06-15:
Thanks @igrr, I mentioned the second enhancement (minimum password security) because it is so easy to implement and makes a significant improvement to the component. Virtually all CLI consoles on commercial devices have at least a minimum level of security. The workaround you propose is interesting, but there appears to be no way to un-register commands, so you could log in that way, but you would have to reset the unit to log out. 

The enhancement proposed is both easy to implement and adds important functionality. I'm new to IDF so I don't know if contributions are allowed, if they are, I'm happy to make these changes and submit for review; if not, I'll do as you suggest and open PRs for both (unless you object to the second).

igrr, 2022-06-15:
> The workaround you propose is interesting, but there appears to be no way to un-register commands, so you could log in that way, but you would have to reset the unit to log out.

It is possible to un-initalize the console and re-initialize it again, as a lighter weight alternative to resetting the device.

We can also accept a PR for un-registering commands, this sounds like a reasonable request!

Typically we don't add additional functionality to core components, which
1. can be easily implemented in the application, or
2. different applications might require slightly different implementations of this functionality, or
3. isn't required for common use cases or applications.

The request for a feature to log into the console hasn't been seen on the IDF feature tracker before. It is something that can be easily implemented in the application (just about 20 lines of code, as shown above). Different users may require different ways the passwords are validated. For these reasons I don't think we will accept a PR for this feature. Of course, if more developers chime in and say that they would like to see such functionality in IDF, we can reconsider this.

I'm sorry if my reply sounds discouraging. For your first request (help command) we definitely welcome a PR.

igrr, 2022-06-15:
Here is the updated example which demonstrates logging out by de-initializing the console REPL.

[0001-add-an-example-of-login-command.patch.txt](https://github.com/espressif/esp-idf/files/8911440/0001-add-an-example-of-login-command.patch.txt)

louielauanu, 2024-05-20:
Hi @dalbert2,

We develop a feature that enable users to decide the help command verbose level, and that will be available soon.

closed 2024-05-30 by espressif-bot