espressif/esp-idf #9159: Compiling bootloader with `-O2` not working

KaeLL, 2022-06-14:
- IDF version: 20f5e180eecccfaff815d707e1fcbba2f4d6a391

If `BOOTLOADER_COMPILER_OPTIMIZATION_PERF=y`, [this](https://github.com/espressif/esp-idf/blob/20f5e180eecccfaff815d707e1fcbba2f4d6a391/components/bootloader_support/src/bootloader_common_loader.c#L160-L164) code triggers Werror despite the ignore directive. Compilation works if I comment out the `-Warray-bounds` pragma, which is weird, but even weirder is that after doing that, the warning for `-Warray-bounds` correctly shows up, but it's erroneously not treated as error. Not sure what's happening. Apparently something changed about either or both of those warnings and/or GCC isn't handling the pragmas correctly.
```
In function 'bootloader_common_reset_rtc_retain_mem',
    inlined from 'bootloader_common_update_rtc_retain_mem' at /esp-idf/components/bootloader_support/src/bootloader_common_loader.c:187:13:
/esp-idf/components/bootloader_support/src/bootloader_common_loader.c:163:5: error: 'memset' writing 16 bytes into a region of size 0 overflows the destination [-Werror=stringop-overflow=]
  163 |     memset(rtc_retain_mem, 0, sizeof(rtc_retain_mem_t));
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
```

ESP-Marius, 2022-06-15:
Hmm, could it be that this is due to the function getting inlined, and the GCC pragmas fails to correctly handle this. If I force it to never be inlined (`__attribute__((noinline))` ) then the issue goes away.

KaeLL, 2022-06-15:
Apparently [it's a bug and seems it was fixed for GCC 12](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=55881). Now what?

0xjakob, 2022-06-16:
@KaeLL Introducing a new GCC version is not exactly trivial. Before we do that, we have to do a lot of tests to be reasonably sure there aren't any regressions (speed, memory, code size, etc.). If we don't, then these problems will appear on user's devices, maybe also yours. Of course, we also need to fix everything that breaks during the update for obvious reasons.

I can understand that some problems that got fixed in the newest version of some upstream project create frustration if they still occur in IDF. But we also have to keep the balance for long-term users on the release branches who need the latest IDF updates but certainly don't want some weird size regression or runtime behavior on their devices in the field (I've worked as application engineer myself). In the end, we can only maintain a limited number of toolchains.

If you have a better idea, feel free to let us know.

That all being said, if the patch is easy to apply, we might be able to backport it into our compiler version during the next toolchain release, but yes, I'm afraid you will have to wait for that.

igrr, 2022-06-16:
Thanks for pointing this out @KaeLL. From the first glance, backporting this fix to GCC 11 looks non-trivial. We will likely introduce helper functions (`memset_region`, `memcpy_region` or similar) to be used specifically when the destination address is obtained from an integer constant or a linker label.

KaeLL, 2022-06-17:
I'm under the impression I may have come across as pedantic in the way I phrased my comment, and if that was the case, it wasn't intentional and I'm sorry. As for the idea part, what about "canary" builds of the toolchain, like browsers? I'm just throwing this to see if it sticks, no idea if the model fits the constraints of toolchain porting/development.

closed 2022-08-09 by KaeLL